espressif/esp-adf #1237: 关于AEC效果问题请教，如何可以让AEC质量变得更好？

laodi-chen, 2024-07-19:
## Environment

- Audio development kit: ESP32-LyraT-Mini
- Audio kit version: v1
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.4.7
- ADF version: v2.6-82-g33ba0c5c
- Build system: idf.py
- Running log:
- Operating system: Windows
- Environment type: ESP Command Prompt
- Using an IDE?: Yes(Visual Studio Code1.87.2)
- Power supply: USB

## Problem Description

#### 因为在空旷场地所以需要声音要大一些，我调整了以下
1. `audio_hal_set_volume(board_handle->audio_hal, 100);` 原为85
2. `#define BOARD_PA_GAIN             (8)`  原为20

#### 并添加了麦克风增益
1. `#define AUDIO_HAL_VOL_DEFAULT 100` 原为70
2. `#define ALGORITHM_STREAM_DEFAULT_AGC_GAIN_DB      45` 原为5
4. `#define ALGORITHM_STREAM_DEFAULT_MASK    (ALGORITHM_STREAM_USE_AEC | ALGORITHM_STREAM_USE_NS | ALGORITHM_STREAM_USE_AGC)` 添加了AGC

#### 对于AEC效果调试如下
1. 我首先调整了参考增益，将3调整为1后确实回音有所改善，但是第二天又变差了不知为何？
`algo_config.ref_linear_factor = 1;`
2. 尝试添加录制增益，但效果变得更差
`algo_config.rec_linear_factor = 3;`
3. 测量信号延迟

#### 测试环境
1. sip服务器`Asterisk`
2. sip工具`Linphone for Win，Linphone for Android`

求指点如何能让效果更好

## Environment

1. 使用的驻极体麦克风参数
[image]
4. 使用的喇叭参数
[image]
5. 网络环境
[image]
6. 摆放位置
[image]

laodi-chen, 2024-07-29:
@jason-mao 老哥来看看怎么搞？

jason-mao, 2024-07-30:
@laodi-chen 你这AEC 是完全没消除。摆放位置的说明中，喇叭内腔 和mic 是完全隔离的吗？也就是说声音只有喇叭正前面一个路径？另外，你把回采的信号和播放信号分开存一份来看看。

ktoto2011, 2024-08-07:
> @laodi-chen 你这AEC 是完全没消除。摆放位置的说明中，喇叭内腔 和mic 是完全隔离的吗？也就是说声音只有喇叭正前面一个路径？另外，你把回采的信号和播放信号分开存一份来看看。
你好，我这边实现voip 的时候也遇到相同的情况AEC 的效果很差，安装的位置和这差不多，请问一下，有什么办法定位和解决吗？

jason-mao, 2024-08-07:
@ktoto2011 结构上保证喇叭的声音只有正向的一个路径到mic，同时mic 加胶圈防振动。你可以把原始音频、参考音频、mic音频存下来分析一下。

laodi-chen, 2024-08-20:
@jason-mao 感谢老哥还记得

后续我又尝试了不同规格的麦克风，以及悬空麦克风喇叭不与外壳产生关联，但是效果都不是很好。

因为项目上的比较急，我用了最笨的办法，当对端讲话时降低设备麦克风增益抑制声音回传。

这周我抽空把发送到设备的声音与接受的回音分离成两个声道录制，再补充一份录音文件。

bakecool, 2025-02-24:
@laodi-chen 主要看你是否真的得到了消音参考（所谓回声的原声）。