espressif/esp-adf #1240: ESP-ADF Release/v2.7

estevaofv, 2024-07-28:
Hello, in README already talking about ESP-ADF
Release/v2.7 compatible with IDF 5.3 when the new ADF version will be releases? or It already have been released and you just forgot to create the tag for v2.7?
Thanks in advance!

jason-mao, 2024-07-29:
@estevaofv The ADF v2.7 is compatiable the IDF v5.3, it's not released, we are working on it.

jason-mao, 2024-07-29:
Related https://github.com/espressif/esp-adf/issues/1205

kscz, 2024-08-23:
Do you happen to have a timeline or just a rough estimate of how soon until the new release? #1205 just has the comment "This won't take long." which is pretty ambiguous.

jason-mao, 2024-12-10:
The ADF v2.7 was released in https://github.com/espressif/esp-adf/releases/tag/v2.7

closed 2024-12-10 by jason-mao