espressif/esp-adf #1242: Iam streaming the audio from sdcard to connected device

MSalman9133, 2024-08-02:
/*
 * SPDX-FileCopyrightText: 2021-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "nvs.h"
#include "nvs_flash.h"
#include "esp_system.h"
#include "esp_log.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"
#include "esp_bt.h"
#include "bt_app_core.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include "esp_gap_bt_api.h"
#include "esp_a2dp_api.h"
#include "esp_avrc_api.h"
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "dirent.h"
#include "driver/i2s.h"
#define MOUNT_POINT "/sdcard"
static const char *TAG = "SDCARD";
#define SDCARD_PWR_CTRL GPIO_NUM_13
/* log tags */
#define BT_AV_TAG             "BT_AV"
#define BT_RC_CT_TAG          "RC_CT"

/* device name */
#define TARGET_DEVICE_NAME    "Mivi DuoPods K2"
#define LOCAL_DEVICE_NAME     "ESP_A2DP_SRC"

#define AUDIO_BUFFER_SIZE 1024
static FILE *audio_file = NULL;  // Global file pointer for the audio file
static uint8_t audio_buffer[AUDIO_BUFFER_SIZE];  // Buffer to read audio data#define AUDIO_BUFFER_SIZE 1024

/* AVRCP used transaction label */
#define APP_RC_CT_TL_GET_CAPS            (0)
#define APP_RC_CT_TL_RN_VOLUME_CHANGE    (1)

enum {
    BT_APP_STACK_UP_EVT   = 0x0000,    /* event for stack up */
    BT_APP_HEART_BEAT_EVT = 0xff00,    /* event for heart beat */
};

/* A2DP global states */
enum {
    APP_AV_STATE_IDLE,
    APP_AV_STATE_DISCOVERING,
    APP_AV_STATE_DISCOVERED,
    APP_AV_STATE_UNCONNECTED,
    APP_AV_STATE_CONNECTING,
    APP_AV_STATE_CONNECTED,
    APP_AV_STATE_DISCONNECTING,
};

/* sub states of APP_AV_STATE_CONNECTED */
enum {
[... 853 lines cut ...]

    if (esp_bluedroid_enable() != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s enable bluedroid failed", __func__);
        return;
    }

#if (CONFIG_EXAMPLE_SSP_ENABLED == true)
    /* set default parameters for Secure Simple Pairing */
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
#endif

    /*
     * Set default parameters for Legacy Pairing
     * Use variable pin, input pin code when pairing
     */
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
    esp_bt_pin_code_t pin_code;
    esp_bt_gap_set_pin(pin_type, 0, pin_code);

    bt_app_task_start_up();
    /* Bluetooth device name, connection mode and profile set up */
    bt_app_work_dispatch(bt_av_hdl_stack_evt, BT_APP_STACK_UP_EVT, NULL, 0, NULL);
}

hbler99, 2024-08-05:
It's recommended to describe your problem in a word which help us understand it better, thx!

MSalman9133, 2024-08-07:
[image]
[image]
1.Iam facing this issue using example code (D:\ESP_ADF\espadf\examples\player\pipeline_a2dp_source_stream)
2.Able to discover the connection coming to audio transmission unable to do

MSalman9133, 2024-08-07:
> It's recommended to describe your problem in a word which help us understand it better, thx!

ignore above code which i was attached right now iam using example code for audio streaming

hbler99, 2024-08-09:
Maybe set `menuconfig > Component config > FAT Filesystem support > Long filename support` can solve your problem. Additionally, just like the description in red line, you need appley the patch in $IDF_PATH

closed 2025-02-17 by jason-mao