espressif/esp-adf #1243: VOIP Example calls don't hang-up with Asteriks

SinnoTogo, 2024-08-03:
Hello:

Here my setup:
esp-idf: V5.1.1
esp-adf: V2.6
Hardware: ESP32-S3-KORVO-2 3.1
PBX: Asterisks running on a rapsberry pi

In general the system works, I can make and receive calls with audio.

But I have problems to hang up calls.

If the call is hang-up by the ESP32 then everything is OK
If the call is hang-up by the other end (ie. Microsip or similar) then the ESP32 doesn't hang-up until I explicitly send the esp_rtc_bye command (ie. by ussing a button press on the board).

Here the SIP flow diagram for a case where the end device hangs-up:
[image]
As you can see the BYE reques is sent by Asteriks but the ESP32 never procesees it.

Next a SIP flow diagra for a case where the ESP32 hangs-up:

[image]

In this case the correct sequence is followed and then the call ends properly.

Any idea or hint how to solve the issue?

Thanks!

TempoTian, 2024-08-08:
HI I have tested with FreeSwitch when peer send bye, ESP32 can received correctly.   
Could you please check on the latest version, just update `esp-adf-libs/esp_media_protocols` to latest version and try whether it fixed or not? We just commit a solution to fix when resend invite ESP32 response not with SDP may cause server send wrong response. Pay attention latest version rename `esp_rtc_init/deinit ` to `esp_rtc_service_init/deinit` to avoid conflict with IDF rtc clock API.
[image]

TempoTian, 2024-08-08:
Here is ESP32 received BYE and response log. You can compare it with your serial log. If possible send me both serial log and related fetched wireshark network packet, I will help to to analysis the possible reasons.
```
 (53903) SIP: 

BYE sip:1505@192.168.137.27:64339;transport=TCP SIP/2.0
Via: SIP/2.0/TCP 52.81.228.8:9670;rport;branch=z9hG4bKmX9Sv2176Qtgr
Max-Forwards: 70
From: <sip:1509@52.81.228.8:9670>;tag=80099F0aBrtcK
To: <sip:1505@52.81.228.8:9670>;tag=-2029613454
Call-ID: 2A20026AF866CBA21DA7A0E86AFD84A97003AEF7F810
CSeq: 86988519 BYE
User-Agent: FreeSWITCH-mod_sofia/1.10.6-release~64bit
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: path, replaces
Reason: Q.850;cause=16;text="NORMAL_CLEARING"
Content-Length: 0

I (53961) SIP: [1970-01-01/00:00:26]<<======================
I (53963) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_END
W (53974) AUDIO_PIPELINE: There are no listener registered
I (53975) AUDIO_PIPELINE: audio_pipeline_unlinked
I (53977) AV_STREAM: _audio_dec task stoped
I (53988) SIP_SERVICE: ESP_RTC_EVENT_HANGUP
W (53989) SIP: CHANGE STATE FROM 96, TO 2, :func: sip_uas_process_req:1043
I (54000) SIP: [1970-01-01/00:00:26]=======WRITE 0418 bytes>>
I (54001) SIP: 

SIP/2.0 200 OK
Via: SIP/2.0/TCP 52.81.228.8:9670;rport;branch=z9hG4bKmX9Sv2176Qtgr
Contact: <sip:1505@192.168.137.27:64339;transport=TCP>
From: <sip:1509@52.81.228.8:9670>;tag=80099F0aBrtcK
To: <sip:1505@52.81.228.8:9670>;tag=-2029613454
Call-ID: 2A20026AF866CBA21DA7A0E86AFD84A97003AEF7F810
CSeq: 86988519 BYE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, UPDATE, CANCEL, OPTIONS, INFO
Content-Length: 0

```

jason-mao, 2024-12-10:
@SinnoTogo Have you solved your problem?

jason-mao, 2025-02-17:
This topic has become inactive, so I'm closing the issue. Please reopen it if you have any questions or need any further assistance.

closed 2025-02-17 by jason-mao

davidmavis, 2025-03-30:
Hi All. Need help with freepbx + ESP-RTC setup.
Can receive and send audio calls, but video doesn't work. Also, ESP is registering with different ports and this cause the problem of sometimes Asterisk not accepting registration.
Does anyone have same problem? Could anyone please post some tips which works?

TempoTian, 2025-03-31:
@davidmavis Can you create a seperate thread to discuss your issue? Please add your detail serial log and server expected behavior, I will help you to check the possible reasons.