espressif/esp-adf #1244: esp_codec_dev: Breaking change in v1.2.0

tore-espressif, 2024-08-06:
esp_codec_dev v1.2.0 now uses I2C Driver-NG https://github.com/espressif/esp-adf/blob/d2613ac466375f34711bf88ab49b5ed047b8df79/components/esp_codec_dev/CHANGELOG.md#features

This should be considered a breaking change as users with esp_codec_dev v1.1.0 are forced to use Driver-NG in their application.

I suggest removing version 1.2.0 and releasing it as v2.0.0

TempoTian, 2024-08-08:
One question, if new bsp like P4 is supported and it use new `i2c_master`, then use new `esp_codec_dev` version. But some old bsp still use old `i2c_driver`. If it use old version of `esp_codec_dev` it is OK, if it use new `esp_codec_dev` it will NG. If I provide a Kconfig to let even new `esp_codec_dev` can support old `i2c_driver`, can it fix current dilemma?   
One use old `esp_codec_dev` with the risk new fix can not used, one must have a method like doc to let user to turn on the KConfig. Or esp-idf I2C module provide a Kconfig then esp_codec_dev and esp-bsp can all use it, which is a better idea? Please give some advices, Thanks!

tore-espressif, 2024-08-12:
I can see two solutions
1. you can support both I2C drivers, configurable via menuconfig (and legacy driver as default?). This will require more work, though
2. We can make a 'big leap' and move to driver-NG. But this should be considered a breaking change, so we need to bump the major version to 2.x.x

Both are OK with me :)

closed 2024-10-22 by tore-espressif