espressif/esp-adf #1245: Unable to open the sd card file while transmitting audio data to connected device through bluetooth a2dp

MSalman9133, 2024-08-08:
[image]
[image]
through using example code (D:\ESP_ADF\esp-adf\examples\player\pipeline_a2dp_source_stream)

MSalman9133, 2024-08-08:
audio_element_set_uri(fatfs_stream_reader, "/sdcard/test.mp3"); // i had set the same path still no luck
1. discovery and connection is absolutely fine when comes to transmission itself open the file is an error

MSalman9133, 2024-08-08:
[image]
even mounted uri path also correct,read operation is also correct,when commes to open the sdcard file return -1.

MSalman9133, 2024-08-08:
@projectgus @justintconroy @pedrominatel @kedars ...??

hbler99, 2024-08-09:
Have you set `menuconfig > Component config > FAT Filesystem support > Long filename support` ?

MSalman9133, 2024-08-09:
> Have you set `menuconfig > Component config > FAT Filesystem support > Long filename support` ?

Tqs for u reply hbler99 but still the same no luck
[image]

[image]

hbler99, 2024-08-09:
You can try set `Max long filename length > 512`

MSalman9133, 2024-08-09:
> You can try set `Max long filename length > 512`

Tqs for ur reply actually my indexer isnt working for search can u please explicitly mention in which file and line no this macro is present it willl be beneficail

hbler99, 2024-08-09:
`menuconfig > Component config > FAT Filesystem support > Max long filename length > 512`

MSalman9133, 2024-08-09:
[image]
if   place 512 save and build  the project after that also again reverting back to 255

MSalman9133, 2024-08-09:
even i had tried smaller name (renamed test.mp3 to t.mp3) i tried with that still the same

hbler99, 2024-08-09:
The SD card might be damaged. I recommend formatting the SD card and then trying again.

MSalman9133, 2024-08-09:
[image]
looking fyn ryt

MSalman9133, 2024-08-09:
> The SD card might be damaged. I recommend formatting the SD card and then trying again.

But  i have tried this example  iam using esp32_lyrat_mini_v1.2 board 
D:\ESP_ADF\esp-adf\examples\player\pipeline_sdcard_mp3_control--->this example is working on the same memory card on same board through audio codac iam getting the mp3 songs this is working fyn when comes trying to sending through blutooth iam facing this file read issue

MSalman9133, 2024-08-09:
@hbler99

hbler99, 2024-08-13:
My configuration is as folllows, which doesn't encounter problems.
[image]
[image]

MSalman9133, 2024-08-13:
[button_press.txt](https://github.com/user-attachments/files/16596196/button_press.txt)
able to run the code now as if know hardcoded path able to stream my requirement is till button press it should not start streaming may i know  what is the issue in above code

MSalman9133, 2024-08-13:
All button should be work previous button and next button and volume up and volume down

MSalman9133, 2024-08-13:
> @hbler99

hbler99, 2024-08-15:
What's your development board?  In my way, the button flag is set by input key service
```
ESP_LOGI(TAG, "[ 3 ] Create and start input key service");
    input_key_service_info_t input_key_info[] = INPUT_KEY_DEFAULT_INFO();
    input_key_service_cfg_t input_cfg = INPUT_KEY_SERVICE_DEFAULT_CONFIG();
    input_cfg.handle = set;
    periph_service_handle_t input_ser = input_key_service_create(&input_cfg);
    input_key_service_add_key(input_ser, input_key_info, INPUT_KEY_NUM);
    periph_service_set_callback(input_ser, input_key_service_cb, (void *)board_handle);

```
```
static esp_err_t input_key_service_cb(periph_service_handle_t handle, periph_service_event_t *evt, void *ctx)
{
    /* Handle touch pad events
           to start, pause, resume, finish current song and adjust volume
        */
    audio_board_handle_t board_handle = (audio_board_handle_t) ctx;
    int player_volume;
    audio_hal_get_volume(board_handle->audio_hal, &player_volume);
    if (evt->type == INPUT_KEY_SERVICE_ACTION_CLICK_RELEASE) {
        ESP_LOGI(TAG, "[ * ] input key id is %d", (int)evt->data);
        switch ((int)evt->data) {
            case INPUT_KEY_USER_ID_PLAY:
                ESP_LOGI(TAG, "[ * ] [Play] input key event");
                send_flag = true;
                ESP_LOGE(TAG, "START recording");
                break;
            case INPUT_KEY_USER_ID_SET:
                ESP_LOGI(TAG, "[ * ] [Set] input key event");
                send_flag = false;
                rb_reset(recv_rb);
                ESP_LOGI(TAG, "STOP recording");
                break;
            case INPUT_KEY_USER_ID_VOLUP:
                ESP_LOGI(TAG, "[ * ] [Vol+] input key event");
                player_volume += 10;
                if (player_volume > 100) {
                    player_volume = 100;
                }
                audio_hal_set_volume(board_handle->audio_hal, player_volume);
                ESP_LOGI(TAG, "[ * ] Volume set to %d %%", player_volume);
                break;
            case INPUT_KEY_USER_ID_VOLDOWN:
                ESP_LOGI(TAG, "[ * ] [Vol-] input key event");
                player_volume -= 10;
                if (player_volume < 0) {
                    player_volume = 0;
                }
                audio_hal_set_volume(board_handle->audio_hal, player_volume);
                ESP_LOGI(TAG, "[ * ] Volume set to %d %%", player_volume);
                break;
        }
    }

    return ESP_OK;
}
```
It's referenced in [pipeline_a2dp_sink_stream](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream)