espressif/esp-adf #1247: When the ESP-ADF can  support Continuous speech recognition for continuous dialogue when wakeup by wakenet

iot-lorawan, 2024-08-12:
Hi,
     When the ESP-ADF can support Continuous speech recognition for continuous dialogue when wakeup by wakenet

Thanks,

hbler99, 2024-08-13:
Currently, we have no plans to support continuous speech recognition.

iot-lorawan, 2024-08-15:
Hi @hbler99 ,
      Is there good solution for this.

Thanks,

hbler99, 2024-08-16:
You can try to comment out the line
```
208            detect_flag = 0;
```
https://github.com/espressif/esp-adf/blob/dac74d81c85b5bc4e9da008031f31394943fa2c7/components/audio_recorder/recorder_sr.c#L208C1-L208C29

iot-lorawan, 2024-08-19:
Hi @hbler99 ,
     Thanks for your response.
     I modify as your suggestion, but it seem dont work well. 
     I change use esp-idf with esp-sr， it work fine.

Best Regard.

hbler99, 2024-08-19:
Congrats!

closed 2024-08-19 by iot-lorawan