espressif/esp-adf #1248: pipeline_http_mp3播放卡顿问题，从已关闭议题(AUD-5291) #1174开启

laodi-chen, 2024-08-14:
@jason-mao 老哥

后来又按照你说的方式调整了输出缓冲区大小，并进行了以下调整，但是效果都无济于事。
使用的例程默认播放地址https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3 

1. 使用了更大的请求空间
```c
    http_cfg.out_rb_size = 800 * 1024;
    http_cfg.request_size = 16384;
    http_cfg.request_range_size = 32768;
```
2. 使用更高的任务优先级
```c
    i2s_cfg.task_prio = configMAX_PRIORITIES - 1;
    i2s_cfg.task_core = 1;
```

这是录音

https://github.com/user-attachments/assets/e92508b1-f40f-48b1-a719-18ab5c43ece5

laodi-chen, 2024-08-20:
@jason-mao 老哥来瞧瞧

jason-mao, 2024-08-20:
@laodi-chen 你获取HTTP 下载端的output ringbuffer 水位, 确认一下在卡顿的时候，是不是没有缓冲数据了？听这个声音应该是阶段性的下载不及时。

laodi-chen, 2024-08-21:
@jason-mao 
我有按照你说的方式调整TCP缓冲区大小，并没有直接改善，但发现了另一种行为。
```c
CONFIG_TCP_SND_BUF_DEFAULT=65535
CONFIG_TCP_WND_DEFAULT=65535
CONFIG_TCP_RECVMBOX_SIZE=128
```

在使用云存储时不知道为什么下载速度就会慢，音频听起来比较卡。
按理说云存储有CDN缓存下载应该更快才对的，哪怕云存储只有5M的带宽也完全足够应付这种小文件的下载了。
测试IIS服务器的对外带宽300M，听起来无比丝滑。
这里还衍生了另一个问题，bin文件在云存储时OTA就会经常失败

1. 腾讯云存储`https://test-1323289210.cos.ap-beijing.myqcloud.com/Relax.mp3`
2. 华为云存储`https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/mp3/Relax.mp3`
3. 测试IIS服务器`http://82.156.206.104:8013/Relax.mp3`

jason-mao, 2024-08-21:
@laodi-chen 在卡顿的时候，获取HTTP 的output ringbuffer 水位, 确认一下在卡顿的时候，是不是没有缓冲数据了？

laodi-chen, 2024-08-22:
@jason-mao 
录制了http下载过程的视频，并配有麦克风收录的声音。
我只改了这个部分不知道是不是你说的HTTP 的output ringbuffer 水位。
`ESP_LOGE(TAG, "req lengh=%d, read=%d, pos=%d/%d", len, rlen, (int)info.byte_pos, (int)info.total_bytes);`

我发现http的加载速度远快于https，但听起来还是会有些许断续感。
第二个视频从1分36秒完全下载完成之后就听着好多了。

HTTPS地址录制的视频`https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/mp3/Relax.mp3`

https://github.com/user-attachments/assets/52dec5a8-1a8d-4e62-a71b-26e5159b4b7f

HTTP地址录制的视频`http://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/mp3/Relax.mp3`

https://github.com/user-attachments/assets/b81b48c8-43ef-421f-9ced-df7693780ce5

jason-mao, 2024-08-22:
看着是和下载速度有关，你在 I2S stream timeout 的地方加点打印，https://github.com/espressif/esp-adf/blob/e218638621d421eacb78b79179bde6c1710393b4/components/audio_stream/i2s_stream.c#L286 ，如果数据有下载不及时，它会进入timeout 分支。IDF 5.x 的话要用https://github.com/espressif/esp-adf/blob/e218638621d421eacb78b79179bde6c1710393b4/components/audio_stream/i2s_stream_idf5.c#L521

laodi-chen, 2024-08-23:
@jason-mao 
模组型号：`ESP32-WROVER-E-N8R8`

我注意到例程在IDF5.x运行良好

### IDF4.4.7
IDF版本：`IDF4.4.7` `38eeba213aa695aabfd6d89aa9f5078dbe5a94c3`
ADF版本：`master` `1f275b5018c73c9cd68b6e858006fdd5c52183b5`

1. 当使用默认HTTP配置总会产生错误
```
ESP_LOGI(TAG, "[2.1] Create http stream to read data");
http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
// http_cfg.out_rb_size = 1024 * 1024;
// http_cfg.request_size = 20 * 1024;
// http_cfg.request_range_size = 30 * 1024;
http_stream_reader = http_stream_init(&http_cfg);

I (3130) HTTP_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
E (35060) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
E (35060) HTTP_STREAM: Invalid HTTP stream, status code = -1
E (35060) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,-1
W (35060) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7
W (35080) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
E (35080) MP3_DECODER: Failed to read audio data (line 129)
W (35080) AUDIO_ELEMENT: [mp3] AEL_IO_ABORT, -3
W (35100) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (35140) HTTP_MP3_EXAMPLE: [ * ] Stop event received
```

https://github.com/user-attachments/assets/0d32122c-987b-4385-b0ba-1be9c4640599

2. 使用`if (r_size == AEL_IO_TIMEOUT)`获取的超时打印

https://github.com/user-attachments/assets/8902ebdc-49ed-48c8-b158-af4c716163f0

### IDF5.2.1
IDF版本：`IDF5.2.1` `a322e6bdad4b6675d4597fb2722eea2851ba88cb`
ADF版本：`master` `1f275b5018c73c9cd68b6e858006fdd5c52183b5`

1. idf5.2.1不会存在超时

https://github.com/user-attachments/assets/b5316a80-5c5f-46ff-9468-2ed2e0dd8171

jason-mao, 2024-08-26:
@laodi-chen 4.4 的分支已经EOL，建议切换到5.2 或者5.3 来。

closed 2024-12-10 by jason-mao