espressif/esp-adf #1252: ESP-Lyrat-mini-V1.2 recording microphone silent in example raw_http and voice_recognition/vad

HyouinSchoolAcc, 2024-08-20:
How do I start to debug  
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, start similar to "How do I..". If it is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [ ] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps above, please follow these instructions so we can get the needed information to help you quickly and effectively.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your issues.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other Items If Possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the `Preview` tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: Please follow the above instructions and provide as many details as possible. It will save a lot of communication time and improve the efficiency of problem solving. The more details you provide, the faster we may be able to reproduce and resolve the issue. Thanks!**

----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:     ESP32-LyraT-Mini-1.2
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version v5.2.2
- [Required] ADF version v2.6-159-gdac74d81
- Build system:         [Make|CMake|idf.py]
- [Required] Running log:   
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // 1.22.0-80-g6c4433a
- Operating system:     [Windows|Linux|macOS]
- (Windows only) Environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell]
- Using an IDE?: [No|Yes (please give details)]
- Power supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

// Detailed problem description goes here.

### Expected Behavior
When the board is programmed the microphone is supposed to pick up sound
### Actual Behavior
All sound recorded/inputted are null and the sound audio is completely silent
### Steps to Reproduce

1. get an esp32-lyrat-mini-v1.2
2. upload raw_http example or the voice_recognition/vad example
3. in the raw_http example the recorded voice will be blank. In the vad example there will not be any updates after "vad setup and running"

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue
All code are exactly as written as the example given, with exception of 
```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
[... 66 lines cut ...]
I (4369) REC_RAW_HTTP: [3.2] Create i2s stream to read audio data from codec chip
I (4379) REC_RAW_HTTP: [3.3] Register all elements to audio pipeline
I (4379) REC_RAW_HTTP: [3.4] Link it together [codec_chip]-->i2s_stream->http_stream-->[http_server]
I (4399) REC_RAW_HTTP: Volume set successfully to 100.
I (4399) REC_RAW_HTTP: Current volume is: 100
I (4399) REC_RAW_HTTP: [ 4 ] Press [Rec] button to record, Press [Mode] to exit
W (17339) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:41
E (53339) REC_RAW_HTTP: [ * ] [Rec] input key event, resuming pipeline ...
W (53339) AUDIO_PIPELINE: Without stop, st:1
W (53339) AUDIO_PIPELINE: Without wait stop, st:1
W (53339) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (53349) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (53359) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
Total bytes written: 61440
Total bytes written: 64800
I (55429) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (55439) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
W (55449) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
I (55449) REC_RAW_HTTP: Got HTTP Response = File 20240820T032231Z_16000_16_1.wav was written, size 64800

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

HyouinSchoolAcc, 2024-08-20:
[sdkconfig.json](https://github.com/user-attachments/files/16668521/sdkconfig.json)
[record_raw_http.zip](https://github.com/user-attachments/files/16668625/record_raw_http.zip)
no crashes - coredump not included

jason-mao, 2024-08-21:
@HyouinSchoolAcc Would please test example from latest master branch?

HyouinSchoolAcc, 2024-08-21:
I'm looking at the diagram and it appears that the I2S peripheral is connected to the ES7243 ADC, and I think it might be helpful to probe the ADC to make sure its responding correctly. I added logI''s to i2s_stream.c under i2s_read but it appears it is not being  registered as my logs never show up during recording the system.

HyouinSchoolAcc, 2024-08-22:
Upon further investigation, it appears that this board, the v1.2, needs its ES7243 initialized with I2C register writes in order to receive audio. I read some documentations in ES7243 and is now able to turn on the ADC chip. I still don't know how it interfaces with the codec chip though.

Bhavesh-Moradiya, 2024-08-24:
I am working on same board and having issue with i2s, I want raw data but because of 2 different IC used in no luck

Bhavesh-Moradiya, 2024-08-24:
I checked BSP packages and found ES8311 I2C address is 0x30 and es7243 0x26 but according to schematic it should be 18 and 20 respectively, if any idea about this please let me know what I am missing

HyouinSchoolAcc, 2024-08-26:
I really couldn't figure out how to initiate the codec chip after 3 days of trying, how about you? Were you able to recieve any audio? I havn't tried to force the pipeline to use the ES7243 chip, but that theoretically is easily achievable. How about you? Any plans?

fakefred, 2024-10-03:
I have the same problem, though I've only tried raw_http.

I opened the recorded audio in Ardour (you can also use Audacity), amplified it, then inspected the signal on the spectrogram. It's just white noise all over 0 to 8 kHz.

[image]

I also took a sample of me knocking on the mic with my finger. The peaks do show up, but the gain is super low. I tried setting the volume with `audio_hal_set_volume(board_handle->audio_hal, 100)`, but it didn't help at all.

[image]

> I checked BSP packages and found ES8311 I2C address is 0x30 and es7243 0x26 but according to schematic it should be 18 and 20 respectively, if any idea about this please let me know what I am missing

@Bhavesh-Moradiya, I think 0x26 is correct. According to the [ES7243 datasheet](http://www.everest-semi.com/pdf/ES7243%20PB.pdf), the 7-bit address is 0b00100xx where xx is ~AD1 ~AD0. Per the [LyraT-mini schematic](https://dl.espressif.com/dl/schematics/SCH_ESP32-LyraT-Mini_V1.2_20220119.pdf), the pins AD1 and AD0 are both grounded, so it would be 0b0010011. Add the 0 to the right and we get 0x26.

> I read some documentations in ES7243 and is now able to turn on the ADC chip.

@HyouinSchoolAcc, could you tell us more?

Bhavesh-Moradiya, 2024-10-08:
@HyouinSchoolAcc 

use below code to initialize both codec IC

```
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
    audio_hal_ctrl_codec(board_handle->adc_hal, AUDIO_HAL_CODEC_MODE_ENCODE, AUDIO_HAL_CTRL_START);

```

gggolddeity, 2025-09-03:
> [@HyouinSchoolAcc](https://github.com/HyouinSchoolAcc)
>
> use below code to initialize both codec IC
>
> ```
>     audio_board_handle_t board_handle = audio_board_init(); [...]

It didn't work for me :(
Do you have other solutions to get sound from mic?