espressif/esp-adf #1255: ESP32 Lyra T mini 1.2 ES8311 and Es7243 I2C address and Raw Audio Data

Bhavesh-Moradiya, 2024-08-24:
Hi, according to schematic ES8311 I2C address is 0x18 and ES7243 0x20,
but I installed ESP-ADF and successfully tested SD card record audio example lyra T 1.1, in that example I found ES8311.c i2c Address is 0x30 and es7243 0x26.
How to read raw data from ES7243(Mic Connected) and How to Play Audio from (ES8311 Speaker Connected)? 
why mini 1.2 has 2 ADC-DAC IC?

Thanks

Bhavesh-Moradiya, 2024-08-24:
I also noted that es7243 connected to i2s1 but its SDOUT pin IO36(S_VP) is only for input. so How to get ADC output data?

hbler99, 2024-08-30:
You can try `audio_element_input(audio_element_handle_t el, char *buffer, int wanted_size) ` to obtain audio data from i2s，which is described detailed in [https://github.com/espressif/esp-adf/blob/bed20d72b9bbcd48183ef3dddcccb0af3b264eb1/components/audio_pipeline/audio_element.c#L373。](https://github.com/espressif/esp-adf/blob/bed20d72b9bbcd48183ef3dddcccb0af3b264eb1/components/audio_pipeline/audio_element.c#L373%E3%80%82)
Additionally, there's no need to run the pipeline or element; simply create and initialize the i2s element.

Bhavesh-Moradiya, 2024-09-02:
hi hbler99, Thanks.

is it possible to run this audio_element.c on ESP32 Lyra T 1.2? Do I need to initialize audio board before?

Please help me with more detail w.r.t Lyra T 1.2, I am confused with the ES7243 and ES8311 codec chip connected to mic and speaker respectively on Mini 1.2.

Thanks again

hbler99, 2024-09-03:
Audio board is supposed to be initialized at the beginning. Hope followed steps provide your reference:
```
static audio_element_handle_t create_i2s_stream(int sample_rates, int bits, int channels, audio_stream_type_t type)
{
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA(CODEC_ADC_I2S_PORT, sample_rates, bits, type);
    audio_element_handle_t i2s_stream = i2s_stream_init(&i2s_cfg);
    mem_assert(i2s_stream);
    ....
    return i2s_stream;
}

ESP_LOGI(TAG, "[ x ] Create and start input key service");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);

ESP_LOGI(TAG, "[x] Create i2s stream to write data to codec chip");
audio_element_handle_t i2s_stream_writer = create_i2s_stream(RATE, BITS, CHANNEL, AUDIO_STREAM_WRITER);

ESP_LOGI(TAG, "[x] Create i2s stream to read data from codec chip");
audio_element_handle_t i2s_stream_reader = create_i2s_stream(RATE, BITS, CHANNEL, AUDIO_STREAM_READER);

int read_bytes = audio_element_input(i2s_stream_reader, (void *)block_buffer, size * sizeof(int16_t));

int write_bytes = audio_element_output(i2s_stream_writer, (void *)block_buffer, size * sizeof(int16_t));
```

Bhavesh-Moradiya, 2024-09-03:
Hi @hbler99 Thanks for help

I code as above and received data but its like noise or distortion only, I am not sure about it.
may be something I did wrong in buffer size or i2s Initialization. can you check code and correct me what's wrong I did.

```
static const char *TAG = "I2S RAW DATA";
static esp_periph_set_handle_t set;

#define i2s_ES8311 0
#define i2s_ES7243 1

#define RECORD_RATE 44100
#define RECORD_CHANNEL 1
#define RECORD_BITS 16

#define PLAYBACK_RATE 44100
#define PLAYBACK_CHANNEL 1
#define PLAYBACK_BITS 16

#define EXAMPLE_BUFF_SIZE 2048

static audio_element_handle_t create_i2s_stream(int PORT, int sample_rates, int bits, int channels, audio_stream_type_t type)
{
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = type;
    i2s_cfg.chan_cfg.id = PORT;
    i2s_cfg.transmit_mode = I2S_COMM_MODE_STD;
    i2s_cfg.std_cfg.clk_cfg.sample_rate_hz = sample_rates;
    i2s_cfg.std_cfg.slot_cfg.data_bit_width = bits;
    i2s_cfg.std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_MONO;
    i2s_cfg.volume = 90;
    i2s_stream_set_channel_type(&i2s_cfg, I2S_CHANNEL_TYPE_ONLY_RIGHT);
    audio_element_handle_t i2s_stream = i2s_stream_init(&i2s_cfg);
    mem_assert(i2s_stream);

    audio_element_set_music_info(i2s_stream, sample_rates, channels, bits);
    return i2s_stream;
}

static audio_element_handle_t create_wav_encoder()
{
    wav_encoder_cfg_t wav_cfg = DEFAULT_WAV_ENCODER_CONFIG();
    return wav_encoder_init(&wav_cfg);
}

static audio_element_handle_t create_wav_decoder()
{
    wav_decoder_cfg_t wav_cfg = DEFAULT_WAV_DECODER_CONFIG();
    return wav_decoder_init(&wav_cfg);
}

void app_main(void)
{

    uint16_t *w_buf = (uint16_t *)calloc(1, EXAMPLE_BUFF_SIZE);
    assert(w_buf);

    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    set = esp_periph_set_init(&periph_cfg);
    audio_board_key_init(set);

    ESP_LOGI(TAG, "[ x ] Create and start input key service");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
    audio_hal_ctrl_codec(board_handle->adc_hal, AUDIO_HAL_CODEC_MODE_ENCODE, AUDIO_HAL_CTRL_START);

    ESP_LOGI(TAG, "[x] Create i2s stream to write data to codec chip");
    audio_element_handle_t i2s_reader_el = create_i2s_stream(i2s_ES7243, RECORD_RATE, RECORD_BITS, RECORD_CHANNEL, AUDIO_STREAM_READER);
    audio_element_handle_t wav_encoder_el = create_wav_encoder();
[... 128 lines cut ...]
Data: 473
Data: 741
Data: 700
Data: 803
Data: 759
Data: 837
Data: 65253
Data: 65387
Data: 65087
Data: 65155
Data: 65016
Data: 65049
Data: 64872
Data: 64959
Data: 64720
Data: 64787
Data: 64615
Data: 64675
Data: 64368
Data: 64502
Data: 64359
Data: 64319
```

thanks

hbler99, 2024-09-04:
I hope the following code will help you.
```
static const char *TAG = "I2S RAW DATA";
static esp_periph_set_handle_t set;

#define RECORD_RATE 44100
#define RECORD_CHANNEL 1
#define RECORD_BITS 16

#define PLAYBACK_RATE 44100
#define PLAYBACK_CHANNEL 1
#define PLAYBACK_BITS 16

#define EXAMPLE_BUFF_SIZE 512

static audio_element_handle_t create_i2s_stream(int sample_rates, int bits, int channels, audio_stream_type_t type)
{
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA(CODEC_ADC_I2S_PORT, sample_rates, bits, type);
    i2s_cfg.std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_MONO;
    i2s_cfg.std_cfg.slot_cfg.slot_mask = I2S_STD_SLOT_LEFT;
    audio_element_handle_t i2s_stream = i2s_stream_init(&i2s_cfg);
    mem_assert(i2s_stream);
    audio_element_set_music_info(i2s_stream, sample_rates, channels, bits);
    return i2s_stream;
}

static audio_element_handle_t create_wav_encoder()
{
    wav_encoder_cfg_t wav_cfg = DEFAULT_WAV_ENCODER_CONFIG();
    return wav_encoder_init(&wav_cfg);
}

static audio_element_handle_t create_wav_decoder()
{
    wav_decoder_cfg_t wav_cfg = DEFAULT_WAV_DECODER_CONFIG();
    return wav_decoder_init(&wav_cfg);
}

void app_main(void)
{

    uint16_t *w_buf = (uint16_t *)audio_calloc(1, EXAMPLE_BUFF_SIZE * sizeof(uint16_t));

    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    set = esp_periph_set_init(&periph_cfg);
    audio_board_key_init(set);

    ESP_LOGI(TAG, "[ x ] Create and start input key service");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
    audio_hal_ctrl_codec(board_handle->adc_hal, AUDIO_HAL_CODEC_MODE_ENCODE, AUDIO_HAL_CTRL_START);

    ESP_LOGI(TAG, "[x] Create i2s stream to write data to codec chip");
    audio_element_handle_t i2s_reader_el = create_i2s_stream(RECORD_RATE, RECORD_BITS, RECORD_CHANNEL, AUDIO_STREAM_READER);
    audio_element_handle_t wav_encoder_el = create_wav_encoder();

    ESP_LOGI(TAG, "[x] Create i2s stream to read data from codec chip");
    audio_element_handle_t i2s_writer_el = create_i2s_stream(PLAYBACK_RATE, PLAYBACK_BITS, PLAYBACK_CHANNEL, AUDIO_STREAM_WRITER);
    audio_element_handle_t wav_decoder_el = create_wav_decoder();

    ESP_LOGI(TAG, "[ 3 ] Set up  event listener");
    audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
    audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);

    ESP_LOGI(TAG, "[ 4 ] Set event listening");
    audio_event_iface_set_listener(esp_periph_set_get_event_iface(set), evt);
    ESP_LOGI(TAG, "[ 5 ] Read i2s Data");
    i2s_stream_set_clk(i2s_reader_el, RECORD_RATE, RECORD_BITS, RECORD_CHANNEL);
    while (1)
    {
        int read_bytes = audio_element_input(i2s_reader_el, (void *)w_buf, EXAMPLE_BUFF_SIZE * sizeof(uint16_t));
        int write_bytes = audio_element_output(i2s_writer_el, (void *)w_buf, EXAMPLE_BUFF_SIZE * sizeof(uint16_t));
    }
}
```
Specially, `i2s_stream_set_clk` is supposed to be placed outside of the `while(1)` loop.

Bhavesh-Moradiya, 2024-09-28:
@hbler99 

Thanks for help brother, its working good.

there is some sound distorted and noisy that need to fix, will check it through "Pipeline" method to get encoded data or resampling.

In code create_i2s_stream function must be initialize separately for both read and write method like below.

 ```
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = type;
    i2s_cfg.chan_cfg.id = port;
    i2s_cfg.std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_MONO;
    i2s_stream_set_channel_type(&i2s_cfg, I2S_CHANNEL_TYPE_ONLY_RIGHT);
    audio_element_handle_t i2s_reader_el = i2s_stream_init(&i2s_cfg);
    mem_assert(i2s_reader_el);
    audio_element_set_music_info(i2s_reader_el, 44100, 16, 1);
    i2s_stream_set_clk(i2s_reader_el, 44100, 16, 1);
    return i2s_reader_el;
```

I achieved my task with I2S > TCP and TCP > I2S using pipeline code and now will do same with new code.

Thanks again for help and support

closed 2024-09-28 by Bhavesh-Moradiya