espressif/esp-adf #1259: esp-adf build error

skylin008, 2024-08-30:
xllin@xllindeMacBook-Pro cli % idf.py set-target esp32
/Users/xllin/software_develop/esp-adf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/Users/xllin/software_develop/esp-adf/examples/cli/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
xllin@xllindeMacBook-Pro cli % idf.py set-target esp32
/Users/xllin/software_develop/esp-adf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/Users/xllin/software_develop/esp-adf/examples/cli/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
xllin@xllindeMacBook-Pro cli % idf.py fullclean       
/Users/xllin/software_develop/esp-adf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: fullclean
Directory '/Users/xllin/software_develop/esp-adf/examples/cli/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
xllin@xllindeMacBook-Pro cli % idf.py clean    
/Users/xllin/software_develop/esp-adf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: clean
Running cmake in directory /Users/xllin/software_develop/esp-adf/examples/cli/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/xllin/software_develop/esp-adf/examples/cli"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-145)") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/xllin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/xllin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/xllin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/jsmn
NOTICE: Skipping optional dependency: espressif/nghttp
Processing 2 dependencies:
[1/2] espressif/esp-dsp (1.4.12)
[2/2] idf (4.4.4)
NOTICE: Skipping optional dependency: espressif/jsmn
NOTICE: Skipping optional dependency: espressif/nghttp
-- Project sdkconfig file /Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig
/Users/xllin/software_develop/esp-adf/components/esp_codec_dev/Kconfig:4: warning: ESP_IDF_VERSION has 'option env="ESP_IDF_VERSION"', but the environment variable ESP_IDF_VERSION is not set
Loading defaults file /Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig.defaults...
/Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig.defaults:27 CONFIG_MAIN_TASK_STACK_SIZE was replaced with CONFIG_ESP_MAIN_TASK_STACK_SIZE
Loading defaults file /Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig.defaults.esp32...
Traceback (most recent call last):
  File "/Users/xllin/software_develop/esp-adf/esp-idf/tools/kconfig_new/confgen.py", line 644, in <module>
    main()
  File "/Users/xllin/software_develop/esp-adf/esp-idf/tools/kconfig_new/confgen.py", line 330, in main
    output_function(deprecated_options, config, temp_file)
  File "/Users/xllin/software_develop/esp-adf/esp-idf/tools/kconfig_new/confgen.py", line 407, in write_header
    deprecated_options.append_header(config, filename)
  File "/Users/xllin/software_develop/esp-adf/esp-idf/tools/kconfig_new/confgen.py", line 186, in append_header
    if new_opt in config.syms and _opt_defined(config.syms[new_opt]):
  File "/Users/xllin/software_develop/esp-adf/esp-idf/tools/kconfig_new/confgen.py", line 173, in _opt_defined
    if opt.orig_type in (kconfiglib.BOOL, kconfiglib.TRISTATE) and opt.str_value != 'n':
AttributeError: module 'kconfiglib' has no attribute 'BOOL'
CMake Error at /Users/xllin/software_develop/esp-adf/esp-idf/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py
  (/Users/xllin/.espressif/python_env/idf4.4_py3.10_env/bin/python;/Users/xllin/software_develop/esp-adf/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/Users/xllin/software_develop/esp-adf/esp-idf/Kconfig;--sdkconfig-rename;/Users/xllin/software_develop/esp-adf/esp-idf/sdkconfig.rename;--config;/Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig;--defaults;/Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig.defaults;--defaults;/Users/xllin/software_develop/esp-adf/examples/cli/sdkconfig.defaults.esp32;--env-file;/Users/xllin/software_develop/esp-adf/examples/cli/build/config.env).
  Error 1
Call Stack (most recent call first):
  /Users/xllin/software_develop/esp-adf/esp-idf/tools/cmake/build.cmake:534 (__kconfig_generate_config)
  /Users/xllin/software_develop/esp-adf/esp-idf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1

How to solve this issue.Thanks!

hbler99, 2024-08-30:
First, manually delete the build folder, and then `cd $ADF_PATH$->./install->. ./export.sh`. After that you can try the steps mentioned above again.

skylin008, 2024-09-03:
Thanks @hbler99 .Yes, I follow the user guide to build.  I had change the esp-idf branch to V5.3 , now the other issue shows:

[root@VM-16-3-opencloudos play_mp3_control]# idf.py set-targets esp32p4
Executing action: set-targets
Running cmake in directory /home/audioDevelop/esp-adf/examples/get-started/play_mp3_control/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/root/.espressif/python_env/idf5.3_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/audioDevelop/esp-adf/examples/get-started/play_mp3_control"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.41.0") 
-- Component directory /home/audioDevelop/esp-adf/components/esp-adf-libs does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/audioDevelop/esp-adf/components/esp-sr does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Updating lock file at /home/audioDevelop/esp-adf/examples/get-started/play_mp3_control/dependencies.lock
Processing 2 dependencies:
[1/2] espressif/jsmn (1.1.0)
[2/2] idf (5.3.0)
CMake Error at /home/audioDevelop/esp-adf/esp-idf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'esp-adf-libs'.
Call Stack (most recent call first):
  /home/audioDevelop/esp-adf/esp-idf/tools/cmake/build.cmake:304 (__build_resolve_and_add_req)
  /home/audioDevelop/esp-adf/esp-idf/tools/cmake/build.cmake:305 (__build_expand_requirements)
  /home/audioDevelop/esp-adf/esp-idf/tools/cmake/build.cmake:311 (__build_expand_requirements)
  /home/audioDevelop/esp-adf/esp-idf/tools/cmake/build.cmake:607 (__build_expand_requirements)
  /home/audioDevelop/esp-adf/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/audioDevelop/esp-adf/examples/get-started/play_mp3_control/build/log/idf_py_stderr_output_44775 and /home/audioDevelop/esp-adf/examples/get-started/play_mp3_control/build/log/idf_py_stdout_output_44775

hbler99, 2024-09-04:
Maybe updating the submodule to latest repository could solve it.

closed 2024-12-10 by jason-mao