espressif/esp-adf #1261: dureos compilation errors and questions

AaronPMM, 2024-09-01:
Environment：
Audio development kit: ESP32-S3-BOX-3
IDF version:5,3 (windows)
ADF version:v2.6-167-ge27cd526

Error description：
As shown in the figure below：

[image]

It seems that duer_profile does not have permission.The document states that the requested profile file should be placed in ADF_PATH/components/dueros_service/duer_profile. In the original SDK, duer_profile is a file instead of a folder, so I deleted it and created a duer_profile folder and placed the profile in it.

I don't know if this is the reason, if so, what kind of permissions are required?

In addition, the document states that tone/dingding.wav in the example needs to be copied to microSD, but there is no place to insert the SDK on ESP32-S3-BOX-3. How can I solve this problem?

JosephTang, 2024-09-02:
Hi ~ @AaronPMM, 
duer_profile should be a file. When got your profile from baidu, copy the profile's content into the `duer_profile`, after that, select ` Generate the nvs partition which include the profile of dueros` in `menuconfig`, a python script will be run to generate a bin, you can also select `flash the generated nvs partition` to flash the bin with the `idf.py flash` command.

No additional permissions required for this file.

esp-box-3 has no sdcard slot. so you can make the tone/dingding.wav as a `flash_tone`. 
A tool [mk_audio_tong.py](https://github.com/espressif/esp-adf/blob/master/tools/audio_tone/mk_audio_tone.py) shows how to make a `flash_tone` binary.
A example [wwe](https://github.com/espressif/esp-adf/blob/master/examples/speech_recognition/wwe) and its [CMakeLists.txt](https://github.com/espressif/esp-adf/blob/master/examples/speech_recognition/wwe/CMakeLists.txt#L34) show how to flash and use this bin.

AaronPMM, 2024-09-02:
> Hi ~ @AaronPMM, duer_profile should be a file. When got your profile from baidu, copy the profile's content into the `duer_profile`, after that, select ` Generate the nvs partition which include the [...]

The duer_profile file I downloaded is a zip package. According to the description in the document (as shown in the figure below), duer_profile should be a folder instead of a file.

JosephTang, 2024-09-02:
The DuerOS example was updated two months ago in the master branch, and this document has also been changed. We will clarify this description in a future version. Thank you for your suggestion. You can replace the content in dueros_profile and rebuild the project.

AaronPMM, 2024-09-02:
> The DuerOS example was updated two months ago in the master branch, and this document has also been changed. We will clarify this description in a future version. Thank you for your suggestion. You can replace the content in dueros_profile and rebuild the project.

Current environment:
I have now found a physical machine with Ubuntu 20.04 and changed the previous window environment to Ubuntu environment

Do you mean that duer_profile is a folder instead of a file? I deleted the duer_profile file, created a duer_profile folder, unzipped the profile and copied the content files to the duer_profile folder. The compilation prompted an error again. It was an open error. So duer_profile is a file, and the document does not describe the application profile in detail. So what should duer_profile be? How to apply for it?
Also, there should be sd initialization in the example, but esp32-s3-box-3 hardware does not have sd. Even if the compilation is successful, the program will not run normally due to sd initialization. Do you have any suggestions?

JosephTang, 2024-09-02:
`duer_profile` is a file. Just copy your profile's content into it.

AaronPMM, 2024-09-02:
> `duer_profile` is a file. Just copy your profile's content into it.

I guess it may be that I rename any file in the compression package to duer_profile. I am trying it, but now after the program is downloaded to the hardware, it has not been working properly. The error is as follows:
I (137356) SDCARD: Using 1-line SD mode, base path=/sdcard
I (137366) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (137366) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (137376) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 E (137426) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107 E (137426) vfs_fat_sdmmc: sdmmc_card_init failed (0x107). in place. W (137436) AUDIO_EVT: There is no space in external queue E (137446) PERIPH_SDCARD: mount sdcard error! I (138356) SDCARD: Using 1-line SD mode, base path=/sdcard I (138356) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 I (138356) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 I (138366) gpio: GPIO[9]| down: 0| Intr:0 E (138436) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107 E (138436) vfs_fat_sdmmc: sdmmc_card_init failed (0x107). E (138436) SDCARD: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place. W (138446) AUDIO_EVT: There is no space in external queue E (138446) PERIPH_SDCARD: mount sdcard error!

shootao, 2024-09-05:
same as [issues/1264](https://github.com/espressif/esp-adf/issues/1264), plaese close this Issue

closed 2024-09-07 by AaronPMM