espressif/esp-adf #1264: Question about using audio flash tone in Dueros routine

AaronPMM, 2024-09-03:
The hardware I have is ESP32-S3-BOX-3. Although ESP-ADF indicates that it supports this hardware, the dueros example does not support it. The example uses sdcard, but ESP32-S3-BOX-3 does not support sdcard. I hope to replace sdcard with aduio flash tone. Since this is my first time using ESP chips, I have a problem with how to put the generated bin file into flash.
I searched for some online articles, which pointed out that the partition table needs to be modified to add a flash_tone partition. There are 2 ".csv" files in this example, partitions_dueros_examples.csv and partitions_dueros_examples_esp32s3.csv. Which one should I modify? Or should I modify both? In addition, what should I modify for the offset and size options?

hbler99, 2024-09-03:
In the future, we will support it. Please be patient.

AaronPMM, 2024-09-03:
> In the future, we will support it. Please be patient.

Can you give me a reply or suggestion on my question? Our company is currently evaluating this project and plans to quickly build a sample based on this hardware. We may not be able to wait for you to complete the routine.

AaronPMM, 2024-09-03:
I made the following changes according to the routine 

1. Add audio_tone_uri.c and audio_tone_uir.h in the main directory 

2. 2. Modify the CMakeList.txt file in the dueros directory

 ```
      # This is a cmake function, which is used to flash the bin file to the specified partition
      function(esptool_py_flash_customize_image target_name image_name offset image)
          idf_build_get_property(build_dir BUILD_DIR)
          file(RELATIVE_PATH image ${build_dir} ${image})
      
          set_property(TARGET ${target_name} APPEND PROPERTY FLASH_FILE
                      "\"${offset}\" : \"${image}\"")
          set_property(TARGET ${target_name} APPEND PROPERTY FLASH_ENTRY
                      "\"${image_name}\" : { \"offset\" : \"${offset}\", \"file\" : \"${image}\" }")
          set_property(TARGET ${target_name} APPEND PROPERTY IMAGES "${offset} ${image}")
      
          if(CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT)
              set_property(TARGET encrypted-${target_name} APPEND PROPERTY FLASH_FILE
                          "\"${offset}\" : \"${image}\"")
              set_property(TARGET encrypted-${target_name} APPEND PROPERTY FLASH_ENTRY
                          "\"${image_name}\" : { \"offset\" : \"${offset}\", \"file\" : \"${image}\" }")
              set_property(TARGET encrypted-${target_name} APPEND PROPERTY IMAGES "${offset} ${image}")
          endif()
      endfunction()
      
      # Flash the custom partition named `flash_tone`.
      set(partition flash_tone)
      idf_build_get_property(project_dir PROJECT_DIR)
      message(STATUS "project_dir = >>>>>>>>>>>>>>>>>>>>>>>>>>>> %s", ${project_dir})
      set(image_file ${project_dir}/tone/audio_tone.bin)
      partition_table_get_partition_info(offset "--partition-name ${partition}" "offset")
      esptool_py_flash_customize_image(flash "${partition}" "${offset}" "${image_file}")
```

   3. Modify the void *duer_audio_start_recorder(rec_event_cb_tcb) function

```
      void *duer_audio_start_recorder(rec_event_cb_t cb)
      {
          if (recorder) {
              return recorder;
          }
      
          audio_element_handle_t i2s_stream_reader;
          audio_pipeline_handle_t pipeline;
          audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
          pipeline = audio_pipeline_init(&pipeline_cfg);
          if (NULL == pipeline) {
              return NULL;
          }
      
          i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA(CODEC_ADC_I2S_PORT, 48000, CODEC_ADC_BITS_PER_SAMPLE, AUDIO_STREAM_READER);
          i2s_cfg.task_core = 1;
          i2s_stream_reader = i2s_stream_init(&i2s_cfg);
      
          raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
          raw_cfg.type = AUDIO_STREAM_READER;
          raw_read = raw_stream_init(&raw_cfg);
      
          audio_pipeline_register(pipeline, i2s_stream_reader, "i2s");
          audio_pipeline_register(pipeline, raw_read, "raw");
      #if 0   
          const char *link_tag[3] = {"i2s", "raw"};
          uint8_t linked_num = 2;
      
          rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
          rsp_cfg.src_rate = 48000;
          rsp_cfg.dest_rate = 16000;
[... 270 lines cut ...]
I (1334) wifi:Init tx cache buffer num: 32
I (1344) wifi:Init static tx FG buffer num: 2
I (1344) wifi:Init static rx buffer size: 1600
I (1354) wifi:Init static rx buffer num: 16
I (1354) wifi:Init dynamic rx buffer num: 32
I (1354) wifi_init: rx ba win: 16
I (1364) wifi_init: tcpip mbox: 32
I (1364) wifi_init: udp mbox: 6
I (1374) wifi_init: tcp mbox: 6
I (1374) wifi_init: tcp tx win: 5744
I (1374) wifi_init: tcp rx win: 5744
I (1384) wifi_init: tcp mss: 1440
I (1384) wifi_init: WiFi/LWIP prefer SPIRAM
I (1394) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (1434) wifi:mode : sta (74:4d:bd:2e:86:bc)
I (1434) wifi:enable tsf
I (1110,tid:3fcbdb70) lightduer_session.c(  44): random = 75933
W (11274) AUDIO_BOARD: The board has already been initialized!
W (11274) WIFI_SSID_MANAGER: There is no ssid stored in flash, please save ssids to flash first
W (11274) WIFI_SERV: No ssid stored in flash, try to connect to wifi set by wifi_service_set_sta_info()
I (11284) WIFI_SERV: Connect to wifi ssid: TWGJ-Wireless-XNY, pwd: XnY!5716
I (11294) AUDIO_THREAD: The media_task task allocate stack on external memory
I (11304) ESP_AUDIO_TASK: media_ctrl_task running...,0x3d015ed0

Is my modification correct? Is there anything missing? Why do they keep rebooting?

shootao, 2024-09-04:
@AaronPMM 
Please apply the attached patch:
1. Support ESP-Box wake-up.
2. Support playing prompt sounds without an SD card.
[dueros_support_box_without_sdcard.zip](https://github.com/user-attachments/files/16862764/dueros_support_box_without_sdcard.zip)

closed 2024-09-07 by AaronPMM