espressif/esp-adf #1267: Compilation problem: bsp/esp-bsp.h:No such file or directory

AaronPMM, 2024-09-05:
I am trying to use an LCD screen in a Dueros example, referring to the esp-box chatgpt_demo, and calling bsp_i2c_init(). When compiling, the error "No such file or directory" appears. Which library is the bsp_i2c_init function in? How do I configure it to use it properly?

shootao, 2024-09-05:
基于最新的ADF 你可以有如下操作
1. 将 [esp-box-3 BSP](https://github.com/espressif/esp-bsp/tree/master/bsp/esp-box-3) 放到你的 components 下面
2. 参考 chatgpt demo, 初始化 lcd
3. 获取 bsp 里面 i2c 的handle， 在codec 初始化前调用  `i2c_bus_set_master_handle`
以上操作应该就可以了

AaronPMM, 2024-09-06:
> 基于最新的ADF 你可以有如下操作
> 
> 1. 将 [esp-box-3 BSP](https://github.com/espressif/esp-bsp/tree/master/bsp/esp-box-3) 放到你的 components 下面
> 2. 参考 chatgpt demo, 初始化 lcd
> 3. 获取 bsp 里面 i2c 的handle， 在codec 初始化前调用  `i2c_bus_set_master_handle`
>    以上操作应该就可以了

替换后提示 ESP-BOX-3 的I2C驱动是旧的（i2c: CONFLICT! driver_ng is not allowed to be used with this old driver），我更换为 driver/i2c_master.h，编译出错，新旧驱动的接口函数不同，请问新的 i2c 驱动的例子应该参考什么？

closed 2024-09-09 by AaronPMM