espressif/esp-adf #1271: ESP32-S3-Korvo-2 V3.0 when running VOIP example uses wrong contact, and the microphones do not pick up any audio

Nathanael29, 2024-09-09:
## Environment

- Audio development kit:      [ESP32-S3-Korvo-2]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v5.1.4
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // master
- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Operating system:     [Windows]
- (Windows only) Environment type: [ESP Command Prompt]
- Power supply:         [USB]

## Problem Description

Running the VOIP example on the ESP32-S3-Korvo-2 V3.0, the device does not use the contact in the same network as the SIP server.
On the rare occasion that it does connect to the SIP server, the microphones do not pick up any audio.

SIP Server is miniSIPserver on Raspberry Pi: https://www.myvoipapp.com/

## Debug Logs

Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18e4
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x308c
entry 0x403c9934
I (27) boot: ESP-IDF v5.1.4-dirty 2nd stage bootloader
I (27) boot: compile time Sep  4 2024 16:14:48
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00200000
I (92) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c150020 size=51c04h (334852) map
I (163) esp_image: segment 1: paddr=00061c2c vaddr=3fc97500 size=08e7ch ( 36476) load
I (170) esp_image: segment 2: paddr=0006aab0 vaddr=40374000 size=05568h ( 21864) load
I (175) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=148ae8h (1346280) map
I (382) esp_image: segment 4: paddr=001b8b10 vaddr=40379568 size=0dec0h ( 57024) load
I (402) boot: Loaded app from partition at offset 0x10000
I (402) boot: Disabling RNG early entropy source...
I (414) cpu_start: Multicore app
I (414) octal_psram: vendor id    : 0x0d (AP)
I (414) octal_psram: dev id       : 0x02 (generation 3)
I (417) octal_psram: density      : 0x03 (64 Mbit)
I (423) octal_psram: good-die     : 0x01 (Pass)
I (428) octal_psram: Latency      : 0x01 (Fixed)
I (433) octal_psram: VCC          : 0x01 (3V)
I (438) octal_psram: SRF          : 0x01 (Fast Refresh)
I (444) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (450) octal_psram: BurstLen     : 0x01 (32 Byte)
I (456) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (462) octal_psram: DriveStrength: 0x00 (1/1)
I (468) MSPI Timing: PSRAM timing tuning index: 5
I (472) esp_psram: Found 8MB PSRAM device
I (477) esp_psram: Speed: 80MHz
I (587) mmu_psram: Instructions copied and mapped to SPIRAM
[... 150 lines cut ...]
REGISTER sip:192.168.68.100:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.82.212:15092;branch=z9hG4bK--655567353;rport
From: <sip:100@192.168.68.100:5060>;tag=1741089145
To: <sip:100@192.168.68.100:5060>
Contact: <sip:100@192.168.82.212:15092;transport=UDP>
Max-Forwards: 70
Call-ID: 5D3036E376EFF037060383577E5329A0AAEB84A01977
CSeq: 1 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml

I (6821) SIP: [1970-01-01/00:00:02]=======================>>
W (9829) SIP: Retransmit again
I (9829) SIP: [1970-01-01/00:00:04]=======WRITE 0598 bytes>>
I (9829) SIP:

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jason-mao, 2024-09-12:
@Nathanael29 I think the ADF latest master has fixed this issue. It's similar to the PR https://github.com/espressif/esp-adf/pull/1265/files

Nathanael29, 2024-09-13:
Just tested with ADF master, the example is working but the microphone gain is not that high. Where can I adjust the gain?