espressif/esp-adf #1275: VOIP例程开启NSNET后运行报错E (42821) MODEL_LOADER: Can not find model in partition table

52blingbling, 2024-09-14:
- Audio development kit:     ESP32-S3-Korvo-2
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): v3
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v5.2
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // master

运行VoIP例程后，开启NS在不讲话时，仍能从手机扬声器端听到些许微弱的杂音（环境噪音不算大，就正常微弱环境音），算法上能不能做到不讲话时连杂音都没有？我想开启NSNET试试效果，结果运行后报错
E (42821) MODEL_LOADER: Can not find model in partition table
E (42828) ALGORITHM_STREAM: Failed to load models
E (42833) ALGORITHM_STREAM: algorithm_stream.c:388 (algo_stream_init): Got NULL Pointer
E (42842) ALGORITHM_STREAM: Error occured

52blingbling, 2024-09-14:
我发现是partitions_voip_example.csv里没有
model | data | spiffs |   | 5168K这一项，但添加后会报错，I (31249) MODEL_LOADER: The storage free size is 3776 KB
I (31255) MODEL_LOADER: The partition size is 5168KB
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x42071dfe
0x42071dfe: srmodel_mmap_init at E:/esp/esp-adf/components/esp-sr/src/model_path.c:349 (discriminator 1)

hbler99, 2024-10-14:
`I (31249) MODEL_LOADER: The storage free size is 3776 KB
I (31255) MODEL_LOADER: The partition size is 5168KB` 这段说明你size设定的太大了，剩余内存只有3776 KB但你设定了5168KB，可以改小点再试一下。

closed 2025-02-17 by jason-mao