espressif/esp-adf #1276: rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

masm31, 2024-09-14:
I am trying to run the pipeline_tcp_client from ESP-ADF on my custom board, which uses an ESP32-S chip. However, I keep encountering a debug message, and my ESP32 keeps restarting in a loop. Could you help me identify the problem and suggest how to prevent these continuous restarts?

part of my terminal:
.
I (56) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (56) boot: compile time Sep 11 2024 18:24:22
I (57) boot: Multicore bootloader
I (62) boot: chip revision: v1.0
I (66) boot.esp32: SPI Speed      : 80MHz
I (70) boot.esp32: SPI Mode       : DIO
I (75) boot.esp32: SPI Flash Size : 4MB
I (79) boot: Enabling RNG early entropy source...
I (85) boot: Partition Table:
I (88) boot: ## Label            Usage          Type ST Offset   Length
I (96) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (103) boot:  1 phy_init         RF data          01 01 0000d000 00003000
I (111) boot:  2 factory          factory app      00 00 00010000 00300000
I (118) boot: End of partition table
I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2cbb8h (183224) map
I (183) esp_image: segment 1: paddr=0003cbe0 vaddr=3ffb0000 size=03438h ( 13368) load
I (187) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=b0c0ch (723980) map
I (393) esp_image: segment 3: paddr=000f0c34 vaddr=3ffb3438 size=00b3ch (  2876) load
I (395) esp_image: segment 4: paddr=000f1778 vaddr=40080000 size=18630h ( 99888) load
I (444) boot: Loaded app from partition at offset 0x10000
I (445) boot: Disabling RNG early entropy source...
I (456) cpu_start: Multicore app
I (465) cpu_start: Pro cpu start user code
I (465) cpu_start: cpu freq: 160000000 Hz
I (465) app_init: Application information:
I (468) app_init: Project name:     tcp_client_example
I (473) app_init: App version:      1
I (478) app_init: Compile time:     Sep 11 2024 18:24:12
I (484) app_init: ELF file SHA256:  192ad9a86...
I (489) app_init: ESP-IDF:          v5.3-dirty
I (494) efuse_init: Min chip rev:     v0.0
I (499) efuse_init: Max chip rev:     v3.99 
I (504) efuse_init: Chip rev:         v1.0
I (509) heap_init: Initializing. RAM available for dynamic allocation:
I (516) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (522) heap_init: At 3FFB8AB8 len 00027548 (157 KiB): DRAM
I (528) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (534) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (541) heap_init: At 40098630 len 000079D0 (30 KiB): IRAM
I (548) spi_flash: detected chip: generic
I (552) spi_flash: flash io: dio
W (556) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (565) main_task: Started on CPU0
I (575) main_task: Calling app_main()
I (575) TCP_CLIENT_EXAMPLE: [ 1 ] Start codec chip
I (575) TCP_CLIENT_EXAMPLE: [2.0] Create audio pipeline for playback
I (585) TCP_CLIENT_EXAMPLE: [2.1] Create i2s stream to write data to codec chip
I (595) TCP_CLIENT_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file
I (595) TCP_CLIENT_EXAMPLE: [2.2] Create tcp client stream to read data
I (605) TCP_CLIENT_EXAMPLE: [2.3] Register all elements to audio pipeline
I (615) TCP_CLIENT_EXAMPLE: [2.4] Link it together tcp-->mp3-->i2s
I (615) TCP_CLIENT_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
W (635) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7176
load:0x40078000,len:15576
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640

hbler99, 2024-09-19:
Have you configured it correctly according to the description, such as `Audio board`?

jason-mao, 2024-12-10:
@masm31 Have you erased the flash and tested it again?

closed 2025-02-17 by jason-mao