espressif/esp-adf #1278: I2C bus driver does not obey menuconfig setting

kscz, 2024-09-16:
The menu setting to force use of the legacy i2c driver is ignored by driver/i2c_bus

[image]

This area appears to be the issue: https://github.com/espressif/esp-adf/blob/master/components/esp_peripherals/CMakeLists.txt#L33-L37

This causes applications which require the legacy i2c driver get the error message "CONFLICT! driver_ng is not allowed to be used with this old driver"

hbler99, 2024-10-16:
You can refer to [solved](https://github.com/espressif/esp-adf/issues/1274#issuecomment-2398746866) to solve this problem.

kscz, 2024-10-27:
I can't read Chinese - could you provide some details of what that issue contains?

hbler99, 2024-10-29:
> I can't read Chinese - could you provide some details of what that issue contains?

The root cause of this problem is that ADF has supported the latest i2c driver of IDF, but the esp32-camera on which this example depends is still using the old i2c driver. The current solution is to set this [line](https://github.com/espressif/esp-adf/blob/3b6e59bd88e7eda1c8e3a9904db0cc32fa124b48/components/esp_peripherals/CMakeLists.txt#L33) to "true" first, and ADF will also fix this issue later.

jason-mao, 2024-12-10:
This topic has become inactive, so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2024-12-10 by jason-mao