espressif/esp-adf #1280: unable to get sample "pipeline_a2dp_sink_and_hfp" to work on ESP32-LyraT-Mini

malto101, 2024-09-20:
## Environment

- Audio development kit:      ESP32-LyraT-Mini 
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): v1
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v5.3
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.7-4-g1bc914b7
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
```

```
--- Using /dev/cu.usbserial-2140 instead...
--- esp-idf-monitor 1.5.0 on /dev/cu.usbserial-2140 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
`�x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx����x���xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx��������xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx��������xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx��������xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx������xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx����x��xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx������xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx������xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx����x��xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx��������xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx��������xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx��������xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx�������xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx����x���xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx����x���xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx��������xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx��������xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx������xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx����x��xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx����x��xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx����xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx��x��xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx������xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx������xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx������xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx�����xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx�����xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx��x���xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx������xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx������xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx����xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx����xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx��x��xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx�x���xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx�x�x��xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx�x�����xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx�x�����xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx�x�����xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx�x����xx�x�x���xxx�xx�x

> 

> �
> 
> xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx�x����xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�xx�x�x���xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�xx�x�����xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�xxx��������xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x�������x�x�xxx�xx�x�����xx�x�x���xxx�xx�x�xxx�x���x�x���x��xx�x�x�xx�xxx���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx�x�����xx�x�x���xxx�xx�x�xxx�x���x�x���x��x�xx�x�xxx�x��x�x�x�x�x�xxxxxxxx�xxx�x���xxxxx��xxx�xxxx�xxx�x���x�x�x��x����x�������x�x�xxx�xx�x���xx�x�x���xxx�xx�x�xxx�x�����xxx�x���xx��x�x�xx�x�x�x�x�xxx�xx�x�xx�x�xxx�x��x�x�xx�����xx����x�x���x��x�xx�x�xxx�x��x�x�x���xx��x�xxx�x���x�x�x��x����x����������x�x��x�xx�x�x��xx���xx��x�x����xxx�x�x�xx��xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x����������x�x��x�x������xx���xx��x�x����xxx�x�x�x������xxx�x�x�xxxx�xxx�x���x�x���xxx�xxxx�xxx�x���x�x���xxxx�xx�xx��x�x���x�x�xx�x�x�xxxx���x�x�xx�x�x�x�x�x�x�x����x�x�x�x�x�x��������x�x�x��x����x�������x�x�xxx�x����x��xx�x�x���xxx�xx�x�xxx�x���x�x���xxx�xx�x�xxx�xx�x�xx�x�x��x��x�x [... cut ...]
I had speaker connected to aux port, But I wasn't able to hear anything.

### Expected Behavior

Able ot communicate with with my phone and have 0 errors during runtime 

### Actual Behavior

unable to hear anything through aux port and had a lot of errors related to I2C during run time

### Steps to Reproduce

- Followed ADF getting-started guide
- in the "pipeline_a2dp_sink_and_hfp" folder, selected board  ESP32-LyraT-Mini from menuconfig
- used idf.py build -p /dev/tty.usb21222 flash monitor

hbler99, 2024-09-29:
In [pipeline_a2dp_sink_and_hfp.c#L404](https://github.com/hbler99/esp-adf/blob/14c79a006dcde1b5530fb8e22d1245fe6eb9107f/examples/get-started/pipeline_a2dp_sink_and_hfp/main/a2dp_sink_and_hfp_example.c#L404), you can rewrite it to match the content of the image below.
[image]

RaresCon, 2024-12-23:
Hi @malto101! Have you resolved this issue? I have a similar problem with a custom setup using a breakout board with ES8388 and ESP32, but these errors occur after the I2S stream is initialized and processed.

malto101, 2024-12-23:
Hey @RaresCon , yes, my case was resolved 
I forgot to close this.

RaresCon, 2024-12-23:
If you have a little time to spare, can you please tell me what was the solution to this?

malto101, 2024-12-23:
It was mainly the on which channel does it get initialized in my case.
I think the current version has got it working pretty well

There is one thing we need to take care is 
If I2S stream is enabled with built-in DAC mode, please don't use I2S_NUM_1. The built-in DAC functions are only supported on I2S0 currently

RaresCon, 2024-12-23:
So the I2C and I2S should be on two separate channels for them two work at the same time? I ask this because the errors in your log where for the I2C communication.

malto101, 2024-12-23:
In my case, both were conflicting, thus the problem

mubbashirshirazi007, 2024-12-24:
hey there can you please tell does this example work the same way like Bluetooth airpods or car dialer works ? 
which uses built in speaker and microphone Like in lyrat mini board and send audio too and fro to connected device
@malto101

malto101, 2024-12-24:
Yep, it works similar to Car Hands free, the main issue is with the quality, but its a working example.

mubbashirshirazi007, 2024-12-24:
hey i have some confussion, Can you please confirm that using this example were you able to send audio using microphone to connected mobile phone Via bluetooth and were able to stream audio back to speaker ?
using A2DP sink and HFP
@malto101

malto101, 2024-12-24:
yes..

RaresCon, 2024-12-24:
I still can not get the I2C to work along with I2S. When I2S is initialized, it works just fine, but the I2C communication is broken and it returns NACKs constantly. Any ideas?

`I (1340) BLUETOOTH_EXAMPLE: [ 2 ] Start codec chip
W (1340) i2c_bus_v2: I2C master handle is NULL, will create new one
D (1350) i2c.common: new bus(0) at 0x3ffd4770
W (1350) i2c.master: Please check pull-up resistances whether be connected properly. Otherwise unexpected behavior would happen. For more detailed information, please read docs
I (1360) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0
I (1370) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0
D (1380) i2c.common: bus clock source frequency: 80000000hz
I (1400) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1400) ES8388_DRIVER: init,out:00, in:00
I (1410) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (1410) BLUETOOTH_EXAMPLE: [ 3 ] Create audio pipeline for playback
I (1420) BLUETOOTH_EXAMPLE: [4] Create i2s stream to write data to codec chip
D (1420) i2s_common: tx channel is registered on I2S0 successfully
D (1430) i2s_common: DMA malloc info: dma_desc_num = 3, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1248
D (1440) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
D (1450) i2s_common: MCLK is pinned to GPIO3 on I2S0
D (1460) i2s_std: The tx channel on I2S0 has been initialized to STD mode successfully
D (1460) i2s_common: i2s tx channel enabled
E (1470) i2c.master: I2C hardware NACK detected
E (1470) i2c.master: I2C transaction unexpected nack detected
E (1480) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
I (1490) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0
I (1500) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0
E (1510) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1510) i2c.master: I2C hardware NACK detected
E (1520) i2c.master: I2C transaction unexpected nack detected
E (1520) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
I (1530) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0
I (1540) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0
E (1550) i2c.master: i2c_master_transmit(1074): I2C transaction failed
I (1560) BLUETOOTH_EXAMPLE: [4.1] Get Bluetooth stream
W (1570) BT_BTC: A2DP Enable with AVRC
I (1580) A2DP_STREAM: Unhandled A2DP event: 4
W (1580) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
I (1590) AUDIO_THREAD: The audio_a2dp_stream_thread task allocate stack on internal memory
I (1600) BLUETOOTH_EXAMPLE: [4.2] Register all elements to audio pipeline
I (1610) BLUETOOTH_EXAMPLE: [4.3] Link it together [Bluetooth]-->bt_stream_reader-->i2s_stream_writer-->[codec_chip]
I (1620) AUDIO_PIPELINE: link el->rb, el:0x3ffd6244, tag:bt, rb:0x3ffd7060
I (1630) BLUETOOTH_EXAMPLE: [ 5 ] Initialize peripherals
I (1630) BLUETOOTH_EXAMPLE: [5.1] Create Bluetooth peripheral
I (1640) BLUETOOTH_EXAMPLE: [5.2] Start all peripherals
I (1640) AUDIO_THREAD: The esp_periph task allocate stack on internal memory
I (1650) BLUETOOTH_EXAMPLE: [ 6 ] Set up  event listener
I (1660) BLUETOOTH_EXAMPLE: [6.1] Listening event from all elements of pipeline
I (1670) BLUETOOTH_EXAMPLE: [ 7 ] Start audio_pipeline
I (1670) AUDIO_ELEMENT: [bt-0x3ffd6244] Element task created
I (1680) AUDIO_THREAD: The i2s task allocate stack on internal memory
I (1690) AUDIO_ELEMENT: [i2s-0x3ffd4e08] Element task created
I (1690) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:139732 Bytes

I (1700) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (1710) AUDIO_PIPELINE: Pipeline started
I (1710) BLUETOOTH_EXAMPLE: [ 8 ] Listen for all pipeline events`