espressif/esp-adf #1281: Cannot run example on Lyrat 4.3 board

GCTechnology, 2024-09-25:
## Environment

- Audio development kit:        ESP32-LyraT
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2):        v4.3
- [Required] Module or chip used:        ESP32-WROVER-E

- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):        v5.3.1
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):        v2.7-4-g1bc914b7
- Build system:        idf.py

- [Required] Running log:         All logs from power-on to problem recurrence

[image]
[image]

- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):    xtensa-esp-elf-gcc (crosstool-NG esp-13.2.0_20240530) 13.2.0
- 
- Operating system:     Linux (WSL Ubuntu 22.04LTS)
- Using an IDE?:           No
- Power supply:            USB

## Problem Description

I2C hardware timeout detected while running example '.\esp-adf\examples\player\pipeline_http_mp3' and '.\esp-adf\examples\player\pipeline_http_select_decoder' then reset core circularly.

### Steps to Reproduce

1. cd project folder '.\esp-adf\examples\player\pipeline_http_select_decoder'.
2. run 'cp sdkconfig.defaults.esp32 sdkconfig'
3. run 'idf.py menucondig' to setup network ssid and pass
4. run 'idf.py build'
5. flash LyraT board with 'flash_download_tool_3.9.7', bootloader location 0x1000, partition-table location 0x8000, app location 0x10000
6. reset target

GCTechnology, 2024-09-26:
[image]

New log '(i2c_bus_write_bytes):I2C Bus WriteReg Error'

GCTechnology, 2024-09-27:
Similar error using micropython `audio_test` example
but CODEC works? Sound outputs successfully.
[image]

hbler99, 2024-09-29:
Maybe there is a mismatch between the development board and the `menuconfig` setting under `Audio Hal -> Audio Board`, please check it.

GCTechnology, 2024-09-29:
> Maybe there is a mismatch between the development board and the `menuconfig` setting under `Audio Hal -> Audio Board`, please check it.

No, I've checked this item. Maybe it's a hardware issue I guess. I will recheck it on newly bought 'LyraT mini v1.2' audio board.

closed 2024-10-31 by GCTechnology