espressif/esp-adf #1282: Lyrat v4.3: setting the registries for the ES8388 doesnt work

Combreaker, 2024-09-27:
I'm working with an ESP32-Lyrat v4.3 board and use an external mic to record sound. In the documentation for the ES8388 chip, it says to set certain registry values and i did so but it doesn't change the sound of the output file.

void AudioRecorder::start()
{
    this->board_handle = audio_board_init();
    audio_hal_ctrl_codec(this->board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_ENCODE, AUDIO_HAL_CTRL_START);
    audio_hal_set_volume(this->board_handle->audio_hal, 100);
    //es8388_write_reg(ES8388_ADCCONTROL1, 0b01110111);
    es8388_write_reg(ES8388_ADCCONTROL10, 0b11111010);
    es8388_write_reg(ES8388_ADCCONTROL2, ADC_INPUT_LINPUT2_RINPUT2);
    //es8388_write_reg(ES8388_ADCCONTROL5, 0b00000110);
    //es8388_write_reg(ES8388_ADCCONTROL8, 0b11000000); //Reg16 - LADCVOL Attenuation
    //es8388_write_reg(ES8388_ADCCONTROL9, 0b11000000); //Reg17 - RADCVOL Attenuation
    es8388_write_reg(ES8388_ADCCONTROL14, 0b01011000);

    this->i2s_config = { 
        .type = AUDIO_STREAM_READER, 
        .transmit_mode = I2S_COMM_MODE_STD, 
        .chan_cfg = { 
            .id = (i2s_port_t)CODEC_ADC_I2S_PORT, 
            .role = I2S_ROLE_MASTER, 
            .dma_desc_num = 3, 
            .dma_frame_num = 312, 
            .auto_clear = true, 
        }, 
        .std_cfg = { 
            .clk_cfg = { 
                .sample_rate_hz = this->config.sample_rate,
                .clk_src = I2S_CLK_SRC_DEFAULT, 
                .mclk_multiple = I2S_MCLK_MULTIPLE_256, 
            }, 
            .slot_cfg = { 
                .data_bit_width = I2S_DATA_BIT_WIDTH_16BIT, 
                .slot_bit_width = I2S_SLOT_BIT_WIDTH_AUTO,
                .slot_mode = I2S_SLOT_MODE_MONO,
                .slot_mask = I2S_STD_SLOT_LEFT,
                .ws_width = I2S_DATA_BIT_WIDTH_16BIT, 
                .ws_pol = false, 
                .bit_shift = true, 
                .msb_right = (I2S_DATA_BIT_WIDTH_16BIT <= I2S_DATA_BIT_WIDTH_16BIT) ? true : false, }, 
                .gpio_cfg = { 
                    .invert_flags = { .mclk_inv = false, .bclk_inv = false, }, 
                }, 
        }, 
        .expand_src_bits = I2S_DATA_BIT_WIDTH_16BIT, 
        .use_alc = false, 
        .volume = 0, 
        .out_rb_size = (8 * 1024), 
        .task_stack = (3584), 
        .task_core = this->config.core, 
        .task_prio = (23), 
        .stack_in_ext = false, 
        .multi_out_num = 1, 
        .uninstall_drv = true, 
        .need_expand = false, 
        .buffer_len = (3600), 
    };

    this->wav_config = DEFAULT_WAV_ENCODER_CONFIG();
    this->wav_encoder = wav_encoder_init(&this->wav_config);

    this->fatfs_config = FATFS_STREAM_CFG_DEFAULT();
    this->fatfs_config.type = AUDIO_STREAM_WRITER;
    this->fatfs_config.task_core = 1;

    this->wav_fatfs_stream_writer = fatfs_stream_init(&this->fatfs_config);
    this->i2s_stream_reader = i2s_stream_init(&this->i2s_config);
    i2s_stream_set_clk(this->i2s_stream_reader, this->config.sample_rate, 16, 1);
    
    audio_element_info_t info = AUDIO_ELEMENT_INFO_DEFAULT();
    audio_element_getinfo(this->i2s_stream_reader, &info);
    audio_element_getinfo(this->wav_fatfs_stream_writer, &info);
    info.bits = 16;
    info.sample_rates = this->config.sample_rate;
    info.channels = 1;
    audio_element_setinfo(this->i2s_stream_reader, &info);
    audio_element_setinfo(this->wav_fatfs_stream_writer, &info);

    this->rb_i2s_to_wav = rb_create(this->config.buffer_size, 1);
    audio_element_set_input_ringbuf(this->wav_encoder, this->rb_i2s_to_wav);
    audio_element_set_output_ringbuf(this->i2s_stream_reader, this->rb_i2s_to_wav);

    this->rb_wav_to_fatstream = rb_create(this->config.buffer_size, 1);
    audio_element_set_input_ringbuf(this->wav_fatfs_stream_writer, this->rb_wav_to_fatstream);
    audio_element_set_output_ringbuf(this->wav_encoder, this->rb_wav_to_fatstream);

    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    this->pipeline = audio_pipeline_init(&pipeline_cfg);

    audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
    audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);
    esp_periph_config_t set_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    auto set = esp_periph_set_init(&set_cfg);
    audio_event_iface_set_listener(esp_periph_set_get_event_iface(set), evt);

    this->recording_thread = std::thread(&AudioRecorder::run, this);
    audio_event_iface_msg_t msg;
    while (1) {
        esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);
        if (ret != ESP_OK) {
            ESP_LOGE(this->TAG.c_str(), "[ E ] Event interface error: %d", ret);
        }
    }
}

TempoTian, 2024-10-09:
Following code set microphone gain for es8388, If you are using right channel you should set to 0..3 bits. please have a try!
[image]

```
static int es8388_set_mic_gain(audio_codec_es8388_t *codec, float db)
{
    es_mic_gain_t gain = db > 0 ? (int) (db / 3) + MIC_GAIN_0DB : MIC_GAIN_0DB;
    int res, gain_n;
    gain_n = (int) gain / 3;
    gain_n = (gain_n << 4) + gain_n;
    res = es8388_write_reg(codec, ES8388_ADCCONTROL1, gain_n); // MIC PGA
    return res;
}
```

jason-mao, 2024-12-10:
@Combreaker Has your issue been resolved？

closed 2025-02-17 by jason-mao