espressif/esp-adf #1286: i2s_stream.h in custom cpp component esp-idf 5.0.0 or higher

coencuppen, 2024-10-01:
Using i2s_stream.h in a custom cpp file leads to errors: 

> i2s_stream.h:233:1: error: designator order for field 'i2s_stream_cfg_t::expand_src_bits' does not match declaration order in 'i2s_stream_cfg_t'

which can be fixed by replacing the i2s_stream_cfg_t struct in 
`\esp-adf\components\audio_stream\include\i2s_stream.h`
under // ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0): 

```C++
typedef struct {
    audio_stream_type_t  type;                  /*!< Type of stream */
    i2s_comm_mode_t      transmit_mode;         /*!< I2S transmit mode */
    i2s_chan_config_t    chan_cfg;              /*!< I2S controller channel configuration */
    i2s_std_config_t     std_cfg;               /*!< I2S standard mode major configuration that including clock/slot/gpio configuration  */
#if SOC_I2S_SUPPORTS_PDM_RX
    i2s_pdm_rx_config_t  pdm_rx_cfg;            /*!< I2S PDM RX mode major configuration that including clock/slot/gpio configuration  */
#endif // SOC_I2S_SUPPORTS_PDM_RX
#if SOC_I2S_SUPPORTS_PDM_TX
    i2s_pdm_tx_config_t  pdm_tx_cfg;            /*!< I2S PDM TX mode major configuration that including clock/slot/gpio configuration  */
#endif // SOC_I2S_SUPPORTS_PDM_TX
#if SOC_I2S_SUPPORTS_TDM
    i2s_tdm_config_t     tdm_cfg;               /*!< I2S TDM mode major configuration that including clock/slot/gpio configuration  */
#endif // SOC_I2S_SUPPORTS_TDM
    i2s_data_bit_width_t expand_src_bits;       /*!< The source bits per sample when data expand */
    bool                 use_alc;               /*!< It is a flag for ALC. If use ALC, the value is true. Or the value is false */
    int                  volume;                /*!< The volume of audio input data will be set. */
    int                  out_rb_size;           /*!< Size of output ringbuffer */
    int                  task_stack;            /*!< Task stack size */
    int                  task_core;             /*!< Task running in core (0 or 1) */
    int                  task_prio;             /*!< Task priority (based on freeRTOS priority) */
    bool                 stack_in_ext;          /*!< Try to allocate stack in external memory */
    int                  multi_out_num;         /*!< The number of multiple output */
    bool                 uninstall_drv;         /*!< whether uninstall the i2s driver when stream destroyed*/
    bool                 need_expand;           /*!< whether to expand i2s data */
    int                  buffer_len;            /*!< Buffer length use for an Element. Note: when 'bits_per_sample' is 24 bit, the buffer length must be a multiple of 3. The recommended value is 3600 */
} i2s_stream_cfg_t;
````

by: 
```C++
typedef struct {
    audio_stream_type_t  type;                  /*!< Type of stream */
    i2s_comm_mode_t      transmit_mode;         /*!< I2S transmit mode */
    i2s_std_config_t     std_cfg;               /*!< I2S standard mode major configuration that including clock/slot/gpio configuration  */
    i2s_chan_config_t    chan_cfg;              /*!< I2S controller channel configuration */
#if SOC_I2S_SUPPORTS_PDM_RX
    i2s_pdm_rx_config_t  pdm_rx_cfg;            /*!< I2S PDM RX mode major configuration that including clock/slot/gpio configuration  */
#endif // SOC_I2S_SUPPORTS_PDM_RX
#if SOC_I2S_SUPPORTS_PDM_TX
    i2s_pdm_tx_config_t  pdm_tx_cfg;            /*!< I2S PDM TX mode major configuration that including clock/slot/gpio configuration  */
#endif // SOC_I2S_SUPPORTS_PDM_TX
#if SOC_I2S_SUPPORTS_TDM
    i2s_tdm_config_t     tdm_cfg;               /*!< I2S TDM mode major configuration that including clock/slot/gpio configuration  */
#endif // SOC_I2S_SUPPORTS_TDM
    bool                 use_alc;               /*!< It is a flag for ALC. If use ALC, the value is true. Or the value is false */
    int                  volume;                /*!< The volume of audio input data will be set. */
    int                  out_rb_size;           /*!< Size of output ringbuffer */
    int                  task_stack;            /*!< Task stack size */
    int                  task_core;             /*!< Task running in core (0 or 1) */
    int                  task_prio;             /*!< Task priority (based on freeRTOS priority) */
    bool                 stack_in_ext;          /*!< Try to allocate stack in external memory */
    int                  multi_out_num;         /*!< The number of multiple output */
    bool                 uninstall_drv;         /*!< whether uninstall the i2s driver when stream destroyed*/
    bool                 need_expand;           /*!< whether to expand i2s data */
    i2s_data_bit_width_t expand_src_bits;       /*!< The source bits per sample when data expand */
    int                  buffer_len;            /*!< Buffer length use for an Element. Note: when 'bits_per_sample' is 24 bit, the buffer length must be a multiple of 3. The recommended value is 3600 */
} i2s_stream_cfg_t;
````

reproducing this behaviour can be done by these steps under the fix of the same issue for idf 4.2:
[https://github.com/espressif/esp-adf/issues/409](url)

hbler99, 2024-10-14:
Thanks for your report, it will be solved as soon.

jason-mao, 2024-12-10:
@coencuppen I think this issue was fixed on d1f7d4a378880b64eadc5d18196c7f08e699e95a

closed 2024-12-10 by jason-mao