espressif/esp-adf #1287: 怎样在ESP32S3-BOX中添加蓝牙连接功能？

yuanfangcn, 2024-10-09:
当前ESP32S3-BOX能够WIFI连接和播放mp3（声音通过喇叭），现在怎样添加 连接蓝牙耳机，或者蓝牙音响 的功能？

hbler99, 2024-10-11:
ESP_ADF支持ESP32S3-BOX，您可以参考[pipeline_bt_sink](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_sink)， 在 `menuconfig > Audio Hal` 中选择ESP32-S3-BOX即可

hbler99, 2024-10-11:
用这个demo[pipeline_bt_source](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_source)

yuanfangcn, 2024-10-14:
了解了。
那么：
examples\player\pipeline_a2dp_source_stream 和 pipeline_bt_source 有何异同？
另外
examples\player\pipeline_a2dp_sink_stream 和 pipeline_bt_sink 有何异同？

hbler99, 2024-10-14:
pipeline_a2dp_xxx 使用`a2dp_stream_init`做初始化,在配置上更加灵活，[play_bt_music_example](https://github.com/espressif/esp-adf/blob/3b6e59bd88e7eda1c8e3a9904db0cc32fa124b48/examples/player/pipeline_a2dp_sink_stream/main/play_bt_music_example.c#L92-L95)中的`esp_bt_controller`和`esp_bluedroid`都能够进行是否选择使用;
pipeline_bt_xxx 使用`bluetooth_service_create_stream`做初始化, 直接将bt和上面选择性使用的函数都直接激活完成。

closed 2024-10-15 by yuanfangcn