espressif/esp-adf #1288: How to get current time of music being played ?

ventureg, 2024-10-10:
## Environment

- Audio development kit:     [none] esp32s3 based board
- Audio kit version                NA
- [Required] Module or chip used:  ESP32-S3-WROOM-1
- [Required] IDF version   v5.3.1
- [Required] ADF version  v2.7
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version xtensa-esp-elf-gcc (crosstool-NG esp-13.2.0_20240530) 13.2.0
- Operating system:    Linux
- Using an IDE?: No
- Power supply:  USB
- LogFile:
[current_time_position.txt](https://github.com/user-attachments/files/17322398/current_time_position.txt)

## Problem Description

**esp_audio_time_get** doesn't return the correct current time, while I was hearing music.

**esp_audio_info_get**(player, &info); can not return the correct time ethier.

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue

```cpp
void music_info(void)
{
    int curr_time_in_us = 0;
    esp_audio_state_t state;
    audio_err_t err = esp_audio_time_get(player, &curr_time_in_us);
    if (ESP_ERR_AUDIO_NO_ERROR != err)
    {
        ESP_LOGE(TAG, "esp_audio_time_get failed %d", err);
    }

    if (NULL == ringbuf_handle)
    {
        err = esp_audio_info_get(player, &info);
        ESP_LOGI(TAG, "esp_audio instance is:%p err %d, output handle %p\r\n", player, err, info.out_el);
    
        ringbuf_handle = audio_element_get_input_ringbuf(info.out_el);
        spawn_task_to_play_mp3(NULL);
    }
    err = esp_audio_info_get(player, &info);
    if (ESP_ERR_AUDIO_NO_ERROR != err)
    {
        ESP_LOGE(TAG, "esp_audio_info_get failed %d", err);
    }
    err = esp_audio_state_get(player, &state);
    if (ESP_ERR_AUDIO_NO_ERROR != err)
    {
        ESP_LOGE(TAG, "esp_audio_state_get failed %d", err);
    }
    ESP_LOGI(TAG, "esp_audio instance is:%p err %d, state is %d, curr %d, time pos %d\r\n", player, err, state.status, curr_time_in_us, info.time_pos);
    return;
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
[current_time_position.txt](https://github.com/user-attachments/files/17322398/current_time_position.txt)

```

```

TempoTian, 2024-10-11:
The API to get current play time is `esp_audio_time_get`

ventureg, 2024-10-12:
As shown in the code snippet,  esp_audio_time_get is used, unfortunately it doesn't work on esp32s3

TempoTian, 2024-10-12:
Is it possible to share your code, so we can check or reproduce it? For `esp_audio` is pure softwre related code, it won't change the behavior when change to different board.

ventureg, 2024-10-12:
Please see the attached source code, I changed the suffix to png, actually it is a C file.

This project sets up a web server, tries to decode music files which are stored in spiff,  and send the decoded PCM stream through websocket to web client.

This file is mainly copied from cli folder.  High level API **ESP AUDIO** is used to play the music. And a raw stream servers as the last audio element in pipeline. The pipeline looks like: spiffs->mp3 decoder -> ReSample filter -> raw stream.

A dedicated task **mp3_player_task** is used to sink PCM data from the ending raw stream, and send it through web socket protocol.

Regularly function **music_info** is called to print the current time info.

[image]

TempoTian, 2024-10-12:
Seems same as cli, I tested with cli after play, can use getpos to get current position correctly. You can compare your code with the cli realization
[image]

jason-mao, 2024-12-10:
This topic has become inactive, so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2024-12-10 by jason-mao