espressif/esp-adf #1290: How to seek to a position  in milisecond

ventureg, 2024-10-10:
I can not seek to a millisecond specified position.

could esp_audio_seek provide the parameter in milisecond ?

Thanks

TempoTian, 2024-10-11:
Currently `esp_audio` only support seek to second not support seek to millisecond

ventureg, 2024-10-12:
Could there be a plan to add this feature ?

jason-mao, 2024-12-10:
@ventureg The seek with millisecond is a big change, we can support it on the next major version.

ventureg, 2024-12-10:
When will the next major version be available ?

| |
Tom
|
|
***@***.***
|
---- Replied Message ----
| From | ***@***.***> |
| Date | 12/10/2024 14:59 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [espressif/esp-adf] How to seek to a position in milisecond (AUD-5750) (Issue #1290) |

@ventureg The seek with millisecond is a big change, we can support it on the next major version.