espressif/esp-adf #1291: 解码后获取数据做fft还不影响pipline发送数据到i2s writer?

bailiqun, 2024-10-10:
sdcard -> fatfs_stream_reader-->decoder-->i2s_stream-->speaker
                                                           | -------> fft 
 总体pipline是这样的,然后我需要获的decoder输出的pcm数据, i2s_stream发送前的数据做fft变换,有什么办法吗?
 
十分感谢

TempoTian, 2024-10-11:
你可以用下面的方法做个tricking，用新的write callback覆盖I2S的write函数
不过为了避免播放断音，最好很快的处理完，如果比较耗时可以先写到一个ringbuf然后在另起一个Task处理
```c

static int i2s_write_cb(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx)
{
    stream_func original_write_func = (stream_func)ctx;
    // Process your data, you must process data in time to avoid I2S write data blocked,
    int ret = original_write_func(el, buf, len, wait_time, ctx);
    if (ret < 0) {
        ESP_LOGE(TAG, "i2s write failed");
    }
    return bytes_write;
}

// Set I2S new write callback function
 audio_element_set_write_cb(i2s_writer, i2s_write_cb, audio_element_get_write_cb(i2s_writer)); // context is original write function

```

bailiqun, 2024-10-11:
帅的一塌糊涂

closed 2024-10-11 by bailiqun