espressif/esp-adf #1293: ESP32 I2C主机模式是否在I2C总线出问题时会导致程序崩溃？

SantaPasserby, 2024-10-12:
## Environment

- [Required] Module or chip used:  [ESP32-WROOM-32E]
- [Required] IDF version：v5.3.1-395-g707d097b01
- [Required] ADF version: v2.7
- Build system:         [idf.py]
- [Required] Running log:         

`ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7524
load:0x40078000,len:16540
load:0x40080400,len:4
--- 0x40080400: _init at ??:?
load:0x40080404,len:4312
entry 0x4008065c
I (29) boot: ESP-IDF v5.3.1-395-g707d097b01-dirty 2nd stage bootloader
I (29) boot: compile time Oct 12 2024 02:30:27
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 40MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00011000 00004000
I (82) boot:  1 otadata          OTA data         01 00 00015000 00002000
I (90) boot:  2 phy_init         RF data          01 01 00017000 00001000
I (97) boot:  3 ota_0            OTA app          00 10 00020000 00300000
I (105) boot:  4 ota_1            OTA app          00 11 00320000 00300000
I (112) boot:  5 storage          Unknown data     01 81 00620000 00400000
I (120) boot: End of partition table
I (124) boot: No factory image, trying OTA 0
I (129) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=0f960h ( 63840) map
I (156) esp_image: segment 1: paddr=0002f988 vaddr=3ffbdb60 size=00690h (  1680) load
I (157) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=25068h (151656) map
I (206) esp_image: segment 3: paddr=00055090 vaddr=3ffbe1f0 size=01e5ch (  7772) load
I (208) esp_image: segment 4: paddr=00056ef4 vaddr=40080000 size=0dab0h ( 55984) load
I (237) boot: Loaded app from partition at offset 0x20000
I (279) boot: Set actual ota_seq=1 in otadata[0]
I (279) boot: Disabling RNG early entropy source...
I (290) quad_psram: This chip is ESP32-D0WD
I (290) esp_psram: Found 8MB PSRAM device
I (290) esp_psram: Speed: 40MHz
I (294) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (301) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (311) cpu_start: Multicore app
I (326) cpu_start: Pro cpu start user code
I (326) cpu_start: cpu freq: 240000000 Hz
I (326) app_init: Application information:
I (331) app_init: Project name:     IPB_AS_APP_ESP32
I (337) app_init: App version:      065255f-dirty
I (342) app_init: Compile time:     Oct 12 2024 02:30:07
I (348) app_init: ELF file SHA256:  fdd1478bc8660eae...
I (354) app_init: ESP-IDF:          v5.3.1-395-g707d097b01-dirty
I (361) efuse_init: Min chip rev:     v3.0
I (365) efuse_init: Max chip rev:     v3.99 
I (370) efuse_init: Chip rev:         v3.0
I (375) heap_init: Initializing. RAM available for dynamic allocation:
I (382) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (388) heap_init: At 3FFC0C40 len 0001F3C0 (124 KiB): DRAM
I (395) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (401) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (407) heap_init: At 4008DAB0 len 00012550 (73 KiB): IRAM
I (414) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
[... 317 lines cut ...]
            }
            if(extend_mcu_read_reg(I2C_ADDR_EXTEND_MCU, REG_ADDR_SERVERIP_ADDR1, &buf) == ESP_OK) {
                val = (addr >> 8) & 0xFF;
                if(buf != val) {
                    extend_mcu_write_reg(I2C_ADDR_EXTEND_MCU, REG_ADDR_SERVERIP_ADDR1, &val);
                }
            }
            if(extend_mcu_read_reg(I2C_ADDR_EXTEND_MCU, REG_ADDR_SERVERIP_ADDR2, &buf) == ESP_OK) {
                val = (addr >> 16) & 0xFF;         
                if(buf != val) {
                    extend_mcu_write_reg(I2C_ADDR_EXTEND_MCU, REG_ADDR_SERVERIP_ADDR2, &val);
                }
            }
            if(extend_mcu_read_reg(I2C_ADDR_EXTEND_MCU, REG_ADDR_SERVERIP_ADDR3, &buf) == ESP_OK) {
                val = (addr >> 24) & 0xFF;
                if(buf != val) {
                    extend_mcu_write_reg(I2C_ADDR_EXTEND_MCU, REG_ADDR_SERVERIP_ADDR3, &val);
                }
            }
        }
        vTaskDelay(pdMS_TO_TICKS(200));
    }
}

```

closed 2024-10-15 by SantaPasserby

theGuen, 2024-11-07:
How did you solve this?

i get no connection on the i2c on a Audio-dev kit 

log is like 

```
9:21:02.776 -> [0;32mI (461) cpu_start: cpu freq: 160000000 Hz[0m
09:21:02.776 -> [0;32mI (461) app_init: Application information:[0m
09:21:02.776 -> [0;32mI (464) app_init: Project name:     play_bt_music[0m
09:21:02.776 -> [0;32mI (469) app_init: App version:      v2.7-18-gff7f39dc-dirty[0m
09:21:02.776 -> [0;32mI (476) app_init: Compile time:     Nov  7 2024 09:02:02[0m
09:21:02.776 -> [0;32mI (482) app_init: ELF file SHA256:  37f125b9e...[0m
09:21:02.776 -> [0;32mI (487) app_init: ESP-IDF:          v5.3.1-dirty[0m
09:21:02.776 -> [0;32mI (492) efuse_init: Min chip rev:     v0.0[0m
09:21:02.776 -> [0;32mI (497) efuse_init: Max chip rev:     v3.99 [0m
09:21:02.776 -> [0;32mI (502) efuse_init: Chip rev:         v3.0[0m
09:21:02.776 -> [0;32mI (507) heap_init: Initializing. RAM available for dynamic allocation:[0m
09:21:02.776 -> [0;32mI (514) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
09:21:02.776 -> [0;32mI (520) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
09:21:02.776 -> [0;32mI (526) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
09:21:02.776 -> [0;32mI (532) heap_init: At 3FFCC670 len 00013990 (78 KiB): DRAM[0m
09:21:02.776 -> [0;32mI (538) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
09:21:02.776 -> [0;32mI (545) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
09:21:02.776 -> [0;32mI (551) heap_init: At 4009BAF4 len 0000450C (17 KiB): IRAM[0m
09:21:02.776 -> [0;32mI (559) spi_flash: detected chip: generic[0m
09:21:02.776 -> [0;32mI (562) spi_flash: flash io: dio[0m
09:21:02.776 -> [0;33mW (566) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m
09:21:02.776 -> [0;32mI (580) coexist: coex firmware version: 4482466[0m
09:21:02.776 -> [0;32mI (585) main_task: Started on CPU0[0m
09:21:02.776 -> [0;32mI (595) main_task: Calling app_main()[0m
09:21:02.776 -> [0;32mI (595) BLUETOOTH_EXAMPLE: [ 1 ] Create Bluetooth service[0m
09:21:02.776 -> [0;32mI (595) BTDM_INIT: BT controller compile version [b022216][0m
09:21:02.776 -> [0;32mI (605) BTDM_INIT: Bluetooth MAC: 78:21:84:d3:be:b2[0m
09:21:02.776 -> [0;32mI (605) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08[0m
09:21:03.288 -> [0;33mW (1265) BT_BTC: btc_avrc_tg_set_rn_supported_evt failed: AVRC TG not yet initialized
09:21:03.319 -> [0m
09:21:03.319 -> [0;33mW (1275) BT_BTC: A2DP Enable with AVRC[0m
09:21:03.319 -> [0;32mI (1285) BLUETOOTH_EXAMPLE: [ 2 ] Start codec chip[0m
09:21:03.319 -> [0;32mI (1285) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
09:21:03.319 -> [0;33mW (1295) i2c_bus_v2: I2C master handle is NULL, will create new one[0m
09:21:03.352 -> D (1295) i2c.common: new bus(0) at 0x3ffd6970[0m
09:21:03.352 -> [0;32mI (1305) gpio: GPIO[33]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [0m
09:21:03.352 -> [0;32mI (1315) gpio: GPIO[32]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [0m
09:21:03.352 -> D (1315) i2c.common: bus clock source frequency: 80000000hz[0m
09:21:03.352 -> [0;32mI (1325) gpio: GPIO[33]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [0m
09:21:03.385 -> [0;32mI (1335) gpio: GPIO[32]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [0m
09:21:03.580 -> [0;31mE (1545) i2c.master: I2C software timeout[0m
09:21:03.580 -> [0;31mE (1545) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed[0m
```

SantaPasserby, 2024-11-07:
@theGuen 
`ADF-V2.7` uses `i2c_bus_v2.c`, which is the V2 version. I used `ADF-V2.6` with `i2c_bus.c` before, and the product has always been stable and has not caused any problems.
Later I verified that `i2c_buf_v2.c` would access the null pointer, but I didn't have time to find out what the problem was.

So I modified the `CMakeLists.txt` file to force the use of `i2c_bus.c` instead of `i2c_bus_v2.c`. The location I modified is `${ADF_PATH}/components/esp_peripherals/CMakeLists.txt`.

```
# if("${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}" VERSION_LESS "5.3")
    list(APPEND COMPONENT_SRCS  ./driver/i2c_bus/i2c_bus.c)
# else()
#     list(APPEND COMPONENT_SRCS  ./driver/i2c_bus/i2c_bus_v2.c)
# endif()
```
[image]