espressif/esp-adf #1297: [lcd_camera]ERROR: 'git config --get remote.origin.url' failed with exit code 1

BigJohnn, 2024-10-23:
## Environment

- Audio development kit:      [ESP32-S3-Korvo-2-LCD]
- Audio kit version (for ESP32-S3-Korvo-2): [v3.1]
- [Required] Module or chip used:  [ESP32-S3-WROOM-1]
- [Required] IDF version (tried v5.0-e5617, v5.2-bdcdd, v5.3-595505):
    // v4.4-191-g83daa6dabb
- [Required] ADF version (v2.7-9cf556):
    // v2.3-171-gaac5912
- Build system:         [idf.py]
- [Required] Building log:        
- Executing action: confserver
Running cmake in directory D:\Codes\esp\esp-adf\examples\display\lcd_camera\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:\softwares\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 D:\Codes\esp\esp-adf\examples\display\lcd_camera"...
-- Found Git: D:/softwares/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 

-- ccache will be used for faster recompilation

-- The C compiler identification is GNU 13.2.0

-- The CXX compiler identification is GNU 13.2.0

-- The ASM compiler identification is GNU

-- Found assembler: D:/softwares/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: D:/softwares/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: D:/softwares/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- Building ESP-IDF components for target esp32s3

Dependencies lock doesn't exist, solving dependencies.

---------------------------ERROR--------------------------

CMake Error at D:/softwares/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: 'git config --get remote.origin.url' failed with exit code 1

  

  

Call Stack (most recent call first):

-----------------------END OF ERROR-----------------------
-- Configuring incomplete, errors occurred!
See also "D:/Codes/esp/esp-adf/examples/display/lcd_camera/build/CMakeFiles/CMakeOutput.log".

---------------------------ERROR--------------------------

  D:/softwares/esp/v5.3.1/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

cmake failed with exit code 1, output of the command is in the D:\Codes\esp\esp-adf\examples\display\lcd_camera\build\log\idf_py_stderr_output_33992 and D:\Codes\esp\esp-adf\examples\display\lcd_camera\build\log\idf_py_stdout_output_33992

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
   xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0, in case of idf-v5.3
- Operating system:     [Windows]
- (Windows only) Environment type: [MSYS2 mingw32 && ESP Command Prompt &&PowerShell tried]
- Using an IDE?: [Yes (vscode plugin, but also install.bat+export.bat tried...)]
- Power supply:         [USB]

## Problem Description

尝试了三个版本的idf，总是报ERROR: 'git config --get remote.origin.url' failed with exit code 1
但是idf-terminal其实是输出正常的：
PS D:\Codes\esp\esp-adf\examples\display\lcd_camera> git config --get remote.origin.url
git@github.com:espressif/esp-adf.git

### Expected Behavior
不报上述error

### Steps to Reproduce

1. 选择target，或menuconfig，或build，都会出

## Other Items If Possible

- 目前只有lcd_camera这个demo会出，想测一下新买的korvo2，相机模组来着，以上。

TempoTian, 2024-10-28:
这个大概率是你的build环境没办法拉github目录吧，这个example需要拉 https://github.com/espressif/esp32-camera.git
你可以手动下载下来放到工程的components目录下面，然后删掉  yml 文件试试。 https://github.com/espressif/esp-adf/blob/master/examples/display/lcd_camera/main/idf_component.yml

BigJohnn, 2024-10-28:
> 这个大概率是你的build环境没办法拉github目录吧，这个example需要拉 https://github.com/espressif/esp32-camera.git 你可以手动下载下来放到工程的components目录下面，然后删掉 yml 文件试试。 https://github.com/espressif/esp-adf/blob/master/examples/display/lcd_camera/main/idf_component.yml

多谢，是这个问题了。 另外请教下，我直接改idf_components.yml里面git链接也是不起作用，可能是啥原因?

```
dependencies:
  espressif/jsmn: "^1.1.0"
  esp32-camera:
  git: git@github.com:espressif/esp32-camera.git  # 这个git clone 是非常顺畅的
```

closed 2024-10-28 by BigJohnn