espressif/esp-adf #1298: M4A parsing fails for files recorded on Android (Google Pixel 8) devices

nathan-swidget, 2024-10-23:
Custom Board using ESP-S3-PICO-1
ESP-IDF: release/v5.3 (707d097b01756687cca18be855a2675d150247ae)  
ESP-ADF: v2.7 (9cf556de500019bb79f3bb84c821fda37668c052) 

Hey, 
I'm running into issues when trying to playback M4A recordings generated from Android devices. The sample m4a file, aswell as recordings from iOS work as expected.

```
E (980) M4A_PARSER: Sample number miss match, line:566
E (990) M4A_PARSER: Error opening audio
E (995) AAC_DECODER: Error mp4 stream, release
E (1000) AUDIO_ELEMENT: [dec] AEL_STATUS_ERROR_OPEN,-1
```
The full log output can be found below.

Example of failing file: https://spades-test-public-access.s3.us-east-1.amazonaws.com/android.m4a

```
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_mem.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "mp3_decoder.h"
#include "aac_decoder.h"
#include "http_stream.h"
#include "board.h"
#include <string.h>

#include <string>

static const char *TAG = "audio test app";

extern const uint8_t audio_start_asm[] asm("_binary_android_m4a_start");
extern const uint8_t audio_end_asm[] asm("_binary_android_m4a_end");

typedef struct  {
    const uint8_t* data;
    int           size;
} buf_stream_t;

typedef struct {
    const uint8_t* data;
    int size;
} buf_stream_cfg_t;

static esp_err_t buf_stream_open(audio_element_handle_t self)
{
    buf_stream_t *buf_stream = (buf_stream_t *)audio_element_getdata(self);
    int ret = audio_element_set_total_bytes(self, buf_stream->size);
    return ret;
}

static audio_element_err_t buf_stream_read(audio_element_handle_t self, char *buffer, int len, TickType_t ticks_to_wait, void *context)
{
    buf_stream_t *buf_stream = (buf_stream_t *)audio_element_getdata(self);
    audio_element_info_t info;
    audio_element_getinfo(self, &info);
    int pos = info.byte_pos;
    int rlen = len;
    if (pos + rlen > buf_stream->size) {
        rlen = buf_stream->size - pos;
    }
    if (rlen) {
        memcpy(buffer, buf_stream->data + pos, rlen);
    }
    printf("Start to read %d/%d\n", pos, buf_stream->size);
    if (rlen == 0) {
        printf("No more data, ret: %d\r\n", rlen);
[... 267 lines cut ...]
Start to read 5120/84596
Start to read 6144/84596
Start to read 7168/84596
Start to read 8192/84596
E (980) M4A_PARSER: Sample number miss match, line:566
E (990) M4A_PARSER: Error opening audio
E (995) AAC_DECODER: Error mp4 stream, release
E (1000) AUDIO_ELEMENT: [dec] AEL_STATUS_ERROR_OPEN,-1
W (1006) AUDIO_ELEMENT: [dec] audio_element_on_cmd_error,7
I (1012) AAC_DECODER: Closed by [7]
Start to read 9216/84596
W (1018) AUDIO_ELEMENT: OUT-[source] AEL_IO_ABORT
I (1024) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
W (1057) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
I (1057) AUDIO_PIPELINE: Pipeline started
W (1057) AUDIO_ELEMENT: [source] Element already stopped
W (1062) AUDIO_ELEMENT: [dec] Element already stopped
W (1067) AUDIO_ELEMENT: [i2s] Element already stopped
I (1073) AUDIO_PIPELINE: audio_pipeline_unlinked
W (1078) AUDIO_ELEMENT: [i2s] Element already stopped
W (1085) AUDIO_ELEMENT: [dec] Element already stopped
I (1090) CODEC_ELEMENT_HELPER: The element is 0x3c0710f0. The reserve data 2 is 0x0.
W (1098) AUDIO_PIPELINE: There are no listener registered
I (1104) main_task: Returned from app_main()
```

TempoTian, 2024-10-29:
After check, we found the m4a use `co64` instead of stco box, which not yet supported by the parser.
[image]

We will check and try to enhance the parse logic.  So your use senario including playback all recorded m4a file from all mobile devices? Use `co64` box waste some storage size for small file ( < 4GB) actually, not quite suitable for IOT devices (which commonly use FATFS).

TempoTian, 2024-10-31:
I have done the test for co64 support, you can dowload following lib and replace into folder `esp-adf-libs/esp_codec/lib/esp32s3` and test whether it fix your issues.
[libesp_processing.zip](https://github.com/user-attachments/files/17581430/libesp_processing.zip)

nathan-swidget, 2024-11-01:
This lib does fix the issue, thanks!

closed 2024-11-01 by nathan-swidget