espressif/esp-adf #1299: 我使用ESP32-S3-Korvo-2板子使用esp-rtc例程与esp-rtc的客户端进行通信，手机端无法收到esp32的声音，并且一段时间会自动挂断

monskate, 2024-10-27:
----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                 ESP_AUDIO-v1.7.3-40-g00ffda8-dbab925-9815982               |
|                     Compile date: Apr 12 2024-02:34:27                     |
------------------------------------------------------------------------------

[0;32mI (3180) ESP_AUDIO_CTRL: Func:media_ctrl_create, Line:355, MEM Total:6502652 Bytes, Inter:211155 Bytes, Dram:211155 Bytes, Dram largest free:135168Bytes

[0m
[0;32mI (3194) MP3_DECODER: MP3 init[0m
[0;33mW (3198) I2S_STREAM_IDF5.x: I2S(2) already startup[0m
[0;32mI (3203) ESP_RTC_Demo: [ 4 ] Initialize rtc service[0m
[0;32mI (3209) ESP_RTC_CORE: esp_rtc ver: 1.1.0 compile date: Feb  5 2024-12:25:35
[0m
[0;32mI (3217) SIP: Conecting...[0m
[0;33mW (3220) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1835[0m
[0;32mI (3228) SIP: [1970-01-01/00:00:01]=======WRITE 0602 bytes>>[0m
[0;32mI (3233) SIP: 

REGISTER sip:192.168.10.108:5060 SIP/2.0

Via: SIP/2.0/UDP 192.168.10.127:11082;branch=z9hG4bK-1783780767;rport

From: <sip:1005@192.168.10.108:5060>;tag=-1844547210

To: <sip:1005@192.168.10.108:5060>

Contact: <sip:1005@192.168.10.127:11082;transport=UDP>

Max-Forwards: 70

Call-ID: FF9DF909A7215939E3E29B9BBF45518BDB2BAA736106

CSeq: 1 REGISTER

Expires: 3600

User-Agent: ESP32 SIP/2.0

Content-Length: 0

Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE

Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri

Allow-Events: presence, kpml

[0m
[0;32mI (3289) SIP: [1970-01-01/00:00:01]=======================>>[0m
[0;32mI (3220) ESP_RTC_CORE: esp_rtc_init done[0m
I (3439) wifi:<ba-add>idx:1 (ifx:0, 06:b8:e6:d8:14:d4), tid:0, ssn:0, winSize:64
[0;32mI (3441) SIP: [1970-01-01/00:00:01]<<=====READ 0638 bytes==[0m
[0;32mI (3441) SIP: 

SIP/2.0 401 Unauthorized

Via: SIP/2.0/UDP 192.168.10.127:11082;branch=z9hG4bK-1783780767;rport=11082

From: <sip:1005@192.168.10.108:5060>;tag=-1844547210

To: <sip:1005@192.168.10.108:5060>;tag=942c776QBBmXc

Call-ID: FF9DF909A7215939E3E29B9BBF45518BDB2BAA736106

CSeq: 1 REGISTER

User-Agent: FreeSWITCH-mod_sofia/1.10.12-release~a88d069d6f~64bit

Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE

[... 778 lines cut ...]
[0m
[0;32mI (22302) SIP: [1970-01-01/00:00:10]<<======================[0m
[0;32mI (22309) RTC_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_BEGIN[0m
[0;32mI (22315) AUDIO_PIPELINE: link el->rb, el:0x3c203a44, tag:algo, rb:0x3c20ad6c[0m
[0;32mI (22323) AUDIO_THREAD: The algo task allocate stack on external memory[0m
[0;32mI (22330) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:6430376 Bytes, Inter:204563 Bytes, Dram:204563 Bytes, Dram largest free:131072Bytes

[0m
[0;33mW (22345) SR_SYS: Data cache recommends 64KB[0m
[0;33mW (22349) SR_SYS: Data cache line recommends 64B[0m
[0;32mI (22364) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (22364) AUDIO_THREAD: The _audio_enc task allocate stack on external memory[0m
[0;32mI (22368) AV_STREAM: audio_enc started[0m
[0;32mI (22372) AUDIO_THREAD: The _audio_dec task allocate stack on external memory[0m
[0;32mI (22374) AUDIO_THREAD: The algo_fetch task allocate stack on external memory[0m
[0;32mI (22388) AV_STREAM: _audio_enc task stoped[0m
[0;32mI (22392) AV_STREAM: audio_dec started[0m
[0;32mI (22397) RTC_SERVICE: ESP_RTC_EVENT_VIDEO_SESSION_BEGIN[0m
[0;32mI (22403) AUDIO_THREAD: The _video_enc task allocate stack on external memory[0m
[0;32mI (22419) ALGORITHM_STREAM: _algo_fetch_task is stopped[0m
[0;33mW (22431) SIP: CHANGE STATE FROM 16, TO 32, :func: sip_uas_process_req:1078[0m

[0;33mW (32001) AV_STREAM: audio decoder ringbuf write timeout[0m

[0;32mI (33745) SIP: Sending keep-alive to server[0m

monskate, 2024-10-27:
[image]

TempoTian, 2024-10-28:
请帮忙提供下具体的ADF版本信息，最新修复了一个因为I2S stream升级导致的读数据卡住的问题，可以检查下你的版本有没有修复。
https://github.com/espressif/esp-adf/commit/41f94ef398e63897bf4f7074a48456b5eb1b737e

monskate, 2024-10-30:
我尝试更改之后，摄像头数据也无法传输过来

monskate, 2024-10-30:
> 请帮忙提供下具体的ADF版本信息，最新修复了一个因为I2S stream升级导致的读数据卡住的问题，可以检查下你的版本有没有修复。 [41f94ef](https://github.com/espressif/esp-adf/commit/41f94ef398e63897bf4f7074a48456b5eb1b737e)

我尝试更改之后，摄像头数据也无法传输过来

TempoTian, 2024-10-31:
提供下log，帮你看下可能什么原因。你也可以在`examples/protocols/components/av_stream/av_stream.c` 加一些log诊断下