espressif/esp-adf #1301: Micropython-adf patch does not apply

sheetaljain01, 2024-10-28:
Hi, 

I'm facing an issue while setting up MicroPython ADF while applying the patch command. Could you help me troubleshoot?"

If you'd like, feel free to provide more details on the specific issue you’re experiencing with the setup, and I can help guide you through the troubleshooting steps!

sheetaljain01@SheetalJain:~$ git apply $HOME/esp/esp-adf/idf_patches/idf_v5.0_freertos.patch
error: patch failed: esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:726
error: esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c: patch does not apply
sheetaljain01@SheetalJain:~$

TempoTian, 2024-10-29:
You should change folder to IDF_PATH and retry. shell command like `cd $IDF_PATH` if you have already do `. export.sh`

jason-mao, 2024-12-10:
@sheetaljain01 Does this resolve your issue?

closed 2025-02-17 by jason-mao