espressif/esp-adf #1304: lcd_camera- undefined reference to `app_main'

mrmonteith, 2024-10-29:
## Environment

- Audio development kit:      |ESP32-S3-Korvo-2|ESP32-S3-Korvo-2-LCD
- Audio kit version ESP32-S3-Korvo-2: v3
- [Required] Module or chip used:  [ESP32-S3-WROOM-1]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v5.3
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.7-12-gd16b0401
- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // xtensa-esp-elf-gcc.exe (crosstool-NG esp-13.2.0_20240530) 13.2.0
- Operating system:     [Windows]
- (Windows only) Environment type: Plain Command Prompt]
- Using an IDE?: [No]
- Power supply:         [USB]

## Problem Description
idf.py build - on lcd_camera receives error - undefined reference to `app_main'

// Detailed problem description goes here.
[1/1] cmd.exe /C "cd /D C:\Users\micha\esp\lcd_camera\buil...sers/micha/esp/lcd_camera/build/bootloader/bootloader.bin"
Bootloader binary size 0x5260 bytes. 0x2da0 bytes (36%) free.
[2/4] Linking CXX executable lcd_camera.elf
FAILED: lcd_camera.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/micha/esp/lcd_camera/build/lcd_camera.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\lcd_camera.elf.rsp -o lcd_camera.elf  && cd ."
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(app_startup.c.obj):(.literal.main_task+0x28): undefined reference to `app_main'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(app_startup.c.obj): in function `main_task':
C:/Espressif/frameworks/esp-idf-v5.3-2/components/freertos/app_startup.c:206:(.text.main_task+0xd8): undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\micha\esp\lcd_camera\build\log\idf_py_stderr_output_16324 and C:\Users\micha\esp\lcd_camera\build\log\idf_py_stdout_output_16324

### Expected Behavior
Exit 0
### Actual Behavior
Exit 1
### Steps to Reproduce
cd %userprofile%\esp\lcd_camera
idf.py set-target esp32s3
idf.py menuconfig
idf.py build

// If possible, attach a picture of your setup/wiring here.
At this point wiring is irrelevant

### Code to Reproduce This Issue

 lcd_camera example code

[sdkconfig.json](https://github.com/user-attachments/files/17563635/sdkconfig.json)

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible
[idf_py_stdout_output_16324.txt](https://github.com/user-attachments/files/17563662/idf_py_stdout_output_16324.txt)
[idf_py_stderr_output_16324.txt](https://github.com/user-attachments/files/17563663/idf_py_stderr_output_16324.txt)

shootao, 2024-11-05:
Hi @mrmonteith 
In your audio board, you need to enable **FUNC_LCD_SCREEN_EN** and **FUNC_CAMERA_EN** 
[ref link](https://github.com/espressif/esp-adf/blob/ff7f39dcf0a0da87671ff4ab422eb258be429049/components/audio_board/esp32_s3_korvo2_v3/board_def.h#L39)

gamename, 2024-12-25:
@shootao @mrmonteith 

I just got the same error. What causes this?  I am on an esp32-s3-DevKitC-1 board. I am just trying to play an MP3 file.

Suggestions?

shootao, 2025-02-13:
Hi @gamename 
Which example are you testing ？