espressif/esp-adf #1305: Play audio on two channels

GCTechnology, 2024-10-31:
Hi, I am using an ESP32S3 module to develop a project with dual channel audio output. What I mean is using the two I2S interfaces to connect two codecs separately and output audio independently.
So I'd like to know if it is possible to use the current ADF library's 'audio_pipeline' audio player to implement the feature? It would be even better if there were corresponding examples.
[image]

jason-mao, 2024-12-10:
@GCTechnology Is your audio in stereo? If yes, I think you should implement an element to change the interleave audio to un-interleave audio， then put the un-interleave audio into two I2S streams.

GCTechnology, 2024-12-12:
> @GCTechnology Is your audio in stereo? If yes, I think you should implement an element to change the interleave audio to un-interleave audio， then put the un-interleave audio into two I2S streams.

Ah, I almost forget this issue I pushed. I have successfully implemented this feature by modifying example 'play_mp3_control'.
Thanks for your reply.

closed 2024-12-12 by GCTechnology