espressif/esp-adf #1306: VOIP 例程，在sip推送的振铃和通话过程存在断续的破音情况，请教一下该怎么优化

ktoto2011, 2024-11-04:
[ √] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
[ √] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
[ √] I have searched the issue tracker for a similar issue and not found a similar issue.

Environment
Audio development kit: ESP32-S3-Korvo-2
Audio kit version (ESP32-S3-Korvo-2): v3
[Required] Module or chip used: ESP32-S3-WROOM-1
[Required] IDF version v4.47
[Required] ADF version v2.6-183-gce64fea5
Build system: [idf.py]
[Required] Running log: All logs from power-on to problem recurrence
Operating system: [Windows]
(Windows only) Environment type: [ESP Command Prompt]
Using an IDE?: vscode
Power supply: USB
## Problem Description

在sip推送的振铃和通话过程存在断续的破音情况（见录像文件），请教一下该怎么优化？但是设备播放本地flash 存储的音频文件是正常的，这过程音量的设置大小是一致的，网络连接的热点强度大概在-38~40dbm 左右，wifi信号很好。

ktoto2011, 2024-11-04:
https://github.com/user-attachments/assets/ed4a5a8e-fe38-486f-9d77-863f23c53397

sip推送过来的振铃，偶尔出现破音

ktoto2011, 2024-11-04:
https://github.com/user-attachments/assets/41aec0a4-b71e-4c22-9c61-4ad7383534ab

播放本地flash的振铃，没有破音，效果很好

ktoto2011, 2024-11-04:
https://github.com/user-attachments/assets/eba3fe7b-b0c3-483c-9c1f-5a4a50696f6d

这个录像是先播放本地的振铃音乐，然后播放接收到sip 推送过来的振铃提示（从欢迎致电....这是sip推送过来的振铃音）

TempoTian, 2024-11-05:
从现象上看应该输入的数据偏慢了一点，可以用audio_sal里面的task打印函数`audio_sys_get_real_time_stats`看下各个task的优先级。目前receive task默认的优先级是20，看下有没有被别的task抢占的情况。也可以在收data的地方打印下看看`static int _receive_audio(unsigned char *data, int len, void *ctx)`, 加时间戳看看进来的数据速率，如果是G711的codec，可以简单的分析下看输入数据量是否能满足实时播放需求

ktoto2011, 2024-11-10:
> 从现象上看应该输入的数据偏慢了一点，可以用audio_sal里面的task打印函数`audio_sys_get_real_time_stats`看下各个task的优先级。目前receive task默认的优先级是20，看下有没有被别的task抢占的情况。也可以在收data的地方打印下看看`static int _receive_audio(unsigned char *data, int len, void *ctx)`, 加时间戳看看进来的数据速率，如果是G711的codec，可以简单的分析下看输入数据量是否能满足实时播放需求

感谢您及时的回应，这边后面更换好一点的喇叭，勉强能接受当前效果

TempoTian, 2024-11-22:
下面的地方改下会好，会损失一些实时性
[image]

closed 2025-02-17 by jason-mao