espressif/esp-adf #1311: I2S at 32 bits high pitch distortion - see attached example

expresspotato, 2024-11-08:
## Environment

- Audio development kit: ESP32-LyraT and custom board
- IDF version: 4.4.8
- ADF version 2.7 and master
- Build system: idf.py
- Operating system: Mac OS
- Using an IDE: VS Code
- Power supply: USB and custom board

## Problem Description

When playing back S32LE (Signed Linear 32 Bit, Little Endian) to the DAC either on the LytraT or our custom board (different DAC), there is some high end distortion for some reason.

It seems to be a problem with 32 bits specifically, as 16 bit content plays fine. 32 bit content created by bit-conversion using esp-adf-libs / esp_audio_effect also exhibits the same problem, which points me to esp-adf as the culprit.

### Expected Behavior

32 Bit i2s should work without distortion

### Actual Behavior

The audio almost sounds identical, but there is a high pitch sound

### Steps to Reproduce

Download: [pipeline_spiffs_mp3.zip](https://github.com/user-attachments/files/17671214/pipeline_spiffs_mp3.zip)

1. Download and flash the example to the LyraT 4.3
2. Only the right channel will output
3. Use high quality headphones or speaker, or both
4. Listen to the sound, it loops a test track of s16le and s32le every 5 seconds

### Code to Reproduce This Issue
See attached zip

TempoTian, 2024-11-13:
I have tested on IDFv5.3 and not reproduce this issue. IDF4.x is end of life, could you please update IDF and have a try? If still have issue, attach a video help to check the distortion look like. Thanks!

expresspotato, 2024-11-18:
Hi @TempoTian, after updating to esp-idf v5.1.5 and using the new i2s driver the issue is is gone.

Regards,
Kevin

closed 2024-12-10 by jason-mao