espressif/esp-adf #1312: VOIP例程播放tone mp3提示音没有声音

Victccmc, 2024-11-09:
[ √] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
[ √] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
[ √] I have searched the issue tracker for a similar issue and not found a similar issue.

### 环境：
```
Audio development kit: ESP32-LyraT-v1,2
Audio kit version: v1
Module or chip used: ESP32-WROVER-B
IDF version: v5.3.1
ADF version: master
Build system: idf.py
Running log:
Operating system: Windows
Environment type: ESP Command Prompt
Using an IDE?: No
Power supply: USB
```
### 问题描述：
例程是从ADF中复制出来，没有做任何修改，编译之后下载到ESP32-LyraT-v1,2开发板，启动之后全程没有任何tone MP3的提示音，在电脑端使用SIP通话是正常，确认audio-esp.bin文件有正确稍等，运行pipeline_flash_tone例程是可以正确播放

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7524
load:0x40078000,len:16552
load:0x40080400,len:4
load:0x40080404,len:4312
entry 0x4008065c
I (29) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (29) boot: compile time Nov  8 2024 19:23:47
I (29) boot: Multicore bootloader
I (34) boot: chip revision: v1.0
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00200000
I (96) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=40134h (262452) map
I (183) esp_image: segment 1: paddr=0005015c vaddr=3ffb0000 size=042d8h ( 17112) load
I (188) esp_image: segment 2: paddr=0005443c vaddr=40080000 size=0bbdch ( 48092) load
I (204) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=fcb64h (1035108) map
I (469) esp_image: segment 4: paddr=0015cb8c vaddr=4008bbdc size=0b89ch ( 47260) load
I (496) boot: Loaded app from partition at offset 0x10000
I (496) boot: Disabling RNG early entropy source...
I (507) quad_psram: This chip is ESP32-D0WD
I (508) esp_psram: Found 8MB PSRAM device
I (508) esp_psram: Speed: 80MHz
I (512) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (519) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (528) cpu_start: Multicore app
I (1044) esp_psram: SPI SRAM memory test OK
I (1052) cpu_start: Pro cpu start user code
I (1052) cpu_start: cpu freq: 240000000 Hz
I (1052) app_init: Application information:
I (1055) app_init: Project name:     voip_app
I (1060) app_init: App version:      1
I (1065) app_init: Compile time:     Nov  9 2024 14:41:54
I (1071) app_init: ELF file SHA256:  d419f1284...
I (1076) app_init: ESP-IDF:          v5.3.1-dirty
I (1082) efuse_init: Min chip rev:     v0.0
[... 702 lines cut ...]

I (198412) VoIP_EXAMPLE: Func:app_main, Line:213, MEM Total:4157396 Bytes, Inter:239131 Bytes, Dram:204239 Bytes, Dram largest free:110592Bytes

I (214433) AUDIO_SYS: | Task              | Run Time    | Per | Prio | HWM       | State   | CoreId   | Stack 
I (214434) AUDIO_SYS: | main              | 2461        | 0%  | 1    | 1404      | Running | 0        | Intr
I (214444) AUDIO_SYS: | IDLE1             | 1015291     |49%  | 0    | 916       | Ready   | 1        | Intr
I (214461) AUDIO_SYS: | IDLE0             | 447784      |22%  | 0    | 920       | Ready   | 0        | Intr
I (214470) AUDIO_SYS: | Tmr Svc           | 1650        | 0%  | 1    | 1260      | Blocked | 7fffffff | Intr
I (214492) AUDIO_SYS: | sip_task          | 2345        | 0%  | 20   | 7928      | Blocked | 0        | Extr
I (214497) AUDIO_SYS: | tiT               | 263         | 0%  | 18   | 2216      | Blocked | 7fffffff | Intr
I (214521) AUDIO_SYS: | esp_periph        | 2449        | 0%  | 5    | 2068      | Blocked | 0        | Intr
I (214523) AUDIO_SYS: | wifi_serv         | 0           | 0%  | 6    | 628       | Blocked | 0        | Extr
I (214534) AUDIO_SYS: | DEC_mp3           | 0           | 0%  | 20   | 2688      | Blocked | 1        | Extr
I (214551) AUDIO_SYS: | Audio_forge       | 0           | 0%  | 5    | 1956      | Blocked | 1        | Extr
I (214560) AUDIO_SYS: | ipc0              | 0           | 0%  | 24   | 400       | Suspended | 0        | Intr
I (214581) AUDIO_SYS: | IN_flash          | 0           | 0%  | 4    | 844       | Blocked | 1        | Intr
I (214586) AUDIO_SYS: | OUT_iis           | 0           | 0%  | 23   | 2608      | Blocked | 1        | Extr
I (214611) AUDIO_SYS: | media_task        | 0           | 0%  | 6    | 1740      | Blocked | 0        | Extr
I (214613) AUDIO_SYS: | input_key_servi   | 0           | 0%  | 5    | 3444      | Blocked | 1        | Extr
I (214623) AUDIO_SYS: | sys_evt           | 0           | 0%  | 20   | 1544      | Blocked | 0        | Intr
I (214641) AUDIO_SYS: | wifi              | 10063       | 0%  | 23   | 3668      | Blocked | 0        | Intr
I (214649) AUDIO_SYS: | esp_timer         | 549291      |27%  | 22   | 3248      | Suspended | 0        | Intr
I (214671) AUDIO_SYS: | ipc1              | 0           | 0%  | 24   | 504       | Suspended | 1        | Intr
I (214676) AUDIO_SYS: | esp_dispatcher    | 0           | 0%  | 10   | 1420      | Blocked | 0        | Intr
```

TempoTian, 2024-11-11:
最近版本有修复可以抓下来确认下： https://github.com/espressif/esp-adf/commit/4971b27f7feb35263e70da92a5692d74a3e81748

Victccmc, 2024-11-11:
> 最近版本有修复可以抓下来确认下： [4971b27](https://github.com/espressif/esp-adf/commit/4971b27f7feb35263e70da92a5692d74a3e81748)

更新最新的Master已经可以播放了，谢谢！

closed 2024-11-11 by Victccmc