espressif/esp-adf #1314: VOIP 例程，sip 多次注册过程，有没有接口或者方法固定端口号？

ktoto2011, 2024-11-10:
[ √] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
[ √] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
[ √] I have searched the issue tracker for a similar issue and not found a similar issue.

Environment
Audio development kit: ESP32-S3-Korvo-2
Audio kit version (ESP32-S3-Korvo-2): v3
[Required] Module or chip used: ESP32-S3-WROOM-1
[Required] IDF version v4.47
[Required] ADF version v2.6-183-gce64fea5
Build system: [idf.py]
[Required] Running log: All logs from power-on to problem recurrence
Operating system: [Windows]
(Windows only) Environment type: [ESP Command Prompt]
Using an IDE?: vscode
Power supply: USB

Problem Description
sip在运行过程中，由于网络波动以及配置的注册保护时长，存在多次注册的情况，目前固件每次注册的使用的端口号都是随机的，这样造成SIP 服务器对上一次端口的连接清理掉（如果此时在通话过程中，服务端无法获知这条旧端口链接通话的状态），通话会有问题。

ktoto2011, 2024-11-10:
[image]

sip 每次注册端口号都是随机变化的

ktoto2011, 2024-11-10:
[image]

这个抓包是，设备端更换端口号重新注册sip,但是它上一个链接（端口号56099）已经建立通话了，此时服务端往旧端口56099给的bye应答，设备端没办法回复。

TempoTian, 2024-11-11:
Register每次注册使用随机的port是出于安全考虑，在RTP连接的过程中重新register对RTP没啥影响，不过register之后服务器还用之前的port发送BYE应该是server有问题，用了缓存的port，我一会改一个版本让在RTP连接的过程中如果发送过期重新Register的动作就用之前的port，你再试试看有没有改善。

TempoTian, 2024-11-11:
试一下attach的库看看， 库本身支持过期之后自动register的之前代码有bug应该导致RTP连接之后不能register了。
[libesp_media_protocols.zip](https://github.com/user-attachments/files/17698934/libesp_media_protocols1.zip)

ktoto2011, 2024-11-11:
> 试一下attach的库看看， 库本身支持过期之后自动register的之前代码有bug应该导致RTP连接之后不能register了。 [libesp_media_protocols.zip](https://github.com/user-attachments/files/17698934/libesp_media_protocols1.zip)

好的，非常感谢您及时回复。我这边先测试下

ktoto2011, 2024-11-11:
> Register每次注册使用随机的port是出于安全考虑，在RTP连接的过程中重新register对RTP没啥影响，不过register之后服务器还用之前的port发送BYE应该是server有问题，用了缓存的port，我一会改一个版本让在RTP连接的过程中如果发送过期重新Register的动作就用之前的port，你再试试看有没有改善。

您好，请问一下这边实现的是在esp32 上面吗，我这边的硬件是esp32s3(ESP32-S3-Korvo-2),idf 是4.4.7  adf 2.7。现在覆盖文件到C:\Espressif\frameworks\esp-adf-v2.7\components\esp-adf-libs\esp_media_protocols\lib\esp32s3，编译报错。

ktoto2011, 2024-11-11:
或者您告知你这边的编译环境版本是那个版本。我试试

zenggb7711, 2024-11-12:
> Register每次注册使用随机的port是出于安全考虑，在RTP连接的过程中重新register对RTP没啥影响，不过register之后服务器还用之前的port发送BYE应该是server有问题，用了缓存的port，我一会改一个版本让在RTP连接的过程中如果发送过期重新Register的动作就用之前的port，你再试试看有没有改善。

另外还有一个问题，如果在有电话打进来INCOMING，还没有建立通话RTP连接时，在这个时间点重新register会使用新的端口号，这个时候是不能正常接听的。
是不是可以在这个过程中，如果发送过期重新Register的动作就用之前的port，保证整个呼叫可以正常建立RTP连接。

zenggb7711, 2024-11-12:
> Register每次注册使用随机的port是出于安全考虑，在RTP连接的过程中重新register对RTP没啥影响，不过register之后服务器还用之前的port发送BYE应该是server有问题，用了缓存的port，我一会改一个版本让在RTP连接的过程中如果发送过期重新Register的动作就用之前的port，你再试试看有没有改善。

我们是不是可以每次开机 Register注册使用随机的port，后面在设备运行过程中，在待机、被呼、通话状态下 都沿用这个port。

TempoTian, 2024-11-12:
目前我的改动是只有RTP 建立成功的时候，过期重新register才会复用port 否则用random的port。
你测试的case应该达到corner case了，我可以加一下，如果状态停留在register状态，就用random的port。应该不需要待机之类的用老的port。

zenggb7711, 2024-11-12:
> 目前我的改动是只有RTP 建立成功的时候，过期重新register才会复用port 否则用random的port。 你测试的case应该达到corner case了，我可以加一下，如果状态停留在register状态，就用random的port。应该不需要待机之类的用老的port。另外上面的改动在通话中的重新register测试应该OK了吧？

RTP连接，通话中重新register是沿用了老的port

TempoTian, 2024-11-12:
把expire改小用我提供的库抓个log我看下呢，我一会编译一个inviting之后bye之前重新register复用port的库你再确认下看。

zenggb7711, 2024-11-12:
> 把expire改小用我提供的库抓个log我看下呢，我一会编译一个inviting之后bye之前重新register复用port的库你再确认下看。

现在expire的时间是300s，大概150s会重新register，需要改到多少？

zenggb7711, 2024-11-12:
> 把expire改小用我提供的库抓个log我看下呢，我一会编译一个inviting之后bye之前重新register复用port的库你再确认下看。

需要在 发送或接收 到inviting之后，bye 或者 canceling 之前 都沿用之前的port

TempoTian, 2024-11-12:
用附件的lib测试下看：
[libesp_media_protocols.zip](https://github.com/user-attachments/files/17710984/libesp_media_protocols.zip)
log关键词：
[image]
[image]

zenggb7711, 2024-11-12:
> > 把expire改小用我提供的库抓个log我看下呢，我一会编译一个inviting之后bye之前重新register复用port的库你再确认下看。
> 
> 现在expire的时间是300s，大概150s会重新register，需要改到多少？

附件是expire改为30秒的日志

81089  收到invite报文
INVITE sip:2024925927068364@223.74.148.250:10128;transport=TCP SIP/2.0

I (81266) SIP: 开始振铃
SIP/2.0 180 Ringing

I (84061) SIP:  重新注册
REGISTER sip:cu.sip.t7n.cn:5070 SIP/2.0

I (84484) MP3_DECODER: Closed   振铃音停止
话机无法正常接听
[日志.txt](https://github.com/user-attachments/files/17711132/default.txt)

zenggb7711, 2024-11-12:
> 用附件的lib测试下看：
> 
> log关键词： ![image](https://private-user-images.githubusercontent.com/110106787/385136105-3c187f0e-004d-455d-b22d-d7075ba5ee53.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzOTA4ODYsIm5iZiI6MTczMTM5MDU4NiwicGF0aCI6Ii8xMTAxMDY3ODcvMzg1MTM2MTA1LTNjMTg3ZjBlLTAwNGQtNDU1ZC1iMjJkLWQ3MDc1YmE1ZWU1My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMlQwNTQ5NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MTk3MDc3NWI5ZTViMDMwNmNlNzcyNjBjZDkxNGU3NTA5NjE4NmY2NjUzYzNhOGVlMzAwNDhiZTYzYWVhZDFkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Xjq8FfFJqPqWrY0Qu7DjP_zo5Pmhrt2cTK-ilCe2Xls) ![image](https://private-user-images.githubusercontent.com/110106787/385136217-ddd07386-2f8c-43b3-8bfd-322117f212a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzOTA4ODYsIm5iZiI6MTczMTM5MDU4NiwicGF0aCI6Ii8xMTAxMDY3ODcvMzg1MTM2MjE3LWRkZDA3Mzg2LTJmOGMtNDNiMy04YmZkLTMyMjExN2YyMTJhMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMlQwNTQ5NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZjEwZjg2ODZlMjIzNWQ0NzIzZWRhNjU2OGJlMDNmNDBlMzAzMzU1OTE2MjZjNzNkM2I0OWNjZTczZjZiOThhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.yXgvKuuQxk5VDyrI2z2NCT6fKLrGofWbDdo1BL82OxY)

附件是expire改为120秒的日志
重新注册是沿用了原来的port
但收到重新register报文后，振铃音停止了，也没有办法接听

I (1119059) SIP: 收的invite报文
INVITE sip:2024925927068364@223.74.148.250:10007;transport=TCP SIP/2.0

I (1119238) SIP:
SIP/2.0 180 Ringing

W (1157378) SIP: Expire on state 16

I (1158585) SIP:
REGISTER sip:cu.sip.t7n.cn:5070 SIP/2.0

W (1158934) SIP: Reuse last sock cleared...................

I (1158992) MP3_DECODER: Closed
话机无法正常接听
[日志文件11121420.zip](https://github.com/user-attachments/files/17712093/11121420.zip)

zenggb7711, 2024-11-12:
> 用附件的lib测试下看：
> log关键词： [...]

为啥重新发register报文后，自动停止播放振铃音了，通话中断了

zenggb7711, 2024-11-12:
> 为啥重新发register报文后，自动停止播放振铃音了，通话中断了

这是抓包截图
[image]

zenggb7711, 2024-11-12:
我们可以按照下面方式处理吗？
出于安全考虑，设备Register每次注册使用随机的port
设备主叫外呼时，发送invite报文 到 发出和接收到 cancel 或者 bye 报文的时候，重新register的port不发生改变
设备被叫接听时，收到invite报文 到发出和接收到 cancel 或者 bye 报文的时候，重新register的port不发生改变
这样设备在主叫 和 被叫，接通或振铃的时候，不会影响正常工作
[image]

TempoTian, 2024-11-12:
Update lib. 我本地测试了下本地挂断是OK的，可以再确认下看。
[Uploading libesp_media_protocols.zip…]()

zenggb7711, 2024-11-13:
> Update lib. 我本地测试了下本地挂断是OK的，可以再确认下看。 [libesp_media_protocols.zip](https://github.com/user-attachments/files/17717038/libesp_media_protocols.zip)

感谢！
新的库文件，我们测试还一个问题。
设备作为主叫外呼，振铃60s没有接听，SIP服务端会自动结束通话。
设备在收到 480Temporarily Unavailable 或  408Request Timeout 报文时，设备会报ESP_RTC_EVENT_ERROR
也没有回复平台ack报文，设备没有正常结束通话，设备再次不能再次外呼。
其他终端呼叫本设备，提示占线正忙

测试1的log
I (5663760) SIP:
INVITE sip:1001@cu.sip.t7n.cn:5070 SIP/2.0

I (5664366) SIP:
SIP/2.0 180 Ringing

I (5724078) SIP:
SIP/2.0 408 Request Timeout

I (5724128) SIP_SERVICE: ESP_RTC_EVENT_ERROR

测试2的log
I (1856076) SIP:
INVITE sip:1001@cu.sip.t7n.cn:5070 SIP/2.0

I (1856702) SIP:
SIP/2.0 180 Ringing

I (1921004) SIP:
SIP/2.0 480 Temporarily Unavailable

I (1921070) SIP: [1970-01-01/00:32:00]<<======================
E (1921075) SIP: CLEAR at 2025 0x3d9a8cd0
I (1921079) SIP_SERVICE: ESP_RTC_EVENT_ERROR
W (1921084) SIP: Not current session method!

[Uploading 主叫振铃超时log.zip…]()

zenggb7711, 2024-11-14:
> 上面提到的两个event没有做特殊处理，所以期望用户可以当作error然后调用bye挂断，我做了一个临时的版本加入上面两个的处理，然后回ACK你可以测试下看效果 [libesp_media_protocols.zip](https://github.com/user-attachments/files/17728194/libesp_media_protocols.zip)

刚刚测试了 在收到408报文没有回复ACK报文
我们是不是可以跟收到404报文一样，结束整个呼叫过程。
在收到404报文后，回复AC报文
I (58927) SIP_SERVICE: ESP_RTC_EVENT_ERROR
I (58940) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_END

以下是log

I (221469) SIP: [1970-01-01/00:03:20]<<=====READ 0393 bytes==
I (221470) SIP:

SIP/2.0 408 Request Timeout
Via: SIP/2.0/TCP 192.168.8.160:53878;received=223.74.148.250;rport=10075;branch=z9hG4bK--1358136718
From: <sip:2024925927068364@cu.sip.t7n.cn:5070>;tag=15570795
To: <sip:1001@cu.sip.t7n.cn:5070>;tag=49e0d6846fdff19c573369bc3abdb93b-5d82
Call-ID: A793C27CF0112B078B6194F8988169CEF91F6DADE81F
CSeq: 13 INVITE
Server: Eos SIP Policy Server
Content-Length: 0

I (221503) SIP: [1970-01-01/00:03:20]<<======================
E (221510) SIP: CLEAR at 2029 0x3d9d0364
I (221514) SIP_SERVICE: ESP_RTC_EVENT_ERROR
W (221519) SIP: Not current session method!

以下是收到404 报文的log

SIP/2.0 404 Not Found
Via: SIP/2.0/TCP 192.168.8.160:58551;received=223.74.148.250;rport=10061;branch=z9hG4bK-1158558672
Max-Forwards: 67
From: <sip:2024925927068364@cu.sip.t7n.cn:5070>;tag=-1269594709
To: <sip:1001@cu.sip.t7n.cn:5070>;tag=F5K1y0jN9jQ1H
Call-ID: 728DF2605015AF2774E1AA1E5674EB980904455073F4
CSeq: 7 INVITE
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, REGISTER, REFER, NOTIFY
Supported: path, replaces
Allow-Events: talk, hold, conference, refer
Reason: Q.850;cause=16;text="NORMAL_CLEARING"
Content-Length: 0
Remote-Party-ID: "1001" <sip:1001@cu.sip.t7n.cn>;party=calling;privacy=off;screen=no
User-Agent: YunEasy

I (58926) SIP: [1970-01-01/00:00:28]<<======================
E (58927) SIP: CLEAR at 2029 0x3d9db2f4
I (58927) SIP_SERVICE: ESP_RTC_EVENT_ERROR
W (58932) SIP: CHANGE STATE FROM 72, TO 2, :func: _sip_uac_process_res:1742
I (58940) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_END
I (58954) ALGORITHM_STREAM: _algo_fetch_task is stopped
W (58954) AUDIO_PIPELINE: There are no listener registered
I (58957) AUDIO_PIPELINE: audio_pipeline_unlinked
I (58964) AV_STREAM: _audio_enc task stoped
I (58964) AFE_VC: afe_se_task quit

I (58972) AFE_VC: afe task destroy finished

I (58980) AFE_VC: exit function: afe_destory

I (58983) AV_STREAM: _audio_dec task stoped
I (58988) SIP: [1970-01-01/00:00:28]=======WRITE 0666 bytes>>
I (58993) SIP:

ACK sip:1001@cu.sip.t7n.cn:5070 SIP/2.0
Via: SIP/2.0/TCP 192.168.8.160:58551;branch=z9hG4bK-1158558672;rport
From: <sip:2024925927068364@cu.sip.t7n.cn:5070>;tag=-1269594709
To: <sip:1001@cu.sip.t7n.cn:5070>;tag=F5K1y0jN9jQ1H
Contact: <sip:2024925927068364@192.168.8.160:58551;transport=TCP>
Max-Forwards: 70
Call-ID: 728DF2605015AF2774E1AA1E5674EB980904455073F4
CSeq: 7 ACK
Expires: 100
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Authorization: Digest username="2024925927068364", realm="cu.sip.t7n.cn", nonce="673540200000d43f113505f99c55786f923dbb34ca4a03de", uri="sip:cu.sip.t7n.cn:5070", response="7d744798812a2480c5a8c4046c1c6808", algorithm=MD5

I (59055) SIP: [1970-01-01/00:00:29]=======================>>

TempoTian, 2024-11-14:
详细的log发我看下，目前代码已经合并处理了。只是W (221519) SIP: Not current session method! 说明seqno不匹配，要看下log看是否server回的seqno是否不匹配

zenggb7711, 2024-11-14:
> 详细的log发我看下，目前代码已经合并处理了。只是W (221519) SIP: Not current session method! 说明seqno不匹配，要看下log看是否server回的seqno是否不匹配

I (1242259) SIP:
SIP/2.0 408 Request Timeout

[主叫振铃超时3.txt](https://github.com/user-attachments/files/17741758/3.txt)

TempoTian, 2024-11-14:
这个你调的register的expire时间过短导致的，容易在invite的中间触发，可以调大试试看。我再Enhance下这部分的逻辑。

zenggb7711, 2024-11-14:
> 这个你调的register的expire时间过短导致的，容易在invite的中间触发，可以调大试试看。我再Enhance下这部分的逻辑。

我们设置这个expire时间断时间，是为了网络的keepalive，是有些IP PBX不回复 options报文。
我先修改expire时间看看

TempoTian, 2024-11-14:
目前的逻辑也是这样的，option参数用keepalive的设定来配置， register是在expire时间达到一半的时候触发。register expire改大，keepalive设定上就可以解决你现在的问题应该

zenggb7711, 2024-11-14:
> 目前的逻辑也是这样的，option参数用keepalive的设定来配置， register是在expire时间达到一半的时候触发。register expire改大，keepalive设定上就可以解决你现在的问题应该

在振铃过程中还会有发送register报文的可能性，改大expire的值只会降低概率
我看其他IP话机包括华为和方位的，在振铃中也会发送register报文，会正常回复ACK报文。

TempoTian, 2024-11-14:
是的，正常配置register timeout为1个小时碰到的概率应该会小很多。可以用附件的lib再确认下看是否OK了。
[libesp_media_protocols.zip](https://github.com/user-attachments/files/17744015/libesp_media_protocols.zip)

zenggb7711, 2024-11-15:
> 是的，正常配置register timeout为1个小时碰到的概率应该会小很多。可以用附件的lib再确认下看是否OK了。 [libesp_media_protocols.zip](https://github.com/user-attachments/files/17744015/libesp_media_protocols.zip)

这个版本已经可以正常回复ACK报文了，我们再多测试一下，有问题再请教，多谢！

ktoto2011, 2024-11-16:
> 是的，正常配置register timeout为1个小时碰到的概率应该会小很多。可以用附件的lib再确认下看是否OK了。 [libesp_media_protocols.zip](https://github.com/user-attachments/files/17744015/libesp_media_protocols.zip)

目前有小概率出现设备主呼后，对方接听，设备端收不到 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 状态。

[接听设备未收到ESP_RTC_EVENT_AUDIO_SESSION_BEGIN .txt](https://github.com/user-attachments/files/17784223/ESP_RTC_EVENT_AUDIO_SESSION_BEGIN.txt)

ktoto2011, 2024-11-16:
这是正常的主呼，对方接听，设备收到的状态；
I (412216) SIP: [2024-11-16/17:14:24]<<======================
E (441824) VoIP_EXAMPLE: [ * ] [PLAY] input key event     （设备拨号）

SIP/2.0 180 Ringing
SIP/2.0 183 Session Progress

I (460805) SIP: [2024-11-16/17:15:18]<<======================
E (460812) SIP: CLEAR at 2033 0x3d9c28b0
I (460816) SIP: Remote ARTP port=45886
[正常的接听流程.txt](https://github.com/user-attachments/files/17784226/default.txt)

I (460820) SIP: Remote RTP addr=220.196.190.61
I (460826) **SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_BEGIN**    （对方接听，设备收到会话开始）

I (463980) SIP: [2024-11-16/17:15:20]<<======================
E (463997) SIP: CLEAR at 2033 0x3d9b25b4
I (463997) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_END

I (464069) SIP_SERVICE: ESP_RTC_EVENT_HANGUP

zenggb7711, 2024-11-17:
> > 是的，正常配置register timeout为1个小时碰到的概率应该会小很多。可以用附件的lib再确认下看是否OK了。 [libesp_media_protocols.zip](https://github.com/user-attachments/files/17744015/libesp_media_protocols.zip)
> 
> 目前有小概率出现设备主呼后，对方接听，设备端收不到 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 状态。
> 
> E (3104484) VoIP_EXAMPLE: [ * ] [PLAY] input key event （设备拨号）
> 
> SIP/2.0 180 Ringing
> 
> I (3108663) SIP: [2024-11-16/16:57:20]<<====================== E (3108664) SIP: CLEAR at 2033 0x3d9b3ea8 W (3108668) SIP: CHANGE STATE FROM 68, TO 68, :func: _sip_uac_process_res:1753 I (3108676) SIP: [2024-11-16/16:57:20]<<=====READ 0920 bytes== I (3108682) SIP:
> 
> SIP/2.0 183 Session Progress
> 
> I (3134728) AFE_VC: afe_se_task quit （未正常收到AFE_VC: afe interface for voice communication)
> 
> I (3134747) SIP_SERVICE: ESP_RTC_EVENT_HANGUP
> 
> [接听设备未收到ESP_RTC_EVENT_AUDIO_SESSION_BEGIN .txt](https://github.com/user-attachments/files/17784250/ESP_RTC_EVENT_AUDIO_SESSION_BEGIN.txt)

帮忙看看底层库文件是基于什么逻辑产生 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 这个event
我们是在收到 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 这个event后，停止播放本地回铃音
这个测试过程中，双方通话是正常的，但是本地回铃音一直在播放。

zenggb7711, 2024-11-17:
> 目前有小概率出现设备主呼后，对方接听，设备端收不到 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 状态。
>
> E (3104484) VoIP_EXAMPLE: [ * ] [PLAY] input key event （设备拨号）
>
> SIP/2.0 180 Ringing
>
> I (3108663) SIP: [...]

帮忙看看底层库文件是基于什么逻辑产生 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 这个event
我们是在收到 ESP_RTC_EVENT_AUDIO_SESSION_BEGIN 这个event后，停止播放本地回铃音
这个测试过程中，双方通话是正常的，但是本地回铃音一直在播放。

TempoTian, 2024-11-18:
Log可以提供全一点的，我看前面register fail了导致的，这里没有stop RTP 的操作导致后续重启起来跳过了。你用下面修改的版本再确认下看：
[libesp_media_protocols.zip](https://github.com/user-attachments/files/17794192/libesp_media_protocols.zip)

ktoto2011, 2024-11-22:
> Log可以提供全一点的，我看前面register fail了导致的，这里没有stop RTP 的操作导致后续重启起来跳过了。你用下面修改的版本再确认下看： [libesp_media_protocols.zip](https://github.com/user-attachments/files/17794192/libesp_media_protocols.zip)

目前还是会出现通话无法挂断的现场。

ktoto2011, 2024-11-22:
上传的日志重现了此现象。设备第一次主呼出去，等待超时，对方不接听；然后设备再主呼出去，对方接听，然后设备按挂断键，RTP 无法正常挂断；

E (192486) VoIP_EXAMPLE: [ * ] [PLAY] input key event      （设备第一次主呼出去）

I (260145) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_END  (不接听，等待超时)

E (260147) VoIP_EXAMPLE: [ * ] [PLAY] input key event           （设备第二次主呼出去）

I (263320) AFE_VC: afe interface for voice communication      （ 对方接听，rtp过程中)

E (267216) VoIP_EXAMPLE: [ * ] [PLAY] input key eventv          (设备按下挂机键...但是后面没有正常结束rtp)

这是不是在第一次等待超时后，太快按下拨打键造成sip通话过程的异常

[1122.txt](https://github.com/user-attachments/files/17853743/1122.txt)

TempoTian, 2024-11-22:
目前从log看，协议层已经没啥问题，这个比较像是应用层的问题。我再细看下

zenggb7711, 2024-12-22:
> Log可以提供全一点的，我看前面register fail了导致的，这里没有stop RTP 的操作导致后续重启起来跳过了。你用下面修改的版本再确认下看： [libesp_media_protocols.zip](https://github.com/user-attachments/files/17794192/libesp_media_protocols.zip)

这个新的协议层lib可以在ESP32模组上使用吗？

TempoTian, 2024-12-23:
这个是在S3上面编译的，不能在32上用，我在准备合并这个修改了，如果需要的话可以帮你编译一个32的版本

zenggb7711, 2024-12-23:
> 这个是在S3上面编译的，不能在32上用，我在准备合并这个修改了，如果需要的话可以帮你编译一个32的版本

帮忙编译一个ESP32版本吧，我们准备在 ESP32-LyraT-Mini开发版上同步开发，多谢

TempoTian, 2024-12-26:
esp32的版本如下：
[libesp_media_protocols_esp32.zip](https://github.com/user-attachments/files/18250908/libesp_media_protocols_esp32.zip)

zenggb7711, 2025-02-01:
> Register每次注册使用随机的port是出于安全考虑，在RTP连接的过程中重新register对RTP没啥影响，不过register之后服务器还用之前的port发送BYE应该是server有问题，用了缓存的port，我一会改一个版本让在RTP连接的过程中如果发送过期重新Register的动作就用之前的port，你再试试看有没有改善。

新年好！如果不考虑安全性，是否可以每次register都固定1个port，例如15060

zenggb7711, 2025-02-06:
> 可以的，如果有这个需求的话可能加一个类似 `fixed_port` 的配置，不过这样的话用户要保证这个端口没有被占用才行。

帮忙各出一个 ESP32和S3的版本吧，多谢！
设备在register时使用`fixed_port` 的配置，固定源port。

TempoTian, 2025-02-06:
用这个版本测试下看[fixed_local_port.zip](https://github.com/user-attachments/files/18688503/fixed_local_port.zip)
加了一下下面的字段
```c
 int                         fixed_local_port;    /*!< Set fixed local port (defaults is 0) */
```

zenggb7711, 2025-02-07:
> 用这个版本测试下看[fixed_local_port.zip](https://github.com/user-attachments/files/18688503/fixed_local_port.zip) 加了一下下面的字段
> 
>  int                         fixed_local_port;    /*!< Set fixed local port (defaults is 0) */

感谢

ktoto2011, 2025-02-13:
> 用这个版本测试下看[fixed_local_port.zip](https://github.com/user-attachments/files/18688503/fixed_local_port.zip) 加了一下下面的字段
> 
>  int                         fixed_local_port;    /*!< Set fixed local port (defaults is 0) */

测试发现端口并没有固定，还是随机的

ktoto2011, 2025-02-13:
[I (9335) wifinew12,0, old12,0, ap25.txt](https://github.com/user-attachments/files/18778440/I.9335.wifinew12.0.old12.0.ap25.txt)

[image]

[image]

ktoto2011, 2025-02-13:
@TempoTian 您好，有空帮忙看下

TempoTian, 2025-02-13:
TCP的也需要吗？目前我加的是UDP的bind到固定端口的操作, TCP bind用的比较少

ktoto2011, 2025-02-13:
> TCP的也需要吗？目前我加的是UDP的bind到固定端口的操作, TCP bind用的比较少

目前我们使用的TCP 的情况比较多，可以帮忙把tcp的注册方式加上吗？

TempoTian, 2025-02-17:
稍等下我来加一下测试下

TempoTian, 2025-02-17:
我测试了下TCP bind不大好用，因为有TIMEWAIT状态，如果close之后马上重新bind connect的话会导致连接失败。TIMEWAIT有比较长的超时时间，所以我不大建议给TCP也加fix port的功能。
```
When this client closes its connection, the TCP stack will typically place the connection in the TIME_WAIT state. Even with SO_REUSEADDR set, if you try to run code (or create a new connection using the same fixed local port) immediately after closing, the bind() call may fail with "Address already in use" because the port is still held in TIME_WAIT. This is normal TCP behavior. 
```

ktoto2011, 2025-02-17:
> 我测试了下TCP bind不大好用，因为有TIMEWAIT状态，如果close之后马上重新bind connect的话会导致连接失败。TIMEWAIT有比较长的超时时间，所以我不大建议给TCP也加fix port的功能。
>
> ```
> When this client closes its connection, the TCP stack will typically place the [...]

好的，感谢您的帮忙，我们看下采用udp的方式吧，这边在测试udp的方式看下还有其他问题没有

robhelvestine, 2025-11-08:
@TempoTian 

Thank you for your help with this Issue. I have also encountered a similar problem on my ESP32S3 system - the port number seems to be random for each registration. Does the .a file you provided on November 17th 2024 fix this issue? Is there a more up to date binary to use?

TempoTian, 2025-11-12:
@swoisz Currently, there is a timeout mechanism that periodically sends OPTIONS or performs re-registration to keep the port active. However, due to issues such as network interruptions, re-registration may fail. In that case, a reconnection is required (closing the old socket and opening a new one).
For TCP, the same local port cannot be reused immediately after closing, so the port cannot be fixed.
For UDP, this issue does not exist because the port can be reused immediately.

TempoTian, 2025-11-13:
@swoisz If there is no refresh use option if provided, maybe it is a bug, we will look into it to double check. Thanks for your info.