espressif/esp-adf #1315: ESP-RTSP Streaming Cilent/Server

ManikandanVaradharaj, 2024-11-13:
**Is your feature request related to a problem? Please describe.**

No,It's not an problem.I'm checking this feature availability for my requirement.

**Describe the solution you'd like**

I'm using Esp32-p4 ev board in that can i add the features for esp-rtsp as a client.with that if i set the ip address it should connect the live stream in rtsp..the streaming will comes form different module with rtsp.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

TempoTian, 2024-11-13:
Yes, you can take the example as a reference: https://github.com/espressif/esp-adf/blob/master/examples/protocols/esp-rtsp/main/rtsp_service.c ignore the server related code.

ManikandanVaradharaj, 2024-11-13:
How can i make an complete app structure for work in esp32-p4 board..whether still ui,assets will be needed

TempoTian, 2024-11-13:
If you need UI support for P4 you can select LVGL, there are also esp_ui https://components.espressif.com/components/espressif/esp-ui/versions/0.1.1~1?language=zh can help to setup the UI related things, currently we have not add UI in the example code, just have code to demo RTSP usage for both server and client.

ManikandanVaradharaj, 2024-11-13:
Orelse any other idea to integrate rtsp in esp32-p4..rather than ui side

TempoTian, 2024-11-13:
Without UI, RTSP can run in service mode that need no UI, you get your stream data then send out through RTSP api, it just like the example do.

ManikandanVaradharaj, 2024-11-13:
Here i need to set the wifi connection first after that set ip address in esp32-p4 ev board,it should connect the server..for this requirement how can i run in the display without UI intreacts.

jimforr, 2025-08-21:
@ManikandanVaradharaj 
you can use cmd: join <ssid> <pass>