espressif/esp-adf #1318: PAUSE WORK BUT RESUME ERROR

hoanganh1911, 2024-11-15:


hoanganh1911, 2024-11-15:
I am currently having the following error, when I press the button, my circuit will pause, but when I press it again, the following error message appears, I think it will resume, I developed from the original example play_sdcard_mp3_control. Thank you for your help.[image]

JokerChow, 2024-12-05:
having same issue. keep watching.

ShaoMingya, 2024-12-18:
应该是版本问题，之前可以。我最近更新了一下，发现这个问题了。

jason-mao, 2024-12-19:
Thanks for your report, let's check and fix it.

ShaoMingya, 2024-12-19:
好像是 adf 此次提交后的问题：6d9c46cccd884df478db3a6e15b85e91359c76b7
切换到d16b0401cc805e184b132aa6c821eccd26163f61后正常。
`git checkout d16b0401cc805e184b132aa6c821eccd26163f61`
`git submodule update --init --recursive`

JokerChow, 2024-12-23:
[ShaoMingya](https://github.com/ShaoMingya), 

Thank you, it works now.

majingjing123, 2024-12-26:
Already fix this bug. You can checkout ADF to commit id: 93a56fcddcabe3be856df547fbc2db566ee2f740.

closed 2024-12-26 by espressif-bot