espressif/esp-adf #1319: rtsp client is not working properly

alpha-alan, 2024-11-19:
## Environment

- Module or chip used:  ESP32-S3-WROOM-1
- IDF version v5.3
- ADF version v2.7
- Running log:         

```
start 2 rtsp://110.1.1.189:8554/video/t1g711.wav
I (63071187) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Aug 15 2024-04:10:02
I (63071189) ESP_RTSP_CLIENT: Conecting...
I (63071189) ESP_RTSP_CLIENT: RTSP Client URL:rtsp://110.1.1.189:8554/video/dl.wav
demo-chime> I (63071252) ESP_RTSP_CLIENT: TX:

OPTIONS rtsp://110.1.1.189:8554/video/dl.wav RTSP/1.0
CSeq: 0
User-Agent: ESP32-RTSP-client 1.1.0

I (63071275) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 0
Date: Fri, Nov 15 2024 02:52:33 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

I (63071288) ESP_RTSP_CLIENT: TX:

DESCRIBE rtsp://110.1.1.189:8554/video/dl.wav RTSP/1.0
CSeq: 1
User-Agent: ESP32-RTSP-client 1.1.0
Accept: application/sdp

I (63071370) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 1
Date: Fri, Nov 15 2024 02:52:33 GMT
Content-Base: rtsp://110.1.1.189:8554/video/dl.wav/
Content-Type: application/sdp
Content-Length: 398

v=0
o=- 1731639153368084 1 IN IP4 110.1.1.189
s=WAV Audio Stream, streamed by the LIVE555 Media Server
i=video/dl.wav
t=0 0
a=tool:LIVE555 Streaming Media v2024.10.31
a=type:broadcast
a=control:*
a=range:npt=0-332.801
a=x-qt-text-nam:WAV Audio Stream, streamed by the LIVE555 Media Server
a=x-qt-text-inf:video/dl.wav
m=audio 0 RTP/AVP 10
c=IN IP4 0.0.0.0
b=AS:1411
a=control:track1

I (63071418) ESP_RTSP_CLIENT: TX:

SETUP rtsp://110.1.1.189:8554/video/dl.wav/trackID=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=30950-30951;mode=record
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0
Session: 3609990

I (63071469) ESP_RTSP_CLIENT: RX:

RTSP/1.0 454 Session Not Found
CSeq: 2
Date: Fri, Nov 15 2024 02:52:33 GMT

I (63071471) ESP_RTSP_CLIENT: TX:

PLAY rtsp://110.1.1.189:8554/video/dl.wav RTSP/1.0
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0

I (63071579) ESP_RTSP_CLIENT: RX:

RTSP/1.0 454 Session Not Found
CSeq: 2
Date: Fri, Nov 15 2024 02:52:33 GMT

I (63071581) ESP_RTSP_CLIENT: TX:

TEARDOWN rtsp://110.1.1.189:8554/video/dl.wav RTSP/1.0
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0
Session: 3609990

I (63071605) RTSP_SERVICE: RTSP_STATE_TEARDOWN
I (63071605) ESP_RTSP_CLIENT: RTSP client task stoped
```

## Problem Description
I use esp-rtsp in the example folder as the RTSP client and live555 as the server. One thing I discovered is that the SETUP sent by esp-rtsp to the server contains a session field, but the session is not found on the server side, so a “session not found” error is reported.

According to my understanding, generally in RTSP requests, the session field should not be included in SETUP requests. Why is there a session field here? How can I solve this "session not found" problem

### Code to Reproduce This Issue
esp-adf\examples\protocols\esp-rtsp

TempoTian, 2024-11-20:
I have just fixed the issue, you can verify use attach lib and replace into esp-adf-libs folder.
[esp_media_protocols.zip](https://github.com/user-attachments/files/17824418/esp_media_protocols.zip)

ManikandanVaradharaj, 2024-11-20:
Can i use the rtsp client protocol in esp32-p4 module

alpha-alan, 2024-11-20:
@TempoTian 

The new library solves the problem of SETUP carrying session. But there are still 2 problems, which makes esp-rtsp still unable to connect to a rtsp server.
Question 1:
When I use live555 as the server, I receive "404 Stream Not Found"

I think this is caused by the url suffix in SETUP

The url sent by esp-rtsp is:
`SETUP rtsp://110.1.1.189:8554/video/t1g711.wav/trackID=0 RTSP/1.0`

For live555, it does not recognize the suffix "trackID=0", and the url it expects to receive is
`SETUP rtsp://110.1.1.189:8554/video/t1g711.wav/track1 RTSP/1.0`
Here "track1" comes from sdp
`a=control:track1`

Considering that live555 is a widely used rtsp server, I hope esp-rtsp be compatible with its behavior.
All log
```
demo-chime> start 2 rtsp://110.1.1.189:8554/video/t1g711.wav
I (2566269) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Nov 20 2024-11:21:23
I (2566271) ESP_RTSP_CLIENT: Conecting...
I (2566272) ESP_RTSP_CLIENT: RTSP Client URL:rtsp://110.1.1.189:8554/video/t1g711.wav
I (2566279) ESP_RTSP_CLIENT: TX:

OPTIONS rtsp://110.1.1.189:8554/video/t1g711.wav RTSP/1.0
CSeq: 0
User-Agent: ESP32-RTSP-client 1.1.0

I (2566353) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 0
Date: Wed, Nov 20 2024 07:59:53 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

I (2566366) ESP_RTSP_CLIENT: TX:

DESCRIBE rtsp://110.1.1.189:8554/video/t1g711.wav RTSP/1.0
CSeq: 1
User-Agent: ESP32-RTSP-client 1.1.0
Accept: application/sdp

I (2566388) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 1
Date: Wed, Nov 20 2024 07:59:53 GMT
Content-Base: rtsp://110.1.1.189:8554/video/t1g711.wav/
Content-Type: application/sdp
Content-Length: 402

v=0
o=- 1732089593547888 1 IN IP4 110.1.1.189
s=WAV Audio Stream, streamed by the LIVE555 Media Server

t=0 0
a=tool:LIVE555 Streaming Media v2024.10.31
a=type:broadcast
a=control:*
a=range:npt=0-29.983
a=x-qt-text-nam:WAV Audio Stream, streamed by the LIVE555 Media Server
a=x-qt-text-inf:video/t1g711.wav
m=audio 0 RTP/AVP 8
c=IN IP4 0.0.0.0
b=AS:64
a=control:track1

I (2566436) RTSP_SERVICE: RTSP_STATE_SETUP
I (2566446) ESP_RTSP_CLIENT: TX:

SETUP rtsp://110.1.1.189:8554/video/t1g711.wav/trackID=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=39923-39924;mode=record
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0

demo-chime> I (2566486) ESP_RTSP_CLIENT: RX:

RTSP/1.0 404 Stream Not Found
CSeq: 2
Date: Wed, Nov 20 2024 07:59:53 GMT

I (2566489) RTSP_SERVICE: RTSP_STATE_TEARDOWN
I (2566500) ESP_RTSP_CLIENT: RTSP client task stoped
```

Question 2：
When I use ZLMeiaKit as RTSP server, I encounter the error "454 Session Not Found", which occurs during the PLAY stage.
I think this is because the PLAY command of esp-rtsp does not carry session.

All log：
```
demo-chime> start 2 rtsp://110.1.1.189:554/live/test
I (43004) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Nov 20 2024-11:21:23
I (43006) ESP_RTSP_CLIENT: Conecting...
I (43006) ESP_RTSP_CLIENT: RTSP Client URL:rtsp://110.1.1.189:554/live/test
I (43039) ESP_RTSP_CLIENT: TX:

OPTIONS rtsp://110.1.1.189:554/live/test RTSP/1.0
CSeq: 0
User-Agent: ESP32-RTSP-client 1.1.0

I (43053) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 0
Date: Wed, Nov 20 2024 07:00:16 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, ANNOUNCE, RECORD, SET_PARAMETER, GET_PARAMETER
Server: ZLMediaKit(git hash:226b87a/2024-11-18T14:08:53+08:00,branch:master,build time:2024-11-20T11:19:02)

I (43078) ESP_RTSP_CLIENT: TX:

DESCRIBE rtsp://110.1.1.189:554/live/test RTSP/1.0
CSeq: 1
User-Agent: ESP32-RTSP-client 1.1.0
Accept: application/sdp

demo-chime> I (43132) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
Content-Base: rtsp://110.1.1.189:554/live/test/
Content-Length: 252
Content-Type: application/sdp
CSeq: 1
Date: Wed, Nov 20 2024 07:00:16 GMT
Server: ZLMediaKit(git hash:226b87a/2024-11-18T14:08:53+08:00,branch:master,build time:2024-11-20T11:19:02)
Session: l9cBm10IEuaH
x-Accept-Dynamic-Rate: 1
x-Accept-Retransmit: our-retransmit

v=0
o=- 0 0 IN IP4 0.0.0.0
s=Streamed by ZLMediaKit(git hash:226b87a/2024-11-18T14:08:53+08:00,branch:master,build time:2024-11-20T11:19:02)
c=IN IP4 0.0.0.0
t=0 0
a=range:npt=now-
a=control:*
m=audio 0 RTP/AVP 8
b=AS:64
a=control:streamid=0

I (43180) RTSP_SERVICE: RTSP_STATE_SETUP
I (43191) ESP_RTSP_CLIENT: TX:

SETUP rtsp://110.1.1.189:554/live/test/trackID=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=32829-32830;mode=record
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0

I (43259) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 2
Date: Wed, Nov 20 2024 07:00:16 GMT
Server: ZLMediaKit(git hash:226b87a/2024-11-18T14:08:53+08:00,branch:master,build time:2024-11-20T11:19:02)
Session: l9cBm10IEuaH
Transport: RTP/AVP/UDP;unicast;client_port=32829-32830;mode=record;server_port=31140-31141;ssrc=7F105C79

I (43284) ESP_RTSP_CLIENT: RTSP Server arport 32829
I (43295) ESP_RTSP_CLIENT: TX:

PLAY rtsp://110.1.1.189:554/live/test RTSP/1.0
CSeq: 3
User-Agent: ESP32-RTSP-client 1.1.0

I (43353) ESP_RTSP_CLIENT: RX:

RTSP/1.0 454 Session Not Found
Connection: Close
CSeq: 3
Date: Wed, Nov 20 2024 07:00:16 GMT
Server: ZLMediaKit(git hash:226b87a/2024-11-18T14:08:53+08:00,branch:master,build time:2024-11-20T11:19:02)
Session: l9cBm10IEuaH

I (43377) ESP_RTSP_CLIENT: TX:

TEARDOWN rtsp://110.1.1.189:554/live/test RTSP/1.0
CSeq: 3
User-Agent: ESP32-RTSP-client 1.1.0
Session: l9cBm10IEuaH

I (43390) RTSP_SERVICE: RTSP_STATE_TEARDOWN
I (43391) ESP_RTSP_CLIENT: RTSP client task stoped
```

TempoTian, 2024-11-20:
@ManikandanVaradharaj  Of cource supported for P4. For esp_media_protocol is  all software realization, it does not dependent on special platform. Currently it only dependent on media_lib_sal in esp-adf-libs, you can copy these two folder into your project components folder, and used it without ADF also. Just happy to use!

TempoTian, 2024-11-20:
@alpha-alan I will download live555 and have a try, Thanks for your info, it is easy to fix, once finished I will notify to you.

ManikandanVaradharaj, 2024-11-20:
@TempoTian  but in adf list of table it shows rtc and rtsp will not support for esp32-p4.can u give clear idea for me that how can i do the rtsp client in esp32-p4 with lvgl ui support.because esp32-p4 there is no option to check whether network is working or not through WiFi.

TempoTian, 2024-11-20:
@ManikandanVaradharaj  You need esp_wifi_remote https://github.com/espressif/esp-wifi-remote to use wifi on P4.  The LWIP and socket related things still run on P4, so there is no need to do any change with the lib. Official document updated in delay. We have release P4 support for esp_media_protocols for long time.

ManikandanVaradharaj, 2024-11-20:
@TempoTian  okay after the wifi connection for rtsp protocol in esp32-p4 which are the libraries needed.that i'll implement with lvgl ui support

TempoTian, 2024-11-20:
@ManikandanVaradharaj  follow this url https://components.espressif.com/components/lvgl/lvgl, you can use command `idf.py add-dependency "lvgl/lvgl^9.2.2"` in your project to download it directly.

alpha-alan, 2024-11-20:
@TempoTian 
I also used mediamtx to test. The error I encountered this time was "400 bad request" esp-rtsp still cannot connect to mediamtx

I noticed a log from mediamtx pointing out the reason:
`transport header contains a invalid mode (record)`

At the same time, I also noticed that the SETUP of esp-rtsp contains a mode=record, but mediamtx does not recognize it.
```
SETUP rtsp://110.1.1.189:8554/live/test/trackID=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=35975-35976;mode=record
```
Hopefully this issue can be resolved along with the two issues I mentioned earlier, thank you.

esp-rtsp all log
```
demo-chime> start 2 rtsp://110.1.1.189:8554/live/test
I (840385) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Aug 15 2024-04:10:02
I (840387) ESP_RTSP_CLIENT: Conecting...
I (840387) ESP_RTSP_CLIENT: RTSP Client URL:rtsp://110.1.1.189:8554/live/test
I (840398) ESP_RTSP_CLIENT: TX:

OPTIONS rtsp://110.1.1.189:8554/live/test RTSP/1.0
CSeq: 0
User-Agent: ESP32-RTSP-client 1.1.0

I (840419) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 0
Public: DESCRIBE, ANNOUNCE, SETUP, PLAY, RECORD, PAUSE, GET_PARAMETER, TEARDOWN
Server: gortsplib

I (840432) ESP_RTSP_CLIENT: TX:

DESCRIBE rtsp://110.1.1.189:8554/live/test RTSP/1.0
CSeq: 1
                                                                                                                                                                                           User-Agent: ESP32-RTSP-client 1.1.0
Accept: application/sdp

I (840511) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 1
Content-Base: rtsp://110.1.1.189:8554/live/test/
Content-Length: 131
Content-Type: application/sdp
Server: gortsplib

v=0
o=- 0 0 IN IP4 127.0.0.1
s=No Name
c=IN IP4 0.0.0.0
t=0 0
m=audio 0 RTP/AVP 8
a=control:trackID=0
a=rtpmap:8 PCMA/8000

I (840536) ESP_RTSP_CLIENT: TX:

SETUP rtsp://110.1.1.189:8554/live/test/trackID=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=35975-35976;mode=record
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0
Session: 6809774

I (840601) ESP_RTSP_CLIENT: RX:

RTSP/1.0 400 Bad Request
CSeq: 2
Server: gortsplib

I (840603) ESP_RTSP_CLIENT: TX:

PLAY rtsp://110.1.1.189:8554/live/test RTSP/1.0
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0

I (840615) ESP_RTSP_CLIENT: TX:

TEARDOWN rtsp://110.1.1.189:8554/live/test RTSP/1.0
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0
Session: 6809774

E (840628) TRANS_TCP: tcp_poll_write select error 104, errno = Connection reset by peer, fd = 54
E (840640) ESP_RTSP_CLIENT: Write failed, error: Socket is not connected
I (840651) RTSP_SERVICE: RTSP_STATE_TEARDOWN
I (840652) ESP_RTSP_CLIENT: RTSP client task stoped
```

mediamtx all log
```
alan@BUilderServer:~/mediamtx$ ./mediamtx mediamtx.yml
2024/11/20 18:00:55 INF MediaMTX v1.9.3
2024/11/20 18:00:55 INF configuration loaded from /home/alan/mediamtx/mediamtx.yml
2024/11/20 18:00:55 INF [RTSP] listener opened on :8554 (TCP), :8000 (UDP/RTP), :8001 (UDP/RTCP)
2024/11/20 18:00:55 INF [RTMP] listener opened on :1935
2024/11/20 18:00:55 INF [HLS] listener opened on :8888
2024/11/20 18:00:55 INF [WebRTC] listener opened on :8889 (HTTP), :8189 (ICE/UDP)
2024/11/20 18:00:55 INF [SRT] listener opened on :8890 (UDP)
2024/11/20 18:00:59 INF [RTSP] [conn 127.0.0.1:51330] opened
2024/11/20 18:00:59 INF [RTSP] [session 52b540c4] created by 127.0.0.1:51330
2024/11/20 18:00:59 INF [RTSP] [session 52b540c4] is publishing to path 'live/test', 1 track (G711)
2024/11/20 18:01:08 INF [RTSP] [conn 110.1.1.131:51472] opened
2024/11/20 18:01:08 INF [RTSP] [session b71ce285] created by 110.1.1.131:51472
2024/11/20 18:01:08 INF [RTSP] [conn 110.1.1.131:51472] closed: transport header contains a invalid mode (record)
2024/11/20 18:01:08 INF [RTSP] [session b71ce285] destroyed: not in use

```

alpha-alan, 2024-11-21:
@TempoTian ，Could you please let me know when I might receive the new library with all the issues resolved?

TempoTian, 2024-11-21:
@alpha-alan I am undertesting now. It won't take long time. It speeds me some time to setup the server for lib555 done. now it is ready to test.

ManikandanVaradharaj, 2024-11-21:
@TempoTian  for rtsp which of the source and header file we can implement in the project that will support for esp32-p4

alpha-alan, 2024-11-21:
@TempoTian 
I understand you need to test before releasing a new library. On the other hand, we will also demonstrate the RTSP client function to customers next Tuesday. Could you please give us a new library before get off work tomorrow?

ManikandanVaradharaj, 2024-11-21:
I'm trying the rtsp protocol inside the video player app like enter the ip address and start the live stream for this i set upto the enter ip address..after ip address it should work through rtsp code for that which libraries would be useful.

ManikandanVaradharaj, 2024-11-21:
Any specific libraries needed for that rather than..esp_media_protocol & media_lib_sal.could i paste in my components that's enough.

TempoTian, 2024-11-21:
NO extra lib needed, just these 2 should OK.

ManikandanVaradharaj, 2024-11-22:
In the lib,it has only esp_rtsp.h in esp_media_protocol, file which is not working with adaptability in the project.

TempoTian, 2024-11-22:
@ManikandanVaradharaj  libpath is as following, the provide lib is used for debug issue, not release yet:
https://github.com/espressif/esp-adf-libs/tree/master/esp_media_protocols
https://github.com/espressif/esp-adf-libs/tree/master/media_lib_sal

ManikandanVaradharaj, 2024-11-22:
For esp32p4..this protocol implementation is not defining.. I'm gets
confused of adding the protocol implementation properly.

TempoTian, 2024-11-22:
P4 Lib is under: https://github.com/espressif/esp-adf-libs/tree/master/esp_media_protocols/lib/esp32p4, ADF already support P4, you can try to use any rtsp example to do test firstly

ManikandanVaradharaj, 2024-11-22:
For the rtsp example try we need to set the lvgl ui setup separately for rtsp right.

TempoTian, 2024-11-25:
@alpha-alan Attach new lib here. 
[libesp_media_protocols.zip](https://github.com/user-attachments/files/17896732/libesp_media_protocols.zip)

please modify APP as following also:

上层要配合做下面的一些改变：
1： 注册codec 的notify
 if (mode == RTSP_CLIENT_PLAY) {
        data_cb.receive_audio = _receive_audio;
        data_cb.stream_codec = _stream_info_cb;
        data_cb.send_audio = NULL;
        data_cb.send_video = NULL;
        rtsp_config.video_enable = false;
        rtsp_config.trans = RTSP_TRANSPORT_UDP;
    }

2：notify中改变av stream的codec设定
static int _stream_info_cb(uint8_t aud_codec, uint8_t vid_codec,void *ctx)
{
    av_stream_handle_t av_stream = (av_stream_handle_t) ctx;
    if (aud_codec == 0) {
        av_audio_dec_change_codec(av_stream, AV_ACODEC_G711U);
    } else if (aud_codec == 8) {
        av_audio_dec_change_codec(av_stream, AV_ACODEC_G711A);
    }
    return 0;
}

3：av_stream添加API av_audio_dec_change_codec
/**
 * @brief      Change audio decoder codec
 *
 * @param[in]  av_stream   The av_stream handle
 * @param[in]  acodec      New audio codec
 *
 * @return
 *     - ESP_OK on success
 *     - ESP_ERR_INVALID_ARG on wrong handle
 */
int av_audio_dec_change_codec(av_stream_handle_t av_stream, av_stream_acodec_t acodec);

int av_audio_dec_change_codec(av_stream_handle_t av_stream, av_stream_acodec_t acodec)
{
    av_stream->config.acodec_type = acodec;
    return ESP_OK;
}

4：enhance av_stream 退出的时候让等待写的操作及时返回
int av_audio_dec_stop(av_stream_handle_t av_stream)
{
    AUDIO_NULL_CHECK(TAG, av_stream, return ESP_ERR_INVALID_ARG);
    if (!av_stream->adec_run) {
        return ESP_OK;
    }

    av_stream->adec_run = false;
    if (av_stream->ringbuf_dec) {
        rb_abort(av_stream->ringbuf_dec);
    }

ManikandanVaradharaj, 2024-11-25:
@TempoTian could you please say how could i test example rtsp in esp32_p4 module

TempoTian, 2024-11-25:
@alpha-alan Please use following lib to test again. Add handle server timeout processing to avoid sever send FIN during RTP receiving.

[libesp_media_protocols.zip](https://github.com/user-attachments/files/17899484/libesp_media_protocols.zip)

TempoTian, 2024-11-25:
@ManikandanVaradharaj  You can use https://github.com/espressif/esp-adf/tree/master/examples/protocols/esp-rtsp to do test. I think you can start another thread to ask your questions to avoid issue mixing together cause some confusing.

TempoTian, 2024-11-27:
For RTSP AAC support， use following lib and follow the readme to do test.
[rtsp_aac.zip](https://github.com/user-attachments/files/17933764/rtsp_aac.zip)

ManikandanVaradharaj, 2024-11-27:
@TempoTian can i use this rtsp_aac for esp32P4 module

TempoTian, 2024-11-27:
Currently I only build it for S3

ManikandanVaradharaj, 2024-11-27:
i need to test for esp32p4 in rtsp client way(receive video and audio),could u do the changes for this

TempoTian, 2024-11-28:
Currently on esp32P4 we support MJPEG and H264 decoder, but we only have hardware MJPEG decoder support upper to 1920x1080 30fps decoding while display, h264 software decode have low resolution support lower than 640x480, I can make a simple example for you.

ManikandanVaradharaj, 2024-11-28:
Give me a example code like  receive the video and play with low resolution.

ManikandanVaradharaj, 2024-11-29:
@TempoTian can I get the example rtsp client code for esp32p4 module.

TempoTian, 2024-11-29:
Please try to use below example to do test
[rtsp_p4.zip](https://github.com/user-attachments/files/17954195/rtsp_p4.zip)

TempoTian, 2024-11-29:
For currently H264 only support baseline decoding of low resolution.
you can use ffmpeg command to do transcode:
`./ffmpeg -re -stream_loop -1 -i  test.mp4 -vf "fps=25,scale=320:240" -c:v libx264 -profile:v baseline -acodec aac -ar 16000 -ac 1 -f rtsp rtsp://xxxxxx/test`

ManikandanVaradharaj, 2024-11-29:
@TempoTian How could i use the above command

ManikandanVaradharaj, 2024-11-29:
@TempoTian  after i build the rtsp_p4_zip,with the credentials changes of my wifi and url.I'm receiving the monitor log as Executing action: monitor
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... ESP32-P4
Running idf_monitor in directory /home/smartbuild/Downloads/rtsp_p4
Executing "/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/smartbuild/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32p4 --revision 0 --decode-panic backtrace /home/smartbuild/Downloads/rtsp_p4/build/rtsp_test.elf -m '/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/smartbuild/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17e4
load:0x4ff2abd0,len:0xb8c
load:0x4ff2cbd0,len:0x3010
entry 0x4ff2abda

TempoTian, 2024-11-29:
You should select first one
[image]

ManikandanVaradharaj, 2024-11-29:
@TempoTian  after change this option.it couldn't able to find my port

ManikandanVaradharaj, 2024-11-29:
Executing action: monitor
Running idf_monitor in directory /home/smartbuild/Downloads/rtsp_p4
Executing "/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/smartbuild/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32p4 --revision 0 --decode-panic backtrace /home/smartbuild/Downloads/rtsp_p4/build/rtsp_test.elf -m '/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/smartbuild/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17e4
load:0x4ff2abd0,len:0xc38
load:0x4ff2cbd0,len:0x2fdc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (26) boot: compile time Nov 29 2024 12:24:23
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (32) boot.esp32p4: SPI Speed      : 80MHz
I (37) boot.esp32p4: SPI Mode       : DIO
I (42) boot.esp32p4: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00200000
I (87) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=40130020 size=714d0h (464080) map
I (199) esp_image: segment 1: paddr=000814f8 vaddr=30100000 size=00020h (    32) load
I (201) esp_image: segment 2: paddr=00081520 vaddr=30100020 size=0003ch (    60) load
I (207) esp_image: segment 3: paddr=00081564 vaddr=4ff00000 size=0eab4h ( 60084) load
I (230) esp_image: segment 4: paddr=00090020 vaddr=40000020 size=124e1ch (1199644) map
I (490) esp_image: segment 5: paddr=001b4e44 vaddr=4ff0eab4 size=07700h ( 30464) load
I (499) esp_image: segment 6: paddr=001bc54c vaddr=4ff16200 size=03aa0h ( 15008) load
I (509) boot: Loaded app from partition at offset 0x10000
I (509) boot: Disabling RNG early entropy source...
I (522) hex_psram: vendor id    : 0x0d (AP)
I (522) hex_psram: Latency      : 0x01 (Fixed)
I (522) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (526) hex_psram: dev id       : 0x03 (generation 4)
I (531) hex_psram: density      : 0x07 (256 Mbit)
I (537) hex_psram: good-die     : 0x06 (Pass)
I (542) hex_psram: SRF          : 0x02 (Slow Refresh)
I (547) hex_psram: BurstType    : 0x00 ( Wrap)
I (553) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (558) hex_psram: BitMode      : 0x01 (X16 Mode)
I (563) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (569) hex_psram: DriveStrength: 0x00 (1/1)
I (575) MSPI DQS: tuning success, best phase id is 2
I (767) MSPI DQS: tuning success, best delayline id is 11
I (767) esp_psram: Found 32MB PSRAM device
I (768) esp_psram: Speed: 200MHz
I (771) hex_psram: psram CS IO is dedicated
I (775) cpu_start: Multicore app
I (1729) esp_psram: SPI SRAM memory test OK
W (1738) clk: esp_perip_clk_init() has not been implemented yet
I (1745) cpu_start: Pro cpu start user code
I (1745) cpu_start: cpu freq: 360000000 Hz
I (1745) app_init: Application information:
I (1748) app_init: Project name:     rtsp_test
I (1753) app_init: App version:      1
I (1758) app_init: Compile time:     Nov 29 2024 12:24:27
I (1764) app_init: ELF file SHA256:  7119e0adf...
I (1769) app_init: ESP-IDF:          v5.3.1
I (1774) efuse_init: Min chip rev:     v0.0
I (1779) efuse_init: Max chip rev:     v0.99 
I (1784) efuse_init: Chip rev:         v0.1
I (1789) heap_init: Initializing. RAM available for dynamic allocation:
I (1796) heap_init: At 4FF1D0C0 len 0001DF00 (119 KiB): RAM
I (1802) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
[... 56 lines cut ...]
E (7179) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (7179) task_wdt:  - IDLE0 (CPU 0)
E (7179) task_wdt: Tasks currently running:
E (7179) task_wdt: CPU 0: main
E (7179) task_wdt: CPU 1: IDLE1
E (7179) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
--- Stack dump detected
Core  0 register dump:
MEPC    : 0x400991be  RA      : 0x400991b0  SP      : 0x4ff2a730  GP      : 0x4ff16a00  
--- 0x400991be: mipi_dsi_host_ll_gen_is_read_fifo_empty at /home/smartbuild/esp/esp-idf/components/hal/esp32p4/include/hal/mipi_dsi_host_ll.h:688
 (inlined by) mipi_dsi_hal_host_gen_read_short_packet at /home/smartbuild/esp/esp-idf/components/hal/mipi_dsi_hal.c:203
0x400991b0: mipi_dsi_hal_host_gen_read_short_packet at /home/smartbuild/esp/esp-idf/components/hal/mipi_dsi_hal.c:201 (discriminator 1)

TP      : 0x4ff2a8b0  T0      : 0x4fc0a778  T1      : 0x48001410  T2      : 0x00000000  
S0/FP   : 0x00000001  S1      : 0x48001414  A0      : 0x48001414  A1      : 0x00000000  
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x500a0000  A5      : 0x00002005  
A6      : 0x00000000  A7      : 0x00000008  S2      : 0x4ff2a78b  S3      : 0x00000000  
S4      : 0x00000006  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
--- 0x4fc0a778: _raise_r in ROM

S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000004  T4      : 0x00000003  T5      : 0x000000da  T6      : 0x000000bb  
MSTATUS : 0x4013de9c  MTVEC   : 0x4ff39804  MCAUSE  : 0x48001454  MTVAL   : 0x00000000  
MHARTID : 0x4800142c

ManikandanVaradharaj, 2024-11-29:
@TempoTian After I changed the option I'm getting log like this..

TempoTian, 2024-11-29:
Maybe you forget to add lcd panel? Does you board run UI ok?

ManikandanVaradharaj, 2024-11-29:
yeah lcd panel connected,before demo repo are i tired and its worked fine.

ManikandanVaradharaj, 2024-11-29:
For this rtsp_p4,display not getting any images and ui

TempoTian, 2024-11-29:
You can review whether config same comparing with your RUN OK project in board_cfg.h
```
Board: ESP32_P4_DEV_V11
i2c: {sda: 7, scl: 8}
i2s: {mclk: 13, bclk: 12, ws: 10, dout: 9, din: 11}
in_out: {codec: ES8311, pa: 53, use_mclk: 1, pa_gain:6}
sdcard: {clk: 43, cmd: 44, d0: 39, d1: 40, d2: 41, d3: 42}
lcd: {
    bus: mipi, ldo_chan: 3, ldo_voltage: 2500, lane_num: 2,
    lane_bitrate: 1000, dpi_clk: 80, bit_depth: 16, fb_num: 2
    dsi_hsync: 40,  dsi_vsync: 4,
    dsi_hbp: 140, dsi_hfp: 120,
    dsi_vbp: 16, dsi_vfp: 16,
    rst: -1,
    width: 720, height: 1280,
}
```

TempoTian, 2024-11-29:
Replace with this board and change config to 
[codec_board.zip](https://github.com/user-attachments/files/17957943/codec_board.zip)

[image]

ManikandanVaradharaj, 2024-11-29:
@TempoTian board_cfg.txt only is there

ManikandanVaradharaj, 2024-11-29:
@TempoTian after i changed the board name.I got errors like /home/smartbuild/esp/esp-idf/components/freertos/esp_additions/include/freertos/task_snapshot.h:8:2: warning: #warning freertos/task_snapshot.h header is no longer used, and will be removed in future versions. [-Wcpp]
    8 | #warning freertos/task_snapshot.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_create':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:194:34: error: 'xTaskHandle' undeclared (first use in this function); did you mean 'xTaskGetHandle'?
  194 |                                 (xTaskHandle *)handle, core) != pdPASS) {
      |                                  ^~~~~~~~~~~
      |                                  xTaskGetHandle
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:194:34: note: each undeclared identifier is reported only once for each function it appears in
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:194:47: error: expected expression before ')' token
  194 |                                 (xTaskHandle *)handle, core) != pdPASS) {
      |                                               ^
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:193:9: error: too few arguments to function 'xTaskCreatePinnedToCore'
  193 |     if (xTaskCreatePinnedToCore(body, name, stack_size, arg, prio,
      |         ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/smartbuild/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:1533,
                 from /home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:30:
/home/smartbuild/esp/esp-idf/components/freertos/esp_additions/include/freertos/idf_additions.h:63:16: note: declared here
   63 |     BaseType_t xTaskCreatePinnedToCore( TaskFunction_t pxTaskCode,
      |                ^~~~~~~~~~~~~~~~~~~~~~~
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_destroy':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:205:18: error: 'xTaskHandle' undeclared (first use in this function); did you mean 'xTaskGetHandle'?
  205 |     vTaskDelete((xTaskHandle)handle);
      |                  ^~~~~~~~~~~
      |                  xTaskGetHandle
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:205:30: error: expected ')' before 'handle'
  205 |     vTaskDelete((xTaskHandle)handle);
      |                ~             ^~~~~~
      |                              )
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_set_priority':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:210:23: error: 'xTaskHandle' undeclared (first use in this function); did you mean 'xTaskGetHandle'?
  210 |     vTaskPrioritySet((xTaskHandle)handle, prio);
      |                       ^~~~~~~~~~~
      |                       xTaskGetHandle
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:210:35: error: expected ')' before 'handle'
  210 |     vTaskPrioritySet((xTaskHandle)handle, prio);
      |                     ~             ^~~~~~
      |                                   )
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:210:5: error: too few arguments to function 'vTaskPrioritySet'
  210 |     vTaskPrioritySet((xTaskHandle)handle, prio);
      |     ^~~~~~~~~~~~~~~~
In file included from /home/smartbuild/esp/esp-idf/components/freertos/esp_additions/include/freertos/idf_additions.h:20:
/home/smartbuild/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:1100:6: note: declared here
 1100 | void vTaskPrioritySet( TaskHandle_t xTask,
      |      ^~~~~~~~~~~~~~~~
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_sleep':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:216:21: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
  216 |     vTaskDelay(ms / portTICK_RATE_MS);
      |                     ^~~~~~~~~~~~~~~~
      |                     portTICK_PERIOD_MS
[1037/1098] Building C object esp-idf/media_lib_sal/CMakeFiles/__idf_media_lib_sal.dir/mem_trace/media_lib_mem_trace.c.obj
ninja: build stopped: subcommand failed.
HINT: You are maybe using pre FreeRTOS V8.0.0 APIs. The backward compatibility of such APIs is no longer enabled by default. Please turn on CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY explicitly to use such APIs.
ninja failed with exit code 1, output of the command is in the /home/smartbuild/Downloads/rtsp_p4/build/log/idf_py_stderr_output_101966 and /home/smartbuild/Downloads/rtsp_p4/build/log/idf_py_stdout_output_101966

ManikandanVaradharaj, 2024-11-29:
@TempoTian After I changed board name,I got the errors in media_lib_os_freertos../home/smartbuild/esp/esp-idf/components/freertos/esp_additions/include/freertos/task_snapshot.h:8:2: warning: #warning freertos/task_snapshot.h header is no longer used, and will be removed in future versions. [-Wcpp]
    8 | #warning freertos/task_snapshot.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_create':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:194:34: error: 'xTaskHandle' undeclared (first use in this function); did you mean 'xTaskGetHandle'?
  194 |                                 (xTaskHandle *)handle, core) != pdPASS) {
      |                                  ^~~~~~~~~~~
      |                                  xTaskGetHandle
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:194:34: note: each undeclared identifier is reported only once for each function it appears in
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:194:47: error: expected expression before ')' token
  194 |                                 (xTaskHandle *)handle, core) != pdPASS) {
      |                                               ^
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:193:9: error: too few arguments to function 'xTaskCreatePinnedToCore'
  193 |     if (xTaskCreatePinnedToCore(body, name, stack_size, arg, prio,
      |         ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/smartbuild/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:1533,
                 from /home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:30:
/home/smartbuild/esp/esp-idf/components/freertos/esp_additions/include/freertos/idf_additions.h:63:16: note: declared here
   63 |     BaseType_t xTaskCreatePinnedToCore( TaskFunction_t pxTaskCode,
      |                ^~~~~~~~~~~~~~~~~~~~~~~
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_destroy':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:205:18: error: 'xTaskHandle' undeclared (first use in this function); did you mean 'xTaskGetHandle'?
  205 |     vTaskDelete((xTaskHandle)handle);
      |                  ^~~~~~~~~~~
      |                  xTaskGetHandle
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:205:30: error: expected ')' before 'handle'
  205 |     vTaskDelete((xTaskHandle)handle);
      |                ~             ^~~~~~
      |                              )
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_set_priority':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:210:23: error: 'xTaskHandle' undeclared (first use in this function); did you mean 'xTaskGetHandle'?
  210 |     vTaskPrioritySet((xTaskHandle)handle, prio);
      |                       ^~~~~~~~~~~
      |                       xTaskGetHandle
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:210:35: error: expected ')' before 'handle'
  210 |     vTaskPrioritySet((xTaskHandle)handle, prio);
      |                     ~             ^~~~~~
      |                                   )
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:210:5: error: too few arguments to function 'vTaskPrioritySet'
  210 |     vTaskPrioritySet((xTaskHandle)handle, prio);
      |     ^~~~~~~~~~~~~~~~
In file included from /home/smartbuild/esp/esp-idf/components/freertos/esp_additions/include/freertos/idf_additions.h:20:
/home/smartbuild/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:1100:6: note: declared here
 1100 | void vTaskPrioritySet( TaskHandle_t xTask,
      |      ^~~~~~~~~~~~~~~~
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c: In function '_thread_sleep':
/home/smartbuild/Downloads/rtsp_p4/components/media_lib_sal/port/media_lib_os_freertos.c:216:21: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
  216 |     vTaskDelay(ms / portTICK_RATE_MS);
      |                     ^~~~~~~~~~~~~~~~
      |                     portTICK_PERIOD_MS
[1037/1098] Building C object esp-idf/media_lib_sal/CMakeFiles/__idf_media_lib_sal.dir/mem_trace/media_lib_mem_trace.c.obj
ninja: build stopped: subcommand failed.
HINT: You are maybe using pre FreeRTOS V8.0.0 APIs. The backward compatibility of such APIs is no longer enabled by default. Please turn on CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY explicitly to use such APIs.
ninja failed with exit code 1, output of the command is in the /home/smartbuild/Downloads/rtsp_p4/build/log/idf_py_stderr_output_101966 and /home/smartbuild/Downloads/rtsp_p4/build/log/idf_py_stdout_output_101966

TempoTian, 2024-12-02:
You should enable `CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY` in menuconfig

ManikandanVaradharaj, 2024-12-02:
@TempoTian After the changes i build and flashed...No In monitor it shows my IP address as wrong i set the URL number main.c is(.0.132) and monitor(.0.134) is different and it doesn't shows whether it connected WiFi or not also.Nothing came in Display  Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-P4
Running idf_monitor in directory /home/smartbuild/Downloads/rtsp_p4
Executing "/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/smartbuild/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32p4 --revision 0 --decode-panic backtrace /home/smartbuild/Downloads/rtsp_p4/build/rtsp_test.elf -m '/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/smartbuild/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17e4
load:0x4ff2abd0,len:0xc38
load:0x4ff2cbd0,len:0x2fdc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (26) boot: compile time Dec  2 2024 10:49:34
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (32) boot.esp32p4: SPI Speed      : 80MHz
I (37) boot.esp32p4: SPI Mode       : DIO
I (42) boot.esp32p4: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00200000
I (87) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=40130020 size=718f0h (465136) map
I (199) esp_image: segment 1: paddr=00081918 vaddr=30100000 size=00020h (    32) load
I (202) esp_image: segment 2: paddr=00081940 vaddr=30100020 size=0003ch (    60) load
I (207) esp_image: segment 3: paddr=00081984 vaddr=4ff00000 size=0e694h ( 59028) load
I (229) esp_image: segment 4: paddr=00090020 vaddr=40000020 size=1255f8h (1201656) map
I (490) esp_image: segment 5: paddr=001b5620 vaddr=4ff0e694 size=07b20h ( 31520) load
I (500) esp_image: segment 6: paddr=001bd148 vaddr=4ff16200 size=03aa0h ( 15008) load
I (509) boot: Loaded app from partition at offset 0x10000
I (510) boot: Disabling RNG early entropy source...
I (522) hex_psram: vendor id    : 0x0d (AP)
I (522) hex_psram: Latency      : 0x01 (Fixed)
I (523) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (526) hex_psram: dev id       : 0x03 (generation 4)
I (532) hex_psram: density      : 0x07 (256 Mbit)
I (537) hex_psram: good-die     : 0x06 (Pass)
I (542) hex_psram: SRF          : 0x02 (Slow Refresh)
I (548) hex_psram: BurstType    : 0x00 ( Wrap)
I (553) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (558) hex_psram: BitMode      : 0x01 (X16 Mode)
I (564) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (570) hex_psram: DriveStrength: 0x00 (1/1)
I (575) MSPI DQS: tuning success, best phase id is 2
I (767) MSPI DQS: tuning success, best delayline id is 11
I (768) esp_psram: Found 32MB PSRAM device
I (768) esp_psram: Speed: 200MHz
I (771) hex_psram: psram CS IO is dedicated
I (776) cpu_start: Multicore app
I (1729) esp_psram: SPI SRAM memory test OK
W (1739) clk: esp_perip_clk_init() has not been implemented yet
I (1745) cpu_start: Pro cpu start user code
I (1746) cpu_start: cpu freq: 360000000 Hz
I (1746) app_init: Application information:
I (1749) app_init: Project name:     rtsp_test
I (1754) app_init: App version:      1
I (1758) app_init: Compile time:     Dec  2 2024 10:49:12
I (1764) app_init: ELF file SHA256:  6bb9a1565...
I (1770) app_init: ESP-IDF:          v5.3.1
I (1775) efuse_init: Min chip rev:     v0.0
I (1779) efuse_init: Max chip rev:     v0.99 
I (1784) efuse_init: Chip rev:         v0.1
[... 124 lines cut ...]
I (4680) rpc_evt: EVT rcvd: Wi-Fi Start
I (4680) rpc_core: <-- RPC_Req  [0x101], uid 5
I (4690) rpc_rsp:  --> RPC_Resp [0x201], uid 5
I (4690) H_API: esp_wifi_remote_connect
I (4690) rpc_core: <-- RPC_Req  [0x11a], uid 6
I (4710) rpc_rsp:  --> RPC_Resp [0x21a], uid 6

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
I (4750) main_task: Returned from app_main()
esp> I (7480) rpc_evt: Event [0x29] received
I (8490) H_API: esp_wifi_remote_connect
I (8490) rpc_core: <-- RPC_Req  [0x11a], uid 7
I (8510) rpc_rsp:  --> RPC_Resp [0x21a], uid 7
I (8510) RTSP_Test: retry to connect to the AP
I (10920) H_API: esp_wifi_remote_connect
I (10920) rpc_core: <-- RPC_Req  [0x11a], uid 8
I (10930) rpc_rsp:  --> RPC_Resp [0x21a], uid 8
I (10930) RTSP_Test: retry to connect to the AP
I (13750) esp_wifi_remote: esp_wifi_internal_reg_rxcb: sta: 0x400d2c18
--- 0x400d2c18: wifi_sta_receive at /home/smartbuild/esp/esp-idf/components/esp_wifi/src/wifi_netif.c:38

I (15250) esp_netif_handlers: sta ip: 192.168.0.134, mask: 255.255.255.0, gw: 192.168.0.1
I (15250) RTSP_Test: got ip:192.168.0.134

ManikandanVaradharaj, 2024-12-02:
@TempoTian Still not i connected speaker for the p4 module.first i need to check video streaming is happening properly or not.

ManikandanVaradharaj, 2024-12-02:
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-P4
Running idf_monitor in directory /home/smartbuild/Downloads/rtsp_p4
Executing "/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/smartbuild/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32p4 --revision 0 --decode-panic backtrace /home/smartbuild/Downloads/rtsp_p4/build/rtsp_test.elf -m '/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/smartbuild/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17e4
load:0x4ff2abd0,len:0xc38
load:0x4ff2cbd0,len:0x2fdc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (26) boot: compile time Dec  2 2024 10:49:34
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (32) boot.esp32p4: SPI Speed      : 80MHz
I (37) boot.esp32p4: SPI Mode       : DIO
I (42) boot.esp32p4: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00200000
I (87) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=40130020 size=718f0h (465136) map
I (199) esp_image: segment 1: paddr=00081918 vaddr=30100000 size=00020h (    32) load
I (202) esp_image: segment 2: paddr=00081940 vaddr=30100020 size=0003ch (    60) load
I (207) esp_image: segment 3: paddr=00081984 vaddr=4ff00000 size=0e694h ( 59028) load
I (229) esp_image: segment 4: paddr=00090020 vaddr=40000020 size=1255f8h (1201656) map
I (490) esp_image: segment 5: paddr=001b5620 vaddr=4ff0e694 size=07b20h ( 31520) load
I (500) esp_image: segment 6: paddr=001bd148 vaddr=4ff16200 size=03aa0h ( 15008) load
I (509) boot: Loaded app from partition at offset 0x10000
I (510) boot: Disabling RNG early entropy source...
I (522) hex_psram: vendor id    : 0x0d (AP)
I (522) hex_psram: Latency      : 0x01 (Fixed)
I (523) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (526) hex_psram: dev id       : 0x03 (generation 4)
I (532) hex_psram: density      : 0x07 (256 Mbit)
I (537) hex_psram: good-die     : 0x06 (Pass)
I (542) hex_psram: SRF          : 0x02 (Slow Refresh)
I (548) hex_psram: BurstType    : 0x00 ( Wrap)
I (553) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (558) hex_psram: BitMode      : 0x01 (X16 Mode)
I (564) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (570) hex_psram: DriveStrength: 0x00 (1/1)
I (575) MSPI DQS: tuning success, best phase id is 2
I (767) MSPI DQS: tuning success, best delayline id is 11
I (768) esp_psram: Found 32MB PSRAM device
I (768) esp_psram: Speed: 200MHz
I (771) hex_psram: psram CS IO is dedicated
I (776) cpu_start: Multicore app
I (1729) esp_psram: SPI SRAM memory test OK
W (1739) clk: esp_perip_clk_init() has not been implemented yet
I (1745) cpu_start: Pro cpu start user code
I (1746) cpu_start: cpu freq: 360000000 Hz
I (1746) app_init: Application information:
I (1749) app_init: Project name:     rtsp_test
I (1754) app_init: App version:      1
I (1758) app_init: Compile time:     Dec  2 2024 10:49:12
I (1764) app_init: ELF file SHA256:  6bb9a1565...
I (1770) app_init: ESP-IDF:          v5.3.1
I (1775) efuse_init: Min chip rev:     v0.0
I (1779) efuse_init: Max chip rev:     v0.99 
I (1784) efuse_init: Chip rev:         v0.1
[... 170 lines cut ...]
E (64200) MEDIA_OS: Error creating RestrictedPinnedToCore rtsp_client
E (64210) ESP_RTSP_CLIENT: Can not start RTSP task!
W (64210) RTSP_SERVICE: Already stopped
I (64220) RTSP_SERVICE: RTSP_STATE_TEARDOWN
E (64220) RTSP_SERVICE: Fail to start RTSP
esp> start 2 rtsp://192.168.0.132:554/live
W (75870) VID_CODEC_REG: Codec mjpeg(hw) already existed
W (75870) VID_CODEC_REG: Codec mjpeg(sw) already existed
W (75870) VID_CODEC_REG: Codec h264(sw) already existed
I (75880) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Nov 29 2024-11:42:50
E (75900) MEDIA_OS: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first

E (75910) MEDIA_OS: Error creating RestrictedPinnedToCore rtsp_client
E (75920) ESP_RTSP_CLIENT: Can not start RTSP task!
W (75930) RTSP_SERVICE: Already stopped
I (75930) RTSP_SERVICE: RTSP_STATE_TEARDOWN
E (75930) RTSP_SERVICE: Fail to start RTSP
esp> start 2 rtsp://192.168.0.132:554/live
W (82800) VID_CODEC_REG: Codec mjpeg(hw) already existed
W (82800) VID_CODEC_REG: Codec mjpeg(sw) already existed
W (82800) VID_CODEC_REG: Codec h264(sw) already existed
I (82810) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Nov 29 2024-11:42:50
E (82820) MEDIA_OS: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first

TempoTian, 2024-12-02:
Patch located here: https://github.com/espressif/esp-adf/tree/master/idf_patches
Select IDF matched version, you can manually copy the content into the header if patch fail.

ManikandanVaradharaj, 2024-12-02:
@TempoTian How could i solve this error

ManikandanVaradharaj, 2024-12-03:
@TempoTian i tired the commands to set the idf freertos patch,but it getting some internal errors.
[image]

ManikandanVaradharaj, 2024-12-03:
@TempoTian After i did the patch requirement,now its connecting with the rtsp but video data is not getting in the display.. esp> start 2 rtsp://192.168.0.132:554/live
I (68044) ESP_RTSP_CLIENT: esp_rtsp ver: 1.1.0 compile date: Nov 29 2024-11:42:50
I (68044) ESP_RTSP_CLIENT: RTSP Client URL:rtsp://192.168.0.132:554/live
I (68044) ESP_RTSP_CLIENT: Connecting...
                                                                                                                                             I (68094) ESP_RTSP_CLIENT: TX:

OPTIONS rtsp://192.168.0.132:554/live RTSP/1.0
CSeq: 0
User-Agent: ESP32-RTSP-client 1.1.0

I (68114) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 0
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER

esp> I (68134) ESP_RTSP_CLIENT: TX:

DESCRIBE rtsp://192.168.0.132:554/live RTSP/1.0
CSeq: 1
User-Agent: ESP32-RTSP-client 1.1.0
Accept: application/sdp

I (68144) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 1
Content-Type: application/sdp
Content-Base: rtsp://192.168.0.132:554
Content-Length: 416

v=0
o=- 10136791 0 IN IP4 192.168.0.132
s=ameba
c=IN IP4 192.168.0.134
t=0 0
m=video 0 RTP/AVP 96
a=rtpmap:96 H264/90000
a=control:streamid=0
a=fmtp:96 packetization-mode=0
m=audio 0 RTP/AVP 97
a=rtpmap:97 mpeg4-generic/8000/1
a=fmtp:97 streamtype=5; profile-level-id=15; mode=AAC-hbr; config=1588; sizeLength=13; indexLength=3; indexDeltaLength=3; constantDuration=1024; Profile=1
a=control:streamid=1

Get video codec 96
Get video location streamid=0
Get audio codec 97
Get audio codec 129 sample rate 8000 channel 1
Get audio location streamid=1
I (68204) ESP_RTSP_CLIENT: TX:

SETUP rtsp://192.168.0.132:554/streamid=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=27550-27551
CSeq: 2
User-Agent: ESP32-RTSP-client 1.1.0

I (68234) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 2
Session: 10136791;timeout=60
Transport: RTP/AVP/UDP;unicast;client_port=27550-27551;server_port=55608-55609

I (68244) ESP_RTSP_CLIENT: RTSP Server vrport 55608
Session timeout 40000
I (68254) RTSP_SERVICE: RTSP_STATE_SETUP
I (68254) ESP_RTSP_CLIENT: TX:

SETUP rtsp://192.168.0.132:554/streamid=1 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=31877-31878
CSeq: 3
User-Agent: ESP32-RTSP-client 1.1.0

I (68274) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 3
Session: 10136791;timeout=60
Transport: RTP/AVP/UDP;unicast;client_port=31877-31878;server_port=55608-55609

I (68294) ESP_RTSP_CLIENT: RTSP Server arport 55608
Session timeout 40000
I (68304) RTSP_SERVICE: RTSP_STATE_SETUP
I (68304) ESP_RTSP_CLIENT: TX:

PLAY rtsp://192.168.0.132:554 RTSP/1.0
CSeq: 4
User-Agent: ESP32-RTSP-client 1.1.0
Session: 10136791

I (68314) ESP_RTSP_CLIENT: RX:

RTSP/1.0 200 OK
CSeq: 4
Session: 10136791

Open with no ADTS header...1 8000..............
I (68324) H264_DEC.SW: tinyh264 version: 1c7f584 
I (68334) RTSP_SERVICE: RTSP_STATE_PLAY
E (68354) H264_DEC: ACCESS UNIT BOUNDARY CHECK
E (68354) H264_DEC.SW: Serious error in decoding, failed to activate param sets
I (68364) RTSP_SERVICE: Video resolution 0x0
convert set to 0x4ff40c58
E (68384) RTSP_SERVICE: Fail to allocate color convert out buffer
I (68384) RTSP_SERVICE: video render task exited
W (70344) ESP_RTP: drop 119

W (72084) ESP_RTP: drop 221

W (72594) ESP_RTP: drop 249

W (76094) ESP_RTP: drop 504

W (79144) ESP_RTP: drop 670

W (80424) ESP_RTP: drop 763

W (81494) ESP_RTP: drop 825

W (81584) ESP_RTP: drop 831

W (81624) ESP_RTP: drop 833

I (110484) ESP_RTSP_CLIENT: TX:

OPTIONS rtsp://192.168.0.132:554 RTSP/1.0
CSeq: 5
User-Agent: ESP32-RTSP-client 1.1.0
Session: 10136791

I (113504) ESP_RTSP_CLIENT: TX:

TEARDOWN rtsp://192.168.0.132:554 RTSP/1.0
CSeq: 5
User-Agent: ESP32-RTSP-client 1.1.0
Session: 10136791

E (113514) RTSP_SERVICE: Video data exceed queue size 2261
I (113514) RTSP_SERVICE: audio render task exit

TempoTian, 2024-12-03:
Currently we only support baseline H264 decode, you can change the push video format to baseline then retry.

ManikandanVaradharaj, 2024-12-03:
@TempoTian Now the data will get in the display to play video or not?

TempoTian, 2024-12-03:
Video data have been received, but decode error, so can not display. You can change format to baseline, so that decode can decode it out, then it can display OK.

[74 more replies]