espressif/esp-adf #1321: rtsp client in esp32_p4

ManikandanVaradharaj, 2024-11-25:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**

the rtsp protocol can implement in esp32_p4 display board can be easy to display the live stream video

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

ManikandanVaradharaj, 2024-11-25:
I'm using esp32_p4 module in esp_brookesia_phone project i added the esp_adf repo and trying to implement rtsp.but I'm not getting proper setup and necessary libraries for the module.could u please suggest me which are the libraries are needed and how the logic will work on esp32_p4 module.

TempoTian, 2024-11-26:
Currently RTSP is implemented in esp_media_protocol and it only dependent on media_lib_sal.
These 2 libraries can be found in:
https://github.com/espressif/esp-adf-libs/tree/master/esp_media_protocols
https://github.com/espressif/esp-adf-libs/tree/master/media_lib_sal

If you use IDF and not ADF you can copy these 2 library and add CMakeLists.txt for them.

If you use ADF, you can modify esp_rtsp example add UI related things into it, also you can see the code to check the API usage:
https://github.com/espressif/esp-adf/tree/master/examples/protocols/esp-rtsp

ManikandanVaradharaj, 2024-11-26:
Now,I'm using esp32_p4 for the esp_devkits repo inside esp_brookesia_phone i pasted the two libraries esp_media_protocal and media_lib_sal.but there i can only get esp_rtsp.h file in esp_media_protocol.inside .obj only present couldn't view the source file and I'm not getting idea how to get the example code for rtsp.

TempoTian, 2024-11-26:
Example code is https://github.com/espressif/esp-adf/tree/master/examples/protocols/esp-rtsp RTSP protocol is in libesp_media_protocol.a there is no open source for it yet.

ManikandanVaradharaj, 2024-11-26:
I installed adf in idf and i tired with example rtsp code but it shows the error In CMakelist and sdkconfig file is not present for esp32p4.

TempoTian, 2024-11-26:
You need export ADF_PATH=XXXXX/esp-adf-internal then retry

ManikandanVaradharaj, 2024-11-26:
which version of idf and adf should i use

TempoTian, 2024-11-26:
IDF v5.3 or above to use esp32p4, ADF can use master branch.

ManikandanVaradharaj, 2024-11-26:
I couldn't install esp-adf-internal file

ManikandanVaradharaj, 2024-11-26:
after i did the WiFi idf_coponent.yml..I'm couldn't set-taget for esp32p4.

Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32p4, new sdkconfig will be created.
Running cmake in directory /home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/smartbuild/.espressif/python_env/idf5.3_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32p4 -DCCACHE_ENABLE=0 /home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp"...
CMake Error at CMakeLists.txt:5 (include):
  include could not find requested file:

    /home/smartbuild/esp/esp-adf/esp-adf-internal/CMakeLists.txt

-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/smartbuild/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/smartbuild/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/smartbuild/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32p4
HINT: Component "espressif/usb_stream" has suitable versions for other targets: "esp32s2", "esp32s3". Is your current target "esp32p4" set correctly?
HINT: Component "espressif/usb_stream" has suitable versions for other targets: "esp32s2", "esp32s3". Is your current target "esp32p4" set correctly?
HINT: Component "espressif/usb_stream" has suitable versions for other targets: "esp32s2", "esp32s3". Is your current target "esp32p4" set correctly?
HINT: Component "espressif/usb_stream" has suitable versions for other targets: "esp32s2", "esp32s3". Is your current target "esp32p4" set correctly?
Processing 9 dependencies:
[1/9] esp32-camera (4335c93ec462a379aa2a3d655b8834972d725190)
[2/9] espressif/cmake_utilities (0.5.3)
[3/9] espressif/eppp_link (0.2.0)
[4/9] espressif/esp32-camera (2.0.15)
[5/9] espressif/esp_hosted (0.0.26)
[6/9] espressif/esp_serial_slave_link (1.1.0)
[7/9] espressif/esp_wifi_remote (0.5.1)
[8/9] espressif/usb_stream (1.3.2)
[9/9] idf (5.3.1)
CMake Error at /home/smartbuild/esp/esp-idf/tools/cmake/component.cmake:251 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "esp32-camera" candidates: "esp32-camera, espressif__esp32-camera"

Call Stack (most recent call first):
  /home/smartbuild/esp/esp-idf/tools/cmake/build.cmake:586 (__component_get_requirements)
  /home/smartbuild/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp/build/log/idf_py_stderr_output_225011 and /home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp/build/log/idf_py_stdout_output_225011

ManikandanVaradharaj, 2024-11-26:
I didn't got any esp-adf-internal folder inside esp-adf.

TempoTian, 2024-11-26:
Following is my build steps, it build OK after mark some code:
```c
2001  cd ~/adf-commit/esp-adf-internal/examples/protocols/esp-rtsp
 2004  idf.py add-dependency "espressif/esp_wifi_remote^0.5.1"
 2005  export ADF_PATH=/home/tempo/adf-commit/esp-adf-internal/
 2006  rm -rf build/
 2007  idf.py set-target esp32p4
 2008  idf.py build
 2009  idf.py menuconfig --> select P4 Function Board
 2010  idf.py build
 2011  code /home/tempo/adf-commit/esp-adf-internal/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal_camera.c --> Comment some GPIO related code
 2012  idf.py build
 2013  code /home/tempo/adf-commit/esp-adf-internal/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal_lcd.c --> Comment lcd init function
 2014  idf.py build

```

ManikandanVaradharaj, 2024-11-26:
here i cloned esp-adf.that location is like /home/smartbuild/esp/esp-adf/examples/protocols/esp-rtsp.

ManikandanVaradharaj, 2024-11-26:
The esp-adf-internal not even found in the github repository also.

ManikandanVaradharaj, 2024-11-26:
Can you please tell me the step by step process of install adf and implement the things for esp32p4 module.

ManikandanVaradharaj, 2024-11-27:
Once again i cloned esp-adf master but i don't got any named folder like esp-adf-internal

smartbuild@smartbuild-Aspire-A314-35:~$ git clone --recursive https://github.com/espressif/esp-adf.git
Cloning into 'esp-adf'...
remote: Enumerating objects: 18572, done.
remote: Counting objects: 100% (5359/5359), done.
remote: Compressing objects: 100% (1362/1362), done.
remote: Total 18572 (delta 4294), reused 4586 (delta 3964), pack-reused 13213 (from 1)
Receiving objects: 100% (18572/18572), 126.87 MiB | 20.58 MiB/s, done.
Resolving deltas: 100% (12360/12360), done.
Updating files: 100% (2223/2223), done.
Submodule 'components/esp-adf-libs' (https://github.com/espressif/esp-adf-libs) registered for path 'components/esp-adf-libs'
Submodule 'components/esp-sr' (https://github.com/espressif/esp-sr.git) registered for path 'components/esp-sr'
Submodule 'esp-idf' (https://github.com/espressif/esp-idf) registered for path 'esp-idf'
Cloning into '/home/smartbuild/esp-adf/components/esp-adf-libs'...
remote: Enumerating objects: 3986, done.        
remote: Counting objects: 100% (1492/1492), done.        
remote: Compressing objects: 100% (772/772), done.        
remote: Total 3986 (delta 631), reused 1450 (delta 589), pack-reused 2494 (from 1)        
Receiving objects: 100% (3986/3986), 648.36 MiB | 16.16 MiB/s, done.
Resolving deltas: 100% (1736/1736), done.
Cloning into '/home/smartbuild/esp-adf/components/esp-sr'...
remote: Enumerating objects: 5289, done.        
remote: Counting objects: 100% (1550/1550), done.        
remote: Compressing objects: 100% (608/608), done.        
remote: Total 5289 (delta 970), reused 1450 (delta 901), pack-reused 3739 (from 1)        
Receiving objects: 100% (5289/5289), 235.03 MiB | 20.74 MiB/s, done.
Resolving deltas: 100% (3015/3015), done.
Cloning into '/home/smartbuild/esp-adf/esp-idf'...
remote: Enumerating objects: 683702, done.        
remote: Counting objects: 100% (19255/19255), done.        
remote: Compressing objects: 100% (6336/6336), done.        
remote: Total 683702 (delta 12654), reused 18394 (delta 12269), pack-reused 664447 (from 1)        
Receiving objects: 100% (683702/683702), 301.35 MiB | 19.90 MiB/s, done.
Resolving deltas: 100% (502892/502892), done.
Submodule path 'components/esp-adf-libs': checked out '891d0b0fe7a04ddf5ddbffc93bd828ce58cc942e'
Submodule path 'components/esp-sr': checked out '394aae67b578cf45ed13dad9c2e0c428f5d599f0'
Submodule path 'esp-idf': checked out 'c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'esp-idf/components/bt/controller/lib_esp32'
Submodule 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib' (https://github.com/espressif/esp32c2-bt-lib.git) registered for path 'esp-idf/components/bt/controller/lib_esp32c2/esp32c2-bt-lib'
Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) registered for path 'esp-idf/components/bt/controller/lib_esp32c3_family'
Submodule 'components/bt/controller/lib_esp32c5/esp32c5-bt-lib' (https://github.com/espressif/esp32c5-bt-lib.git) registered for path 'esp-idf/components/bt/controller/lib_esp32c5/esp32c5-bt-lib'
Submodule 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib' (https://github.com/espressif/esp32c6-bt-lib.git) registered for path 'esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
Submodule 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib' (https://github.com/espressif/esp32h2-bt-lib.git) registered for path 'esp-idf/components/bt/controller/lib_esp32h2/esp32h2-bt-lib'
Submodule 'components/bt/esp_ble_mesh/lib/lib' (https://github.com/espressif/esp-ble-mesh-lib.git) registered for path 'esp-idf/components/bt/esp_ble_mesh/lib/lib'
Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'esp-idf/components/bt/host/nimble/nimble'
Submodule 'components/cmock/CMock' (https://github.com/ThrowTheSwitch/CMock.git) registered for path 'esp-idf/components/cmock/CMock'
Submodule 'components/esp_coex/lib' (https://github.com/espressif/esp-coex-lib.git) registered for path 'esp-idf/components/esp_coex/lib'
Submodule 'components/esp_phy/lib' (https://github.com/espressif/esp-phy-lib.git) registered for path 'esp-idf/components/esp_phy/lib'
Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'esp-idf/components/esp_wifi/lib'
Submodule 'components/heap/tlsf' (https://github.com/espressif/tlsf.git) registered for path 'esp-idf/components/heap/tlsf'
Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'esp-idf/components/json/cJSON'
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'esp-idf/components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'esp-idf/components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'esp-idf/components/mqtt/esp-mqtt'
[... cut ...]
Submodule path 'esp-idf/components/openthread/lib': checked out '203c78501e9a6ea9ca3a929e6f9b6b9691ef16ee'
Submodule path 'esp-idf/components/openthread/openthread': checked out 'f32c18bc0840f400182456e58ae3900fc2fb4af7'
Submodule path 'esp-idf/components/protobuf-c/protobuf-c': checked out 'abc67a11c6db271bedbb9f58be85d6f4e2ea8389'
Submodule path 'esp-idf/components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
Submodule path 'esp-idf/components/unity/unity': checked out 'bf560290f6020737eafaa8b5cbd2177c3956c03f'
[image]

ManikandanVaradharaj, 2024-11-27:
@TempoTian hai,could I get the solution of how to workout rtsp in esp32p4.

TempoTian, 2024-11-27:
Which board are you using? If I have the board on hand, maybe can write a example for you.

ManikandanVaradharaj, 2024-11-27:
esp32_p4_function_eve_board.