espressif/esp-adf #1322: esp-rtsp support in esp32p4

ManikandanVaradharaj, 2024-11-27:
I'm trying with rtsp support in esp32p4 module.
In terminal i set the ADF path,i created the dependency for wifi in the esp-rtsp,i changed the menuconfig Audio sal for for esp32p4..still I'm getting the build errors.

Executing action: all (aliases: build)
Running ninja in directory /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32p4
HINT: Component "espressif/usb_stream" has suitable versions for other targets: "esp32s2", "esp32s3". Is your current target "esp32p4" set correctly?
HINT: Component "espressif/usb_stream" has suitable versions for other targets: "esp32s2", "esp32s3". Is your current target "esp32p4" set correctly?
Processing 11 dependencies:
[11/11] idf (5.3.1)
-- Project sdkconfig file /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig
Loading defaults file /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults...
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:5 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO 
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:22 CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM was replaced with CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM 
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:23 CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM was replaced with CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM 
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:24 CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM was replaced with CONFIG_ESP_WIFI_STATIC_TX_BUFFER_NUM 
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:25 CONFIG_ESP32_WIFI_IRAM_OPT was replaced with CONFIG_ESP_WIFI_IRAM_OPT 
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:26 CONFIG_ESP32_WIFI_RX_IRAM_OPT was replaced with CONFIG_ESP_WIFI_RX_IRAM_OPT 
/home/smartbuild/esp-adf/examples/protocols/esp-rtsp/sdkconfig.defaults:32 CONFIG_UDP_RECVMBOX_SIZE was replaced with CONFIG_LWIP_UDP_RECVMBOX_SIZE 
-- Adding linker script /home/smartbuild/esp/esp-idf/components/riscv/ld/rom.api.ld
-- App "esp-rtsp" version: v2.7-26-gae35314d-dirty
-- Adding linker script /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.api.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.rvfp.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.wdt.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.systimer.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.version.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/esp_rom/esp32p4/ld/esp32p4.rom.newlib.ld
-- Adding linker script /home/smartbuild/esp/esp-idf/components/soc/esp32p4/ld/esp32p4.peripherals.ld
CMake Warning (dev) at managed_components/espressif__esp_hosted/CMakeLists.txt:7:
  Syntax Warning in cmake code at column 302

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Chip is esp32p4, /home/smartbuild/esp/esp-adf/components/esp-adf-libs
-- Current board name is CONFIG_ESP32_P4_FUNCTION_EV_BOARD
-- USB_STREAM: 1.3.2
-- Component idf::espressif__esp_hosted will be linked with -Wl,--whole-archive
-- Components: adf_utils app_trace app_update audio_board audio_hal audio_mixer audio_pipeline audio_recorder audio_sal audio_stream av_stream battery_service bluetooth_service bootloader bootloader_support bt cli_console clouds cmock console coredump_upload_service cxx display_service driver dueros_service efuse esp-adf-libs esp-sr esp-tls esp_actions esp_adc esp_app_format esp_bootloader_format esp_codec_dev esp_coex esp_common esp_dispatcher esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_event_cast esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_peripherals esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump espressif__cmake_utilities espressif__eppp_link espressif__esp-dsp espressif__esp32-camera espressif__esp_hosted espressif__esp_serial_slave_link espressif__esp_wifi_remote espressif__jsmn espressif__nghttp espressif__usb_stream esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 input_key_service json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread ota_service partition_table playlist protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport tone_partition ulp unity usb vfs wear_levelling wifi_provisioning wifi_service wpa_supplicant
-- Component paths: /home/smartbuild/esp/esp-adf/components/adf_utils /home/smartbuild/esp/esp-idf/components/app_trace /home/smartbuild/esp/esp-idf/components/app_update /home/smartbuild/esp/esp-adf/components/audio_board /home/smartbuild/esp/esp-adf/components/audio_hal /home/smartbuild/esp/esp-adf/components/audio_mixer /home/smartbuild/esp/esp-adf/components/audio_pipeline /home/smartbuild/esp/esp-adf/components/audio_recorder /home/smartbuild/esp/esp-adf/components/audio_sal /home/smartbuild/esp/esp-adf/components/audio_stream /home/smartbuild/esp/esp-adf/examples/protocols/components/av_stream /home/smartbuild/esp/esp-adf/components/battery_service /home/smartbuild/esp/esp-adf/components/bluetooth_service /home/smartbuild/esp/esp-idf/components/bootloader /home/smartbuild/esp/esp-idf/components/bootloader_support /home/smartbuild/esp/esp-idf/components/bt /home/smartbuild/esp/esp-adf/examples/protocols/components/cli_console /home/smartbuild/esp/esp-adf/components/clouds /home/smartbuild/esp/esp-idf/components/cmock /home/smartbuild/esp/esp-idf/components/console /home/smartbuild/esp/esp-adf/components/coredump_upload_service /home/smartbuild/esp/esp-idf/components/cxx /home/smartbuild/esp/esp-adf/components/display_service /home/smartbuild/esp/esp-idf/components/driver /home/smartbuild/esp/esp-adf/components/dueros_service /home/smartbuild/esp/esp-idf/components/efuse /home/smartbuild/esp/esp-adf/components/esp-adf-libs /home/smartbuild/esp/esp-adf/components/esp-sr /home/smartbuild/esp/esp-idf/components/esp-tls /home/smartbuild/esp/esp-adf/components/esp_actions /home/smartbuild/esp/esp-idf/components/esp_adc /home/smartbuild/esp/esp-idf/components/esp_app_format /home/smartbuild/esp/esp-idf/components/esp_bootloader_format /home/smartbuild/esp/esp-adf/components/esp_codec_dev /home/smartbuild/esp/esp-idf/components/esp_coex /home/smartbuild/esp/esp-idf/components/esp_common /home/smartbuild/esp/esp-adf/components/esp_dispatcher /home/smartbuild/esp/esp-idf/components/esp_driver_ana_cmpr /home/smartbuild/esp/esp-idf/components/esp_driver_cam /home/smartbuild/esp/esp-idf/components/esp_driver_dac /home/smartbuild/esp/esp-idf/components/esp_driver_gpio /home/smartbuild/esp/esp-idf/components/esp_driver_gptimer /home/smartbuild/esp/esp-idf/components/esp_driver_i2c /home/smartbuild/esp/esp-idf/components/esp_driver_i2s /home/smartbuild/esp/esp-idf/components/esp_driver_isp /home/smartbuild/esp/esp-idf/components/esp_driver_jpeg /home/smartbuild/esp/esp-idf/components/esp_driver_ledc /home/smartbuild/esp/esp-idf/components/esp_driver_mcpwm /home/smartbuild/esp/esp-idf/components/esp_driver_parlio /home/smartbuild/esp/esp-idf/components/esp_driver_pcnt /home/smartbuild/esp/esp-idf/components/esp_driver_ppa /home/smartbuild/esp/esp-idf/components/esp_driver_rmt /home/smartbuild/esp/esp-idf/components/esp_driver_sdio /home/smartbuild/esp/esp-idf/components/esp_driver_sdm /home/smartbuild/esp/esp-idf/components/ [... cut ...]
-- Build files have been written to: /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/build
ninja: error: '/home/smartbuild/esp/esp-adf/components/esp-adf-libs/esp_h264/lib/esp32p4/libesp_h264.a', needed by 'esp-rtsp.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/build/log/idf_py_stderr_output_8426 and /home/smartbuild/esp-adf/examples/protocols/esp-rtsp/build/log/idf_py_stdout_output_8426

ManikandanVaradharaj, 2024-11-27:
esp_h264 has only include for s3 module not available for p4 module.This thing making an build error

ManikandanVaradharaj, 2024-11-27:
In adf examples for rtsp.In rtsp_service.c file there is no function for receive video,if i'm using rtsp client means how can i get the live video stream.

ManikandanVaradharaj, 2024-11-27:
I want play a rtsp live video using my url in esp32p4

Lextan276, 2025-04-25:
@TempoTian you write about pull or push, but is it posible to create server and stream video from esp32-p4 camera there, or i only can connect to server located in my network and push video?

TempoTian, 2025-04-27:
Yes you can also create RTSP server, but current demo have not adapter for P4 yet. You can add a feature request, we will develop it soon.