espressif/esp-adf #1325: SOLVED: 20% faster fp32 FFT in ASM for ESP32 and ESP32-S3

f4lc0n-asm, 2024-12-09:
Hello,

here are the 20% faster ASM drop-in replacements for [dsps_fft2r_fc32_ae32_.S](https://github.com/espressif/esp-dsp/blob/master/modules/fft/float/dsps_fft2r_fc32_ae32_.S) and [dsps_fft2r_fc32_aes3_.S](https://github.com/espressif/esp-dsp/blob/master/modules/fft/float/dsps_fft2r_fc32_aes3_.S) from ESP-DSP repository.

I achieved this by:
- Replacing `slli / add.n` pair with `addx8` on 3 occasions.
- Carefully reordering instructions for much fewer pipeline stalls and preventing the assembler to automatically insert a NOP in front of a hw loop so that the first instruction in the hw loop entirely fits within a naturally aligned four byte region. Consult **Xtensa ISA Summary** (`LOOPNEZ` description) for more info - it's vital for achieving the maximum loop speed!
- Removing some superfluous instructions (forgotten debugging stuff).

The following table shows the number of cycles for various FFT routines:
```
   N |   ANSI |  ESP32 | ESP32 O|ESP32-S3|ESP32-S3 O
  16 |   1277 |    977 |    817 |    930 |    784
  32 |   2972 |   2318 |   1934 |   2207 |   1853
  64 |   6835 |   5379 |   4483 |   5124 |   4290
 128 |  15514 |  12264 |  10216 |  11689 |   9767
 256 |  34785 |  27565 |  22957 |  26286 |  21932
 512 |  77160 |  61234 |  50994 |  58419 |  48689
1024 | 169583 | 134711 | 112183 | 128568 | 107062
2048 | 369782 | 293948 | 244796 | 280637 | 233531
4096 | 800893 | 636993 | 530497 | 608322 | 505920

```
Note 1: Passing the 3rd argument `float* dsps_fft_w_table_fc32` to FFT functions is actually a good idea. Because when you do parallel computations of say 2 FFTs with different Ns, you need 2 different table pointers for them. The table pointer cannot be thus hardcoded in the FFT functions, as attempted.

Note 2: ESP32-S3 ASM version is just 4% faster than ESP32 ASM version because of some pipeline stalls which cannot be avoided.

Cheers!

f4lc0n

[fft2r_fp32_xtensa.zip](https://github.com/user-attachments/files/18064574/fft2r_fp32_xtensa.zip) (7-Zip)

dmitry1945, 2025-03-20:
Hi @f4lc0n-asm 
I have updated the esp-dsp with your changes: https://github.com/espressif/esp-dsp/blob/master/modules/fft/float/dsps_fft2r_fc32_ae32_.S

Please let me know if it's fine for you.
And,
thank you very much!

f4lc0n-asm, 2025-03-21:
Hello @dmitry1945, you are very welcome! The code is sitting there comfortably :)

An ESP32-P4 devkit just landed on my desk. As an introductory project I am developing a fast float FIR (without the ESP32-P4 DSP extension). If everything goes well, I will publish it during the weekend. Still don't have specifications of ESP32-P4 DSP extension so this may be just a plain exercise…

What do you think about my [3-cycle/tap float FIR routines for ESP32](https://github.com/espressif/esp-dsp/issues/92)?