espressif/esp-adf #1327: Voip test error

liujkek22, 2024-12-16:
Question:
Testing the ADF VoIP example using ESP-IDF 5.3 + ESP-ADF 2.6 + FreeSWITCH (Windows version, deployed in the local area network) + MicroSip software, with ESP32-S3-Korvo-2 V3.0 (connected to the local area network).

During the test between ESP32-S3 and MicroSip with FreeSWITCH, the ESP32 can successfully call MicroSip. However, when MicroSip tries to call the ESP32, the ESP32 does not receive the call. Logs only show "keep-alive" activity.

Could anyone provide suggestions or guidance to resolve this issue? Thank you!

liujkek22, 2024-12-16:
### **add some infor for frreswitch,ESP32ip:192.168.137.214,and local ip172.40.202.176,**
2024-12-16 13:16:02.815589 98.89% [INFO] sofia_glue.c:1659 sofia/internal/1002@192.168.137.214:10276 sending invite call-id: (null)
2024-12-16 13:16:07.015930 98.89% [NOTICE] sofia.c:8736 Hangup sofia/internal/1001@172.40.202.176 [CS_EXECUTE] [NORMAL_UNSPECIFIED]
2024-12-16 13:16:07.035871 98.89% [NOTICE] switch_ivr_bridge.c:1023 Hangup loopback/voicemail-a [CS_EXCHANGE_MEDIA] [ORIGINATOR_CANCEL]
2024-12-16 13:16:07.035871 98.89% [NOTICE] mod_loopback.c:585 Hangup loopback/voicemail-b [CS_EXECUTE] [ORIGINATOR_CANCEL]
2024-12-16 13:16:07.035871 98.89% [NOTICE] switch_core_session.c:1762 Session 3 (sofia/internal/1001@172.40.202.176) Ended
2024-12-16 13:16:07.035871 98.89% [NOTICE] switch_core_session.c:1766 Close Channel sofia/internal/1001@172.40.202.176 [CS_DESTROY]
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1762 Session 5 (loopback/voicemail-a) Ended
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1766 Close Channel loopback/voicemail-a [CS_DESTROY]
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1762 Session 6 (loopback/voicemail-b) Ended
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1766 Close Channel loopback/voicemail-b [CS_DESTROY]
2024-12-16 13:16:32.005978 99.20% [NOTICE] switch_ivr_originate.c:3820 Hangup sofia/internal/1002@192.168.137.214:10276 [CS_CONSUME_MEDIA] [NO_ANSWER]
2024-12-16 13:16:32.005978 99.20% [NOTICE] switch_ivr_originate.c:3059 Cannot create outgoing channel of type [user] cause: [NO_ANSWER]
**2024-12-16 13:16:32.005978 99.20% [INFO] mod_dptools.c:3635 Originate Failed.  Cause: NO_ANSWER**
EXECUTE [depth=0] sofia/internal/1001@172.40.202.176 answer()

liujkek22, 2024-12-16:
### **add some infor for frreswitch,ESP32ip:192.168.137.214,and local ip172.40.202.176,**
wee can see that esp32 no answer,ESP32 just keep alive and "Sending keep-alive to server"
2024-12-16 13:16:02.815589 98.89% [INFO] sofia_glue.c:1659 sofia/internal/1002@192.168.137.214:10276 sending invite call-id: (null)
2024-12-16 13:16:07.015930 98.89% [NOTICE] sofia.c:8736 Hangup sofia/internal/1001@172.40.202.176 [CS_EXECUTE] [NORMAL_UNSPECIFIED]
2024-12-16 13:16:07.035871 98.89% [NOTICE] switch_ivr_bridge.c:1023 Hangup loopback/voicemail-a [CS_EXCHANGE_MEDIA] [ORIGINATOR_CANCEL]
2024-12-16 13:16:07.035871 98.89% [NOTICE] mod_loopback.c:585 Hangup loopback/voicemail-b [CS_EXECUTE] [ORIGINATOR_CANCEL]
2024-12-16 13:16:07.035871 98.89% [NOTICE] switch_core_session.c:1762 Session 3 (sofia/internal/1001@172.40.202.176) Ended
2024-12-16 13:16:07.035871 98.89% [NOTICE] switch_core_session.c:1766 Close Channel sofia/internal/1001@172.40.202.176 [CS_DESTROY]
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1762 Session 5 (loopback/voicemail-a) Ended
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1766 Close Channel loopback/voicemail-a [CS_DESTROY]
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1762 Session 6 (loopback/voicemail-b) Ended
2024-12-16 13:16:07.056978 98.89% [NOTICE] switch_core_session.c:1766 Close Channel loopback/voicemail-b [CS_DESTROY]
2024-12-16 13:16:32.005978 99.20% [NOTICE] switch_ivr_originate.c:3820 Hangup sofia/internal/1002@192.168.137.214:10276 [CS_CONSUME_MEDIA] [NO_ANSWER]
2024-12-16 13:16:32.005978 99.20% [NOTICE] switch_ivr_originate.c:3059 Cannot create outgoing channel of type [user] cause: [NO_ANSWER]
**2024-12-16 13:16:32.005978 99.20% [INFO] mod_dptools.c:3635 Originate Failed.  Cause: NO_ANSWER**
EXECUTE [depth=0] sofia/internal/1001@172.40.202.176 answer()

TempoTian, 2024-12-17:
Better to fetch wireshark network packet and uart log all together, then I can help you to check the possible issues.

closed 2025-02-17 by jason-mao