espressif/esp-adf #1333: 通过pipline获取ringbuf对象写入数据后结尾会出现两秒喇叭尖啸

bailiqun, 2024-12-24:
通过建立一个管道,然后in是一个原始输入节点, 然后输入的是pcm原始数据,建立管道后他获取输入的rb来控制发声.
    现在是可以发声的,也按照预期执行了,但是声音结尾会出现哔的尖啸声,不知道什么情况.

  raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
  raw_cfg.type = AUDIO_STREAM_READER;
  raw_cfg.out_rb_size = 960000;
  in_stream = raw_stream_init(&raw_cfg);
    
    audio_decoder_t auto_decode[] = {
        DEFAULT_ESP_OGG_DECODER_CONFIG(),
        DEFAULT_ESP_MP3_DECODER_CONFIG(),
        DEFAULT_ESP_WAV_DECODER_CONFIG(),
        DEFAULT_ESP_PCM_DECODER_CONFIG(),
        DEFAULT_ESP_AAC_DECODER_CONFIG(),
        DEFAULT_ESP_M4A_DECODER_CONFIG(),
        DEFAULT_ESP_OPUS_DECODER_CONFIG(),
        DEFAULT_ESP_FLAC_DECODER_CONFIG()
    };
    esp_decoder_cfg_t auto_dec_cfg = DEFAULT_ESP_DECODER_CONFIG();
    decoder_el = esp_decoder_init(&auto_dec_cfg, auto_decode, sizeof(auto_decode) / sizeof(audio_decoder_t));

    rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_cfg.src_rate = 16000;
    rsp_cfg.src_ch = 1;
    rsp_cfg.dest_rate = 16000;
    rsp_cfg.dest_ch = 1;
    rsp_cfg.stack_in_ext = true;
    rsp_handle = rsp_filter_init(&rsp_cfg);

    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    *pipeline = audio_pipeline_init(&pipeline_cfg);
    audio_pipeline_register(*pipeline, in_stream, "in");
    audio_pipeline_register(*pipeline, decoder_el, "codec");
    audio_pipeline_register(*pipeline, rsp_handle, "filter");
    audio_pipeline_register(*pipeline, i2s_writer, "i2s_writer");
    const char *link_tag_base[4] = {"in", "codec", "filter", "i2s_writer"};
    audio_pipeline_link(*pipeline, &link_tag_base[0], 4);
以上为pipline构建部分代码
----------------------------------------------------------

下面是通过表签名获取ringbuff从而来写入pcm数据的代码
static esp_err_t  cli_hi(esp_periph_handle_t periph, int argc, char *argv[]){

    // audio_element_handle_t el_out = audio_pipeline_get_el_by_tag(pipline, "filter");
    // rsp_filter_change_src_info(el_out, 16000, 1, 16);
    
    audio_element_handle_t el_in = audio_pipeline_get_el_by_tag(pipline, "in");
    ringbuf_handle_t in_rb = audio_element_get_output_ringbuf(el_in);

    int16_t* buffer = (int16_t*)voice_baby_here; 
    size_t buffer_length = sizeof(voice_baby_here)*sizeof(int16_t);
    size_t audio_play_bytes = 0;
    while(true){
        size_t i2s_write_bytes = fmax(0, fmin(buffer_length - audio_play_bytes, 1024));
        rb_write(in_rb, (char*)((int16_t*)(buffer + audio_play_bytes/2)), i2s_write_bytes, pdMS_TO_TICKS(1000));
        rb_done_write(in_rb);
        audio_play_bytes += i2s_write_bytes;
        int rb_available = rb_bytes_filled(in_rb);
        ESP_LOGI("####", "%d / %d  / %d", audio_play_bytes, buffer_length, rb_available);
        if(audio_play_bytes >= buffer_length){
            break;
        }
    }
    return ESP_OK;
}

这样用主要有时候网络获取数据并不均匀,还需要等待导致卡顿, 获取管道rb更灵活一些.麻烦大佬帮忙解释解释

TempoTian, 2024-12-30:
你可以在写入最后一笔数据后再写入一笔0 data看看是否还有尖啸声，有的话可以检查一下原始的输入数据
你的代码本身是支持边写入边播放的，只是你做到cli里面了，导致loop卡住了cli，你可以另外创建一个task来做往ring buffer写数据的动作

bailiqun, 2025-01-01:
十分感谢, 我试试看

closed 2025-01-01 by bailiqun