espressif/esp-adf #1334: I2C fails with hw NACK on ES8388 while I2S stream is running

RaresCon, 2024-12-24:
## Environment

- Audio development kit: none - ESP32-WROOM-32D with ES8388 breakout board
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): -
- [Required] Module or chip used:  ESP32-WROOM-32D
- [Required] IDF version: 5.3.1
- [Required] ADF version: e9f72307c73159711b33775e543df22d6e607449 (v2.7.1)
- Build system: idf.py
- [Required] Running log: 
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF c8fc5f64-dirty 2nd stage bootloader
I (31) boot: compile time Dec 24 2024 20:01:10
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00300000
I (92) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=31d38h (204088) map
I (175) esp_image: segment 1: paddr=00041d60 vaddr=3ffbdb60 size=04ca4h ( 19620) load
I (182) esp_image: segment 2: paddr=00046a0c vaddr=40080000 size=0960ch ( 38412) load
I (198) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=bb420h (767008) map
I (461) esp_image: segment 4: paddr=0010b448 vaddr=4008960c size=11620h ( 71200) load
I (502) boot: Loaded app from partition at offset 0x10000
I (502) boot: Disabling RNG early entropy source...
I (515) cpu_start: Multicore app
I (523) cpu_start: Pro cpu start user code
I (523) cpu_start: cpu freq: 160000000 Hz
I (523) app_init: Application information:
I (526) app_init: Project name:     bt_sink_demo
I (531) app_init: App version:      94a59cc2-dirty
I (537) app_init: Compile time:     Dec 24 2024 20:43:05
I (543) app_init: ELF file SHA256:  d8acfe86e...
I (548) app_init: ESP-IDF:          c8fc5f64-dirty
I (553) efuse_init: Min chip rev:     v0.0
I (558) efuse_init: Max chip rev:     v3.99
I (563) efuse_init: Chip rev:         v3.1
I (568) heap_init: Initializing. RAM available for dynamic allocation:
I (575) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (581) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (587) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (593) heap_init: At 3FFCC628 len 000139D8 (78 KiB): DRAM
I (600) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (606) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (612) heap_init: At 4009AC2C len 000053D4 (20 KiB): IRAM
I (620) spi_flash: detected chip: generic
I (623) spi_flash: flash io: dio
I (639) coexist: coex firmware version: 4482466
I (640) main_task: Started on CPU0
I (640) main_task: Calling app_main()
[... 72 lines cut ...]
### Actual Behavior

Getting HW NACKs for any I2C command after the initialization of the I2S stream.

### Steps to Reproduce

1. Use ESP32-WROOM-32D with ES8388
2. Try to change the volume after the device is connected through BT or try to change the volume in the code

// If possible, attach a picture of your setup/wiring here.

[image]

### Code to Reproduce This Issue

The code is the one from the example, the code can be found on my [fork](https://github.com/RaresCon/esp-adf/tree/new_local_branch), the modifications for my setup can be found in the HEAD commit.

## Debug Logs

The logs can be found in the running log section.

## Other Items If Possible

- [ * ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
[bt_sink_demo.zip](https://github.com/user-attachments/files/18241663/bt_sink_demo.zip)

RaresCon, 2025-01-27:
Is there any update for this issue?

ALToast, 2025-02-08:
Hi @RaresCon 
Software solution reference: [bugfix/improve_es8311_i2c_stability](https://github.com/espressif/esp-adf/commit/8fc8a4a639851f3d2fba62145d511e954aa75069)
Hardware solution: Add 33pF/33Ω filters to the CLK and SDA lines near the ES8311 I2C interface.

RaresCon, 2025-02-08:
> Hi @RaresCon 
> Software solution reference: [bugfix/improve_es8311_i2c_stability](https://github.com/espressif/esp-adf/commit/8fc8a4a639851f3d2fba62145d511e954aa75069)
> Hardware solution: Add 33pF/33Ω filters to the CLK and SDA lines near the ES8311 I2C pin.

Hi @ALToast 
Thanks for your response. I am using a ES8388, is there a software solution for it as well? I am sure the registers are not the same as on ES8311.
I will also try the filtering solution.

ALToast, 2025-02-10:
Hi @RaresCon 
In terms of software, the ES8388 does not have registers for the I2C. Please try reducing the I2C frequency and adjusting the [glitch_ignore_cnt](https://github.com/espressif/esp-adf/blob/master/components/esp_peripherals/driver/i2c_bus/i2c_bus_v2.c#L120) parameter.
If you have tried a hardware solution, please let me know the results. Thanks

RaresCon, 2025-02-16:
> Hi [@RaresCon](https://github.com/RaresCon) In terms of software, the ES8388 does not have registers for the I2C. Please try reducing the I2C frequency and adjusting the [...]

Hi @ALToast
I have tried the software solution above, reducing the I2C frequency to 70K, 50K, and 30K, none has worked. I also tried increasing the [glitch_ignore_cnt](https://github.com/espressif/esp-adf/blob/master/components/esp_peripherals/driver/i2c_bus/i2c_bus_v2.c#L120) to 15 while using these frequencies.

Regarding the hardware possible solution, I don't have the caps and resistors yet, but I will try it next week and tell you if I have any success.

GCTechnology, 2025-02-17:
I got the same issue on ES8388 codec.

[image]

It will show up at start, or sometimes when adjusting the volume. It's a random problem.

GCTechnology, 2025-02-17:
> I got the same issue on ES8388 codec.
> 
> [image]
> 
> [image]
> 
> It will show up at start, or sometimes at the volume adjustment. It's a random problem. It basically doesn't affect the playback, but it will affect the volume adjustment. The manifestation is that when an error occurs, only one of the audio channels may have its volume adjusted.

Similar issue on https://github.com/espressif/esp-adf/issues/1334#issuecomment-2663380488 and https://github.com/espressif/esp-idf/issues/14030

ALToast, 2025-02-26:
Hi @GCTechnology 
I have conducted multiple tests using the ESP32-LyraT development board, and the issue mentioned above did not occur on this board. I would like to know if your hardware is based on the [ESP32-LyraT SCH](https://dl.espressif.com/dl/schematics/ESP32-LYRAT_V4.3-20220119.pdf) reference. 
[image]

ALToast, 2025-02-27:
> > Hi [@RaresCon](https://github.com/RaresCon) In terms of software, the ES8388 does not have registers for the I2C. Please try reducing the I2C frequency and adjusting the [glitch_ignore_cnt](https://github.com/espressif/esp-adf/blob/master/components/esp_peripherals/driver/i2c_bus/i2c_bus_v2.c#L120) parameter. If you have tried a hardware solution, please let me know the results. Thanks
> 
> Hi [@ALToast](https://github.com/ALToast) I have tried the software solution above, reducing the I2C frequency to 70K, 50K, and 30K, none has worked. I also tried increasing the [glitch_ignore_cnt](https://github.com/espressif/esp-adf/blob/master/components/esp_peripherals/driver/i2c_bus/i2c_bus_v2.c#L120) to 15 while using these frequencies.
> 
> Regarding the hardware possible solution, I don't have the caps and resistors yet, but I will try it next week and tell you if I have any success. I will try to write a QSPI solution to change the I2C register writing.
> 
> Is there any other way to change the volume of the I2S stream? Could there be a bug in the I2C driver?
Hi @RaresCon 

The issue is caused by the slave chip not returning the ACK signal. If you capture the I2C waveform, you will be able to observe this phenomenon. The low-pass filtering resistors on the I2C pins in the [ESP32-LyraT SCH](https://dl.espressif.com/dl/schematics/ESP32-LYRAT_V4.3-20220119.pdf) are also based on the slave chip's recommendations.

[image]

ALToast, 2025-03-06:
Hi @RaresCon @GCTechnology
Any progress on the issue?

RaresCon, 2025-03-06:
> Hi @RaresCon @GCTechnology
> Any progress on the issue?

Hi @ALToast 
Unfortunetly, no, I had some other issues to resolve and I couldn't get the low value capacitors and resistors yet. I will try next week to buy them somewhere.

GCTechnology, 2025-03-07:
> Hi [@RaresCon](https://github.com/RaresCon) [@GCTechnology](https://github.com/GCTechnology) Any progress on the issue?

So far, no solution has been found. I‘m trying to solve this problem because it was found and posted by the customer as a bug. :(

Rck27, 2025-03-09:
> .

Any updates to this problem?

Rck27, 2025-03-10:
I tried it using LPF and HPF, but the error still persist[image]

Here's some picture of the signal

RaresCon, 2025-03-22:
Hi @ALToast,

I have tried to add the resistors and capacitors on the SDA/SCL lines between the slave board and the ESP-WROOM-32D and it does not solve the problem. I guess it might be because of the length of the connections between the two.

GCTechnology, 2025-03-23:
> Hi [@ALToast](https://github.com/ALToast),
> 
> I have tried to add the resistors and capacitors on the SDA/SCL lines between the slave board and the ESP-WROOM-32D and it does not solve the problem. I guess it might be because of the length of the connections between the two.

We also tried to add resistors to shorten the rising time of SCL to reach the limit of 300ns, it worked. However, it did not completely solve the problem. It merely reduced the frequency of the occurrence of errors.
[image]

Then I add a retry mechanism to the codec transceiver interface to avoid this. Modify `es_write_reg`
 and `es_read_reg `at `es8388.c`.

RaresCon, 2025-03-24:
> > Hi [@ALToast](https://github.com/ALToast),
> > I have tried to add the resistors and capacitors on the SDA/SCL lines between the slave board and the ESP-WROOM-32D and it does not solve the problem. I guess it might be because of the length of the connections between the two.
> 
> We also tried to add resistors to shorten the rising time of SCL to reach the limit of 300ns, it worked in my case. However, it did not completely solve the problem. It merely reduced the frequency of the occurrence of errors. [image]
> 
> Then I add a retry mechanism to the codec transceiver interface to avoid this. Modify `es_write_reg` and `es_read_reg `at `es8388.c`.

So the resistor (100R) has been added inline with the signal of SCL, similar to the LyraT-4.3 schematic?

RaresCon, 2025-03-24:
@ALToast, should this schematic be used instead of LyraT-4.3's? I found this in the user guide for ES8388.

[image]