espressif/esp-adf #1337: 如何判断播放语音流结束

liu0, 2024-12-27:
player = esp_audio_create(&cfg);
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
    xTaskCreate(esp_audio_state_task, "player_task", 4096, cfg.evt_que, 1, NULL);
    // Create readers and add to esp_audio
     tone_stream_cfg_t tone_cfg = TONE_STREAM_CFG_DEFAULT();
     tone_cfg.type = AUDIO_STREAM_READER;
     esp_audio_input_stream_add(player, tone_stream_init(&tone_cfg));

    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.out_rb_size = 50*1024;
    raw_cfg.type = AUDIO_STREAM_WRITER;
    raw_play_handle = raw_stream_init(&raw_cfg);
    esp_audio_input_stream_add(player, raw_play_handle);
    mp3_decoder_cfg_t mp3_dec_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_dec_cfg.task_core = 1;
    esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER, mp3_decoder_init(&mp3_dec_cfg)); 
我用raw_stream_write(raw_play_handle, (char *)&(mqtt_data.chunk), (mqtt_data.chunk_size));  写入然后播放是可以的。但是我想知道通过raw_stream_write写入的音频什么时候播放结束。现在尝试用audio_err_t err = esp_audio_callback_set(player, audio_event_callback, NULL);  void audio_event_callback(esp_audio_state_t *audio, void *ctx) {
    if (audio == NULL) {
        printf("音频状态为空\n");
        return;
    }

    // 根据音频状态输出不同的状态信息
    switch (audio->status) {
        case AUDIO_STATUS_UNKNOWN:
            printf("音频状态: 未知\n");
            break;
        case AUDIO_STATUS_RUNNING:
            printf("音频状态: 播放中\n");
            break;
        case AUDIO_STATUS_PAUSED:
            printf("音频状态: 暂停\n");
            break;
        case AUDIO_STATUS_STOPPED:
            printf("音频状态: 停止\n");
            break;
        case AUDIO_STATUS_FINISHED:
            printf("音频状态: 播放完成\n");
            break;
        case AUDIO_STATUS_ERROR:
            printf("音频状态: 错误, 错误码: %d\n", audio->err_msg);
            break;
        default:
            printf("未知的音频状态\n");
            break;
    }
    // 打印媒体源类型
    printf("媒体源类型: %d\n", audio->media_src);
}这个回调没办法回调打印出结束。。然后通过xTaskCreate(esp_audio_state_task, "player_task", 4096, cfg.evt_que, 1, NULL);这种
static void esp_audio_state_task (void *para)
{
    QueueHandle_t que = (QueueHandle_t) para;
    esp_audio_state_t esp_state = {0};
    while (1) {
        xQueueReceive(que, &esp_state, portMAX_DELAY);
        ESP_LOGE(TAG, "esp_auido status:%x,err:%x\n", esp_state.status, esp_state.err_msg);
        if ((esp_state.status == AUDIO_STATUS_FINISHED)
            || (esp_state.status == AUDIO_STATUS_ERROR)) {
            int time = 0;
            int duration = 0;
            esp_audio_time_get(player, &time);
            esp_audio_duration_get(player, &duration);
            ESP_LOGE(TAG, "[ * ] End of time:%d ms, duration:%d ms", time, duration);
            AUDIO_MEM_SHOW(TAG);
        }
    }
    vTaskDelete(NULL);
}也不行。这两种办法只能打印出tone提示音播放的，也就是直接调用esp_audio_sync_play(player, tone_uri[TONE_TYPE_DINGDONG], 0);这种的可以获取什么时候播放结束。请问通过raw_stream_write写入的音频什么时候播放结束要怎么知道。不断的调用audio_err_t esp_audio_state_get(esp_audio_handle_t handle, esp_audio_state_t *state);吗。请指教

TempoTian, 2024-12-31:
如果你的最前级是raw stream要想让整个pipeline进入Finished状态，你可以在写完最后一笔数据之后写入长度为0的数据，这样对应的element能够进入Done状态后续的element也才能进入done状态
https://github.com/espressif/esp-adf/blob/master/components/audio_pipeline/audio_element.c#L444

liu0, 2025-01-04:
现在是在写入最后一包的时候，调用这个esp_audio_stop(player, TERMINATION_TYPE_DONE);可以在audio_event_callback中检测到完成

closed 2025-01-04 by liu0