espressif/esp-adf #1338: Getting cryptic message when trying to play m4a/aac file using espressif/esp_audio_codec

gamename, 2024-12-28:
## Environment

- Audio development kit: None
- Audio kit version: N/A
- [Required] ESP32-S3-DevKitC-1-N16
- [Required] v5.3.1
- [Required] v2.7-47-g93a56fcd (but that is misleading - see below)
- Build system: idf.py
- [Required] Running log: 
[squawk_log.txt](https://github.com/user-attachments/files/18264551/squawk_log.txt)

- Compiler version 
```xtensa-esp32-elf-gcc --version
xtensa-esp-elf-gcc (crosstool-NG esp-13.2.0_20240530) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.```

- Operating system:     MacOS
- (Windows only) N/A
- Using an IDE?: vscode
- Power supply: USB

## Problem Description
I am trying to play an m4a/aac file that is embedded in my app. I added the [esp_audio_codec](https://components.espressif.com/components/espressif/esp_audio_codec/versions/2.0.3) to my project to help with this.

Attached to my esp32-s3-DevKitC microcontroller is a MAX98357A amp which, in turn, is connected to a small speaker.

### Expected Behavior
The m4a file should play on my speakers

### Actual Behavior
I hear nothing and get a cryptic message in my log saying:
```E (445) M4A_PARSE: Skip this audio track 0 select 137```

### Steps to Reproduce
1. Wire the system as per the pictures (below).
1. run `idf.py build flash monitor`

// If possible, attach a picture of your setup/wiring here.
See the github repo below, it has a picture

### Code to Reproduce This Issue
https://github.com/gamename/test-squawk

## Debug Logs

Did not crash.

## Other Items If Possible
[sdkconfig.txt](https://github.com/user-attachments/files/18264516/sdkconfig.txt)
[test-squawk.elf.txt](https://github.com/user-attachments/files/18264519/test-squawk.elf.txt)

TempoTian, 2024-12-30:
I have check this issue, there is a typo error in the test code, I see same error in you code, https://github.com/gamename/test-squawk/blob/main/main/test-squawk.c#L152 
```c
//simp_dec_all_t *all_cfg = (simp_dec_all_t *)&cfg->dec_cfg;
/simp_dec_all_t *all_cfg = (simp_dec_all_t *)cfg->dec_cfg;
```
Also I check the provided m4a is not supported currently for its both stco and stsz contain no table item. I have converted it for you, you can use it to do verify please rename it to m4a. Meanwhile I see you have not enable SPIRam support, so there may be not enough space to hold the MP4 tables, you can use menuconfig to enable it
```
Component Config --> ESP PSRAM 
```

https://github.com/user-attachments/assets/448f3f74-bad8-4ab2-a4cb-1b93eb6fc0c8

gamename, 2024-12-30:
@TempoTian 

> I have converted it for you
Thank you! That was very kind. How did you convert it? I would like an example of the process so I can use it in the future.

gamename, 2024-12-30:
@TempoTian 

1. I couldn't download the mp4 file you created, so I did a conversion using `vlc` to m4v.
2. I don't have PSRAM, so I cannot enable that.  But, I am using an esp32-s3-DevKitC-1-N16 (i.e. there is 16MB of ram).
3. I made the source code update per your instructions.  The repo at https://github.com/gamename/test-squawk has all the updates.

Everything compiled and ran, but no sound. :(

Here is the log:
```log
Executing action: all (aliases: build)
Running ninja in directory /Users/tennis/src/test-squawk/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
NOTICE: Processing 2 dependencies:
NOTICE: [1/2] espressif/esp_audio_codec (2.0.3)
NOTICE: [2/2] idf (5.3.2)
-- Project sdkconfig file /Users/tennis/src/test-squawk/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
CMake Deprecation Warning at /Users/tennis/esp/esp-idf-v5.3.2/components/mbedtls/mbedtls/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- App "test-squawk" version: 1ed0edf-dirty
-- Adding linker script /Users/tennis/src/test-squawk/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/tennis/src/test-squawk/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.2/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Component idf::main will be linked with -Wl,--whole-archive
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump espressif__esp_audio_codec esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/tennis/esp/esp-idf-v5.3.2/components/app_trace /Users/tennis/esp/esp-idf-v5.3.2/components/app_update /Users/tennis/esp/esp-idf-v5.3.2/components/bootloader /Users/tennis/esp/esp-idf-v5.3.2/components/bootloader_support /Users/tennis/esp/esp-idf-v5.3.2/components/bt /Users/tennis/esp/esp-idf-v5.3.2/components/cmock /Users/tennis/esp/esp-idf-v5.3.2/components/console /Users/tennis/esp/esp-idf-v5.3.2/components/cxx /Users/tennis/esp/esp-idf-v5.3.2/components/driver /Users/tennis/esp/esp-idf-v5.3.2/components/efuse /Users/tennis/esp/esp-idf-v5.3.2/components/esp-tls /Users/tennis/esp/esp-idf-v5.3.2/components/esp_adc /Users/tennis/esp/esp-idf-v5.3.2/components/esp_app_format /Users/tennis/esp/esp-idf-v5.3.2/components/esp_bootloader_format /Users/tennis/esp/esp-idf-v5.3.2/components/esp_coex /Users/tennis/esp/esp-idf-v5.3.2/components/esp_common /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_ana_cmpr /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_cam /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_dac /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_gpio /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_gptimer /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_i2c /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_i2s /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_isp /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_jpeg /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_ledc /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_mcpwm /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_parlio /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_pcnt /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_ppa /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_rmt /Users/tennis/esp/esp-idf-v5.3.2/components/esp_driver_sdio / [... cut ...]
I (439) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (449) SQUAWK_PLAYER: I2S configured successfully
I (449) M4A_PARSE: Get channel 2 bits:16 sample_rate 44100
I (459) M4A_PARSE: Set audio format:3

I (459) M4A_PARSE: Correct audio sample_rate:44100 channel 2
I (649) M4A_PARSE: All sample sent
I (649) SQUAWK_PLAYER: Decode for 7 cpu: 15.07%
I (649) SQUAWK_PLAYER: Playback succeeded for embedded file
I (649) SQUAWK_PLAYER: Free heap size after playback: 374324
I (659) main_task: Returned from app_main()
```

TempoTian, 2024-12-31:
From the log, it already decode OK, I see your code, you need write data to i2s in order to play it.
Currently your code just do decode
```c
static int simple_decoder_write_pcm(uint8_t *data, int size) {
    adjust_volume(data, size, volume);
    write_ctx.decode_size += size;
   // Add your write data to i2s flow like
  //   int ret = i2s_channel_write(tx_chan, data, size, &bytes_written, 1000);
    return size;
}
```

gamename, 2024-12-31:
@TempoTian That worked!  Closing...

closed 2024-12-31 by gamename